evcc-io/evcc #9388: Daheimladen in ModBus

mergeli, 2023-08-15:
### Describe the bug

Die Daheimladen WB ist im MODBus system konfiguriert. Die neueste Software von Daheimladen wurde installiert.
Aber sie läuft nicht, wenn evcc gestartet ist.
Wenn evcc gestoppt wird und der Befehl   sudo evcc -l debug charger eingegeben wird, wird das ausgegeben:

[main  ] INFO 2023/08/15 17:57:24 evcc 0.118.11
[main  ] INFO 2023/08/15 17:57:24 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/15 17:57:24 using sqlite database: /root/.evcc/evcc.db
Power:          0W
Energy:         3602.2kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 232V 233V
Charge status:  A
Enabled:        false
Identifier:     

Bei gestartetem evcc jedoch das:
sudo systemctl start evcc
pi@raspberrypi:~ $ sudo evcc -l debug charger
[main  ] INFO 2023/08/15 17:57:56 evcc 0.118.11
[main  ] INFO 2023/08/15 17:57:56 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/15 17:57:56 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/08/15 17:58:06 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'daheimladen-mb': current limit: read tcp 192.168.178.44:53606->192.168.178.21:502: i/o timeout

### Steps to reproduce

                          

pi@raspberrypi:~ $ sudo systemctl stop evcc
pi@raspberrypi:~ $ sudo evcc -l debug charger
[main  ] INFO 2023/08/15 17:57:24 evcc 0.118.11
[main  ] INFO 2023/08/15 17:57:24 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/15 17:57:24 using sqlite database: /root/.evcc/evcc.db
Power:          0W
Energy:         3602.2kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 232V 232V 233V
Charge status:  A
Enabled:        false
Identifier:     
pi@raspberrypi:~ $ sudo systemctl start evcc
pi@raspberrypi:~ $ sudo evcc -l debug charger
[main  ] INFO 2023/08/15 17:57:56 evcc 0.118.11
[main  ] INFO 2023/08/15 17:57:56 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/15 17:57:56 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/08/15 17:58:06 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'daheimladen-mb': current limit: read tcp 192.168.178.44:53606->192.168.178.21:502: i/o timeout
pi@raspberrypi:~ $ 

### Configuration details

```yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 192.168.178.20  
  name: grid1
- type: template
  template: rct-power 
  usage: pv  
  host: 192.168.178.20  
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 192.168.178.20 
  name: battery3
  
chargers:
- name: wallbox5
  type: template
  template: daheimladen-mb
  host: 192.168.178.21
  port: 502
 # template: daheimladen 
 # token:  
 # stationid:   
 # name: wallbox5

vehicles:
- name: Ioniq5
  type: template
  template: hyundai
  title: IONIQ 5
  user:   
  password:   
  capacity: 72 
  language: de 
  mode: pv  
  minSoC: 15  
  targetSoC: 80  
  minCurrent: 6  
  maxCurrent: 16  
  cloud: false

- name: CupraBorn
  type: template
  template: cupra
  title: CupraBorn
  user: #Benutzerkonto (zB E-Mail Adresse, UserID etc)
  password: #xyz 
  capacity: 58
  mode: pv
  minSoC: 15
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 16
 # cloud: false  

loadpoints:
- title: Garage
  charger: wallbox5
  vehicles: CupraBorn #Ioniq5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
- title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 200
  bufferSoc: 85
  prioritySoc: 80
   
  maxGridSupplyWhileBatteryCharging: 50

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.41 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0704 # [currency]/kWh
```

### Log details

```text
[site  ] DEBUG 2023/08/15 18:18:04 ----
[lp-1  ] DEBUG 2023/08/15 18:18:04 charge power: 0W
[site  ] DEBUG 2023/08/15 18:18:05 pv power: 9162W
[site  ] DEBUG 2023/08/15 18:18:05 battery soc: 97%
[site  ] DEBUG 2023/08/15 18:18:05 battery power: -2W
[site  ] DEBUG 2023/08/15 18:18:05 grid power: -8782W
[site  ] DEBUG 2023/08/15 18:18:05 site power: -8583W
[lp-1  ] DEBUG 2023/08/15 18:18:05 charge voltages: [234 232 235]V
[lp-1  ] DEBUG 2023/08/15 18:18:05 detected connected phases: 3p
[lp-1  ] DEBUG 2023/08/15 18:18:05 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/15 18:18:05 charge total import: 3602.200kWh
[lp-1  ] DEBUG 2023/08/15 18:18:05 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11

VolkerK62, 2023-08-15:
Die Fehlermeldung ist normal. Der service von evcc belegt die Modbus Verbindung und es ist nur eine möglich.

mergeli, 2023-08-15:
Alles klar. Vielen Dank.

closed 2023-08-15 by andig