evcc-io/evcc #9397: Fahrzeug wird als offline angezeigt

redhawk07, 2023-08-16:
### Describe the bug

Nachdem lange Zeit die Software einwandfrei funktioniert hat, wird mein Cupra jetzt in der Web-Oberfläche als offline angezeigt. Folgende Meldung bekomme ich im Log per sudo journalctl -fau evcc:

Aug 16 07:53:00 ubuntu systemd[1]: Started evcc.
Aug 16 07:53:00 ubuntu evcc[68411]: [main  ] INFO 2023/08/16 07:53:00 evcc 0.118.11
Aug 16 07:53:00 ubuntu evcc[68411]: [main  ] INFO 2023/08/16 07:53:00 using config file: /etc/evcc.yaml
Aug 16 07:53:07 ubuntu evcc[68411]: [main  ] ERROR 2023/08/16 07:53:07 creating vehicle car failed: cannot create vehicle 'template': cannot create vehicle 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/xxxxxxx/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout

Rufe ich evcc vehicle auf, während der Dienst nicht läuft, bekomme ich folgende Ausgabe:
evcc vehicle
[main  ] INFO 2023/08/16 08:06:00 evcc 0.118.11
[main  ] INFO 2023/08/16 08:06:00 using config file: /etc/evcc.yaml
[main  ] ERROR 2023/08/16 08:06:07 creating vehicle car failed: cannot create vehicle 'template': cannot create vehicle 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/xxxxxx/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/xxxxxx/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout
Capacity: 58.0kWh
Features: [Offline]

Rufe ich evcc vehicle auf, während der Dienst läuft, bekomme ich folgende Ausgabe mit korrekten Werten:
[main  ] INFO 2023/08/16 08:08:53 evcc 0.118.11
[main  ] INFO 2023/08/16 08:08:53 using config file: /etc/evcc.yaml
Soc:            39%
Capacity:       58.0kWh
Charge status:  A
Range:          142km
Odometer:       12584km
Finish time:    not available
Climate active: false
Target Soc:     100%
OnIdentified:   TargetSoc:80
Features:       []

Version 0.118.11 läuft bei mir als Dienst auf Ubuntu.

### Steps to reproduce

Keine besonderen Schritte. Der Fehler tritt auch nach einem Neustart des Dienstes auf.

### Configuration details

```yaml
vehicles:
- type: template
  template: cupra
  title: Cupra BORN
  user: xxxx
  password: xxxx
  vin: xxxx
  capacity: 58
  targetSoC: 80
  name: car
```

### Log details

```text
[main  ] INFO 2023/08/16 08:01:52 evcc 0.118.11
[main  ] INFO 2023/08/16 08:01:52 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/16 08:01:52 starting ui and api at :7070
[db    ] INFO 2023/08/16 08:01:52 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2023/08/16 08:01:54 site config:
[site  ] INFO 2023/08/16 08:01:54   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/08/16 08:01:54     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/16 08:01:54     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/16 08:01:54     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/08/16 08:01:54   vehicles:
[site  ] INFO 2023/08/16 08:01:54     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/08/16 08:01:54 loadpoint 1:
[lp-1  ] INFO 2023/08/16 08:01:54   mode:        pv
[lp-1  ] INFO 2023/08/16 08:01:54   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/08/16 08:01:54   meters:      charge ✓
[lp-1  ] INFO 2023/08/16 08:01:54     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/08/16 08:01:54 phase timer inactive
[lp-1  ] DEBUG 2023/08/16 08:01:54 pv timer inactive
[lp-1  ] DEBUG 2023/08/16 08:01:54 guard timer inactive
[lp-1  ] INFO 2023/08/16 08:01:54 vehicle updated: unknown -> Cupra BORN
[lp-1  ] DEBUG 2023/08/16 08:01:54 set target soc: 80
[site  ] DEBUG 2023/08/16 08:01:54 ----
[lp-1  ] DEBUG 2023/08/16 08:01:54 charge power: 0W
[site  ] DEBUG 2023/08/16 08:01:54 pv power: 1454W
[site  ] DEBUG 2023/08/16 08:01:55 battery soc: 84%
[site  ] DEBUG 2023/08/16 08:01:55 battery power: -1079W
[site  ] DEBUG 2023/08/16 08:01:55 grid power: 16W
[site  ] DEBUG 2023/08/16 08:01:55 giving priority to battery charging at soc: 84%
[site  ] DEBUG 2023/08/16 08:01:55 site power: 366W
[lp-1  ] DEBUG 2023/08/16 08:01:56 vehicle odometer: 12584km
[lp-1  ] DEBUG 2023/08/16 08:01:56 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/16 08:01:56 charge total import: 1503.603kWh
[lp-1  ] DEBUG 2023/08/16 08:01:56 charger status: B
[lp-1  ] DEBUG 2023/08/16 08:01:56 charge start energy: 1.5e+03kWh
[lp-1  ] INFO 2023/08/16 08:01:56 car connected
[lp-1  ] DEBUG 2023/08/16 08:01:56 pv timer elapse
[lp-1  ] DEBUG 2023/08/16 08:01:56 pv timer inactive
[lp-1  ] DEBUG 2023/08/16 08:01:56 charger: guard elapse
[lp-1  ] DEBUG 2023/08/16 08:01:56 guard timer inactive
[lp-1  ] DEBUG 2023/08/16 08:01:56 charge total import: 1503.603kWh
[lp-1  ] DEBUG 2023/08/16 08:01:56 charger vehicle id: xxxxxxx
[lp-1  ] DEBUG 2023/08/16 08:01:56 vehicle status: A
[lp-1  ] DEBUG 2023/08/16 08:01:56 vehicle soc: 39%
[lp-1  ] DEBUG 2023/08/16 08:01:56 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/08/16 08:01:56 vehicle range: 142km
[lp-1  ] DEBUG 2023/08/16 08:01:56 pv charge current: 0A = 0A + -0.531A (366W @ 3p)
[lp-1  ] DEBUG 2023/08/16 08:01:56 pv enable timer reset
[lp-1  ] DEBUG 2023/08/16 08:01:56 pv timer inactive
[site  ] DEBUG 2023/08/16 08:01:56 ----
[lp-1  ] DEBUG 2023/08/16 08:01:56 charge power: 0W
[site  ] DEBUG 2023/08/16 08:01:56 pv power: 1463W
[site  ] DEBUG 2023/08/16 08:01:57 battery soc: 84%
[site  ] DEBUG 2023/08/16 08:01:57 battery power: -1096W
[site  ] DEBUG 2023/08/16 08:01:57 grid power: 36W
[site  ] DEBUG 2023/08/16 08:01:57 giving priority to battery charging at soc: 84%
[site  ] DEBUG 2023/08/16 08:01:57 site power: 386W
[lp-1  ] DEBUG 2023/08/16 08:01:57 vehicle odometer: 12584km
[lp-1  ] DEBUG 2023/08/16 08:01:57 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/16 08:01:57 charge total import: 1503.603kWh
[lp-1  ] DEBUG 2023/08/16 08:01:57 charger status: B
[lp-1  ] DEBUG 2023/08/16 08:01:57 pv charge current: 0A = 0A + -0.559A (386W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11

StefanSchoof, 2023-08-16:
> Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/xxxxxxx/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout

Es kann keine Verbindung zwischen deinen evcc und dem VW Server hergestellt werden. Entweder temporäre Server Probleme oder dein Netzwerk hat ein Problem oder dein Internet Provider. (Und theoretisch kann noch sich die Server Addresse geändert haben, aber erfahrungsgemäß sind die anderen Dinge sehr viel öfter)

andig, 2023-08-16:
Temporärer Serverfehler bei VW…

closed 2023-08-16 by andig

hermann-59, 2023-09-05:
Ich habe seit mehreren Tagen dasselbe Problem. Bei mir liefert "evcc vehicle" folgendes:
`openhabian@smarthome:~ $ evcc vehicle
[main  ] INFO 2023/09/05 12:19:12 evcc 0.119.4
[main  ] INFO 2023/09/05 12:19:12 using config file: /home/openhabian/evcc.yaml
[db    ] INFO 2023/09/05 12:19:12 using sqlite database: /home/openhabian/.evcc/evcc.db
[eebus ] INFO 2023/09/05 12:19:12 Local SKI:  xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
[main  ] ERROR 2023/09/05 12:19:14 creating vehicle Cupra failed: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
Capacity: 13.0kWh
Features: [Offline]
`

Irgendein Zugriffsproblem. Aber worauf?

andig, 2023-09-05:
https://github.com/evcc-io/evcc/issues/9606