evcc-io/evcc #9398: Cfos: Add 1p3p

premultiply, 2023-08-16:
Ich habe auch die Cfos Wallbox (noch mit dem alten Controller - habe aber kürzlich den neuen Controller Rev. F, der die Phasenumschaltung unterstützt, als Nachrüstkit bestellt). Daher bin ich auch an einer zukünftigen Unterstützung durch evcc interessiert :-)

Die Phasenumschaltung im eingesteckten Zustand ist über das modbus Register 8087 möglich. Das Umschalten dauert dann (lt. Cfos Support-Forum) 120 secs.

_Originally posted by @ckomma in https://github.com/evcc-io/evcc/discussions/9103#discussioncomment-6734811_

andig, 2023-08-16:
Duplikat: https://github.com/evcc-io/evcc/issues/9282

ckomma, 2023-09-04:
Ich habe meine Wallbox nun auf die Solar-Variante upgegraded (das ist die Variante, die mit dem neuen Controller Rev. F, sowie einem zusätzlichen Schütz, die 1p3p Umschaltung unterstützt).
Die Phasenumschaltung bei eingestecktem Ladekabel muss in der CFOS Web-GUI durch den User aktiviert werden.
- Vor Aktivierung gibt /cnf?cmd=modbus&device=evse&read=8113 den Wert 0 zurück.
- Nach Aktivierung gibt /cnf?cmd=modbus&device=evse&read=8113 den Wert 256 zurück.

andig, 2023-09-04:
> Aus Sicherheitsgründen wird die Phasenumschaltung bei unseren Wallboxen erst mit dem nächsten Ladevorgang wirksam, d.h., beim nächsten Einstecken des Ladekabels.

Und hat sich daran jetzt etwas geändert?

ckomma, 2023-09-04:
> > Aus Sicherheitsgründen wird die Phasenumschaltung bei unseren Wallboxen erst mit dem nächsten Ladevorgang wirksam, d.h., beim nächsten Einstecken des Ladekabels.
> 
> Und hat sich daran jetzt etwas geändert?

Aus den Release Notes der "cFos Firmware-Update: Version 1.22":
"Phasenumschaltung 1-/3-phasig: Diese Funktion wurde seit dem letzten Update erweitert. Mit v1.22 können Sie nun die Phasen umschalten, ohne das Ladekabel ein- und auszustecken. Wir arbeiten weiter an der automatischen Umschaltung und bitten um etwas Geduld."
siehe auch https://www.cfos-emobility.de/de/cfos-charging-manager/documentation/phase-switching.htm (Kapitel "Phasenumschaltung während eines Ladevorgangs").

Die "automatische Umschaltung" bezieht sich auf die automatische. regelbasierte Umschaltung durch deie "Cfos-Manager" Software. Die manuelle / API-basierte Umschaltung ist jetzt schon möglich (über modbus Register 8087).

andig, 2023-09-04:
Bitte mal drauf schauen. Passen die Registerwerte? Stimmt die 256 für die Erkennung (oder wäre das eher eine 1?) und können die Phasen direkt geschrieben werden?

premultiply, 2023-09-04:
> Der Anwender muss jedoch über eine Sicherheitsabfrage in der Web-GUI (siehe mein obiger Kommentar) bestätigen, dass er die "Phasenumschaltung bei eingestecktem Ladekabel" (auf eigenes Risiko) aktivieren möchte.

Das klingt aber so als ob die Software über die API (noch?) keinerlei internen Schaltschutz und obligatorisches Timing bietet.

Wollen wir das wirklich riskieren?

andig, 2023-09-04:
Klar, wenn der Anwender des Produktes das so will!

premultiply, 2023-09-04:
Passt. Auf der cFos-Seite ist es auch so beschrieben:

> Bei der Phasenumschaltung im eingesteckten Zustand wird ggf. das Laden angehalten, das Schütz abgeschaltet und der Spannungs-Pegel der Steuerleitung CP für 120sec auf einen Wert nahe 0V gesetzt. Das soll dem Auto signalisieren, dass der Ladestecker ausgesteckt ist und das Auto dazu bringen die Ladeeinheit neu zu initialisieren.

Also alles gut.

andig, 2023-09-06:
@all? Falls sich den PR niemand anschauen möchte würde ich hier zu machen. Merge kann es nur geben wenn mal jemand schaut ob die Register und Werte passen.

closed 2023-09-08 by andig

julred, 2024-01-17:
Ich bin ein bisschen verwirrt. Ich habe auch die cfos Power Brain Solar Wallbox - also mit 1p3p Umschaltung.
Ich sehe ich Quelltext, dass für die Erkennung das Modbus Register 8113 verwendet wird.
Laut Cfos Homepage ist dies Register aber für das Auslesen eines angehefteten S0 Zählers vorgesehen,
Kann jemand bestätigen, dass die Phasenumschaltung, bzw die Erkennung ob das Gerät diese unterstützt, funktioniert?
https://www.cfos-emobility.de/de/cfos-power-brain/modbus-registers.htm

andig, 2024-01-17:
@ckomma kannst Du bestätigen ob CFOS funktioniert? @julred wir können nur mit den verfügbaren Informationen arbeiten. Frag gerne bei Cfos an, wie sich das 1p3p Feature erkennen lässt. In der Doku ist es nicht offensichtlich.

julred, 2024-01-17:
> @ckomma kannst Du bestätigen ob CFOS funktioniert? @julred wir können nur mit den verfügbaren Informationen arbeiten. Frag gerne bei Cfos an, wie sich das 1p3p Feature erkennen lässt. In der Doku ist es nicht offensichtlich.

Habe eine Anfrage beim Support platziert - werde berichten.

michaelmuessing, 2024-01-22:
@ckomma interessant, dass es bei dir funktioniert. 
Bei meiner Cfos Solar funktioniert die automatische Phasenumschaltung mit evcc nicht, ohne evcc mit cfos Lastmanagement wohl. 
Nur wenn ich kurz in der cfos Modbus deaktiviere, ohne evcc einphasig lade und dann wieder Modbus aktiviere, sind alle folgenden Ladevorgänge immerhin einphasig, EVCC denkt aber laut Gui-Anzeige noch es würde dreiphasig laden. 

Die Cfos-Gui ist dann auch nicht konsistent, sondern zeigt in der Wallbox-Kachel 4100W einphasig an, der Wallbox-Zähler ID101 aber korrekt dreiphasig. 

Da es bei dir funktioniert: Hast du evtl. noch andere Einstellungen gemacht als ich?

Cfos: 
Freischaltung Phasenumschaltung mit SN-Eingabe
Lastmanagement deaktiviert
Modbus aktiviert
In den Wallbox-Einstellungen Autom. Phasenumschaltung aktiviert

EVCC: 
in der yaml beim loadpoint 
phases: 0
mincurrent: 6
maxcurrent: 16

und beim cfos-Meter: abweichend von den nicht funktionierenden Standardeinstellungen (ID2 + Port 4702):
id: 1
port: 4701

julred, 2024-01-30:
> Ok- was davon wäre 1p3p? Bit 10? Bit 10 und 11?

Nach meiner Einschätzung sollte zuerst geschaut werden ob Bit 8 gesetzt ist -> Dann ist eine Phasenumschaltung grundsätzlich möglich. Je nachdem wie viel Überschuss vorhanden ist wird dreiphasig geladen [Modbus Register 8087 = 0] oder bei weniger als 4,2 kW Überschuss nur einphasig [Modbus Register 8087 = 1].

https://www.cfos-emobility.de/de/cfos-power-brain/modbus-registers.htm

andig, 2024-01-30:
Ok, ich war mir nicht sicher, ob 1p3p nicht äquivalent mit "2. Relais" ist. Dann können wir die Prüfung gerne auf 8. Bit ändern. PR folgt.

julred, 2024-01-31:
@andig 
Ich glaube die bestehende evcc Implementierung war schon korrekt. Es wird explizit auf 256 abgeprüft.
Es scheint so, dass die Doku seitens cfos nicht sauber ist. Ich habe bei mir das Modbus Register 8113 ausgelesen. Wenn ich die Phasenumschaltung von extern zulasse, lautet mein Registereintrag 256. Also dein PR sollte rückgängig gemacht werden.
Tut mir leid für die Irritationen.

andig, 2024-01-31:
Es wurde auf == Bit 8 geprüft, jetzt wir && Bit 8 geprüft, das macht schon einen Unterschied. Oder hast du jetzt einen Fehler? Welchen?

julred, 2024-01-31:
Okay, ja du hast Recht. Es wird jetzt nur bitweise geprüft. Nehme meine Aussage zurück.

andig, 2024-02-01:
Bitte trace log und neues Issue mit Ausgabe von `/api/state`. Hängt evtl. mit https://github.com/evcc-io/evcc/pull/11571 zusammen.

sebster6, 2024-02-01:
Hey @andig - sorry vielmals, ich kämpfe jetzt schon eine ganze Weile damit, die Trace Logs aus meinem Home Assistant Docker Addon rauszufriemeln bzw. die API passend abzurufen. In der Qualität möchte ich ungerne ein neues Issue eröffnen, hoffe aber dennoch, dass dieses Trace Log ggf. hilft?
Ich werde derweil weiter schauen, wie ich qualifiziertere Logs oder die API-Ausgabe beisteuern kann. 

```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/01 17:24:10 evcc 0.124.0
[main  ] INFO 2024/02/01 17:24:10 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/01 17:24:10 starting ui and api at :7070
[db    ] INFO 2024/02/01 17:24:10 using sqlite database: /data/evcc.db
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT * FROM `settings` 32 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[shelly] TRACE 2024/02/01 17:24:10 GET http://192.168.178.39/shelly
[shelly] TRACE 2024/02/01 17:24:10 {"type":"SHPLG-S","mac":"E09806B5821F","auth":false,"fw":"20230109-114426/v1.12.2-g32055ee","discoverable":false,"num_outputs":1,"num_meters":1}
[db    ] TRACE 2024/02/01 17:24:10 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/02/01 17:24:10 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[cfos  ] TRACE 2024/02/01 17:24:10 modbus: send 00 01 00 00 00 06 01 03 1f b1 00 01
[cfos  ] TRACE 2024/02/01 17:24:10 modbus: recv 00 01 00 00 00 05 01 03 02 01 00
[lp-1  ] DEBUG 2024/02/01 17:24:15 vehicle soc: 30%
[lp-1  ] DEBUG 2024/02/01 17:24:15 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/02/01 17:24:15 vehicle range: 92km
[cache ] TRACE 2024/02/01 17:24:15 lp-1/vehicleSoc: 30
[cfos  ] TRACE 2024/02/01 17:24:15 modbus: send 00 08 00 00 00 06 01 03 1f 9e 00 01
[cache ] TRACE 2024/02/01 17:24:15 lp-1/vehicleTargetSoc: 80
[cache ] TRACE 2024/02/01 17:24:15 lp-1/chargeRemainingDuration: 2562047h47m16.854775807s
[cache ] TRACE 2024/02/01 17:24:15 lp-1/chargeRemainingEnergy: 32222.222222222223
[cache ] TRACE 2024/02/01 17:24:15 lp-1/vehicleRange: 92
[cfos  ] TRACE 2024/02/01 17:24:15 modbus: recv 00 08 00 00 00 05 01 03 02 00 01
[cache ] TRACE 2024/02/01 17:24:15 lp-1/enabled: true
[cache ] TRACE 2024/02/01 17:24:15 lp-1/mode: pv
[lp-1  ] DEBUG 2024/02/01 17:24:15 pv charge current: 0A = 6A + -17.2A (3952W @ 1p)
[cache ] TRACE 2024/02/01 17:24:15 lp-1/planProjectedStart: 0001-01-01 00:00:00 +0000 UTC
[cache ] TRACE 2024/02/01 17:24:15 lp-1/planOverrun: false
[lp-1  ] DEBUG 2024/02/01 17:24:15 site power 3952W >= 1200W disable threshold
[lp-1  ] DEBUG 2024/02/01 17:24:15 pv disable in 0s
[lp-1  ] DEBUG 2024/02/01 17:24:15 pv disable timer elapsed
[cfos  ] TRACE 2024/02/01 17:24:15 modbus: send 00 09 00 00 00 06 01 06 1f 9e 00 00
[cache ] TRACE 2024/02/01 17:24:15 lp-1/pvAction: disable
[cache ] TRACE 2024/02/01 17:24:15 lp-1/pvRemaining: 0s
[... cut ...]
[cache ] TRACE 2024/02/01 17:25:12 lp-1/guardAction: enable
[cache ] TRACE 2024/02/01 17:25:12 greenShareHome: 0.9984202211690363
[cache ] TRACE 2024/02/01 17:25:12 greenShareLoadpoints: 0
[cache ] TRACE 2024/02/01 17:25:12 lp-1/guardRemaining: 4m3.031911401s
[cache ] TRACE 2024/02/01 17:25:12 tariffPriceHome: 0.0825007898894155
[cache ] TRACE 2024/02/01 17:25:12 tariffCo2Home: 0.2511848341232312
[cache ] TRACE 2024/02/01 17:25:12 tariffPriceLoadpoints: 0.399
[cache ] TRACE 2024/02/01 17:25:12 tariffCo2Loadpoints:   ##159
```

andig, 2024-02-01:
Phasenumschaltung wird nicht erkannt

andig, 2024-02-02:
Ohje. In https://github.com/evcc-io/evcc/commit/4c84322c5cacbf4ad5bce2a6ff1c371763563d44 behoben.

julred, 2024-02-09:
Ich habe Rückmeldung vom Support erhalten bzgl des Modbus Register 8113:

```
Bit 8 gibt an, ob in der UI der Relaiswechsel bei eingestecktem Auto
erlaubt wurde.

Bit 9 ist 1, wenn bei der Produktion die Information gespeichert wurde,
ob die Wallbox mit zweitem Relais ausgestattet wurde. Früher
ausgelieferte Geräte ohne dieses Relais haben diese Information nicht,
können aber vom Kunden nachgerüstet worden sein. Wenn Bit 9 0 ist,
bleiben wegen dieser Unbestimmtheit im Web-Interface manche
Eingabefelder sichtbar, die u.U. keine Funktion haben.

Bit 10 sagt, wenn Bit 9 1 ist, ob ein zweites Relais (bei Auslieferung)
angeschlossen ist.
```

Ich denke, so wie es momentan implementiert ist (nur auf Bit 8 prüfen) ist insofern korrekt. Der evcc Nutzer wird ja wissen, ob er eine Box hat mit Phasenumschaltung (Brain Power Solar), oder halt nicht. Evcc muss nur wissen, ob diese Funktion in der cfos aktiviert wurde.