evcc-io/evcc #9412: Easee max charge current 15A: timeout und ignoring inconsistent phases: 1p < 3p observed active

JHammerschmid, 2023-08-17:
Frage: 
1. Wie schnell muss denn die easee geantwortet haben bevor es zum Timeout kommt. (Ticket wegen der Timeouts bereits bei easee offen), Vorschlag mit local api  (Antwort von easee: steht im Backlog, also nicht im scope)
2. ignoring inconsistent phases: 1p < 3p observed active bekomme ich massenhaft. Was bedeutet das im Klartext? Ich vermute im Zusammenhang mit den max charge current timeouts funktioniert die Phasenumschaltung nicht vernünftig, ist aber nur eine Vermutung.
Danke für eure Unterstüzung
[config.txt](https://github.com/evcc-io/evcc/files/12367178/config.txt)

GrimmiMeloni, 2023-08-17:
1. 10 Sekunden. Scheint etwas straff zu sein, allerdings ist im Nightly auch ein Change der das ganze unwichtiger macht, weil der Loadpoint nachzieht, wenn die Easee zu spät dran ist.

2. Es sind scheinbar mehr Phasen aktiv (3) als der Loadpoint erwarte (1). Ob dies mit einer scheiternden Phasenumschaltung zusammenhängt ist ohne Logfile nicht zu beurteilen.

JHammerschmid, 2023-08-17:
Jetz bin ich etwas überfordert. 
Wie komme ich an das nightly?
Parallel werde ich das das log auf debug, easee auf trace, und interval auf 20s einschalten.