evcc-io/evcc #9415: Hausverbrauch 0 bei Ladevorgang über Tasmota Steckdose

flole123, 2023-08-17:
### Describe the bug

Hi in die Runde,

ich habe neben meiner Elli-Wallbox noch einen zweiten Loadpoint mit einer Tasmota Steckdose konfiguriert. Jedoch wird mein Hausverbrauch immer mit 0W angegeben, sobald ich einen Ladevorgang über die Tasmota Steckdose starte. Ich vermute den Fehler in meiner Konfiguration, da ich mir bei der Tasmota Konfig sehr unsicher bin - aber ich kriege ihn nicht behoben.

Vielen Dank schonmal für die Hilfe und liebe Grüße!

### Steps to reproduce

1. Tasmota Steckdose konfigurieren
2. Ladevorgang über interface und Loadpoint mit konfigurierter Tasmota Steckdose starten
3. Hausverbrauch sinkt auf 0, Ladevorgang läuft und wird mit ca. 2,9kW angezeigt. 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: #

interval: 5s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: #  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: #  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: #  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
- name: tasmota_charger 
  type: template 
  template: tasmota 
  usage: charge 
  host: 192.168.178.130 

chargers:
- type: template
  template: ellipro
  ski: #
  ip: #
  name: wallbox5
- type: template
  template: tasmota 
  host: 192.168.178.130  
  user: admin  
  password: '****'  
  channel: 1  
  standbypower: 15  
  name: wallbox6

vehicles:
- type: template
  template: bmw 
  title: iX  
  user: #  
  password: #
  vin: #  
  capacity: 71  
  name: ev1
  icon: car
- type: template
  template: offline 
  title: e-up!  
  capacity: 32  
  name: ev2
  icon: car

loadpoints:
- title: Elli Wallbox
  charger: wallbox5
  vehicle: ev1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  priority: 1
- title: Autosteckdose
  charger: wallbox6
  vehicle: ev2
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  guardduration: 30s
  priority: 0

site:
  title: #
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: # # [currency]/kWh

eebus:
  certificate:
      private: |
          #
      public: |
          #
```

### Log details

```text
[lp-2  ] INFO 2023/08/17 14:43:46 car connected
[lp-2  ] DEBUG 2023/08/17 14:43:46 pv timer elapse
[lp-2  ] DEBUG 2023/08/17 14:43:46 pv timer inactive
[lp-2  ] DEBUG 2023/08/17 14:43:46 charger: guard elapse
[lp-2  ] DEBUG 2023/08/17 14:43:46 guard timer inactive
[lp-2  ] DEBUG 2023/08/17 14:43:46 charge total import: 36.695kWh
[lp-2  ] DEBUG 2023/08/17 14:43:46 charge start energy: 36.7kWh
[lp-2  ] INFO 2023/08/17 14:43:46 start charging ->
[lp-2  ] DEBUG 2023/08/17 14:43:46 wake-up timer: stop
[lp-2  ] DEBUG 2023/08/17 14:43:46 max charge current: 16A
[site  ] DEBUG 2023/08/17 14:43:46 ----
[lp-1  ] DEBUG 2023/08/17 14:43:46 charge power: 0W
[lp-2  ] DEBUG 2023/08/17 14:43:46 charge power: 2927W
[site  ] DEBUG 2023/08/17 14:43:48 pv power: 3505W
[site  ] DEBUG 2023/08/17 14:43:51 battery soc: 65%
[site  ] DEBUG 2023/08/17 14:43:51 battery power: -837W
[site  ] DEBUG 2023/08/17 14:43:52 grid power: 1W
[site  ] DEBUG 2023/08/17 14:43:56 grid currents: [-3.73 -3.71 7.69]A
[site  ] DEBUG 2023/08/17 14:43:56 site power: -836W
[lp-2  ] DEBUG 2023/08/17 14:43:56 charge total import: 36.703kWh
[lp-2  ] DEBUG 2023/08/17 14:43:56 charger status: C
[site  ] DEBUG 2023/08/17 14:43:56 ----
[lp-1  ] DEBUG 2023/08/17 14:43:56 charge power: 0W
[lp-2  ] DEBUG 2023/08/17 14:43:57 charge power: 2932W
[site  ] DEBUG 2023/08/17 14:43:58 pv power: 3517W
[site  ] DEBUG 2023/08/17 14:44:01 battery soc: 65%
[site  ] DEBUG 2023/08/17 14:44:01 battery power: -837W
[site  ] DEBUG 2023/08/17 14:44:02 grid power: -0W
[site  ] DEBUG 2023/08/17 14:44:05 grid currents: [-3.72 -3.71 7.69]A
[site  ] DEBUG 2023/08/17 14:44:05 site power: -837W
[lp-1  ] DEBUG 2023/08/17 14:44:05 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/17 14:44:05 charger status: A
[site  ] DEBUG 2023/08/17 14:44:05 ----
```

### What type of operating system are you running?

Linux

### Version

0.118.11

thierolm, 2023-08-17:
Nimm/Lösche mal die `tasmota_charger` Konfiguration aus der `meters` Section ...

Die `wallbox6` Konfiguration in der `chargers`Section reicht aus.

flole123, 2023-08-17:
Danke für den schnellen Hinweis. Habe ich rausgenommen. Das Problem bleibt jedoch bestehen. LG

thierolm, 2023-08-17:
Das liegt dann eher an deinem Huawei grid Meter. Kannst du mal ein Log liefern, dass den Grid-Wert vor dem Anschalten und nach dem Anschalten des Chargers zeigt? Das Log oben fängt zu spät an. Es zeigt aber ganz klar, dass der `grid`  Wert der `site` auf 1 bzw 0 W geht. Ich kenne mich mit dem Huawei Dongle nicht aus ...

flole123, 2023-08-17:
Okay, hier mal die Infos zur Site:

`[site  ] INFO 2023/08/17 15:33:24 site config:
[site  ] INFO 2023/08/17 15:33:24   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/08/17 15:33:24     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/08/17 15:33:24     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/08/17 15:33:24     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/08/17 15:33:24   vehicles:
[site  ] INFO 2023/08/17 15:33:24     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2023/08/17 15:33:24     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/08/17 15:33:24 loadpoint 1:
[lp-1  ] INFO 2023/08/17 15:33:24   mode:        pv
[lp-1  ] INFO 2023/08/17 15:33:24   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/08/17 15:33:24   meters:      charge ✓
[lp-1  ] INFO 2023/08/17 15:33:24     charge:    power ✓ energy ✗ currents ✓
[lp-2  ] INFO 2023/08/17 15:33:24 loadpoint 2:
[lp-2  ] INFO 2023/08/17 15:33:24   mode:        pv
[lp-2  ] INFO 2023/08/17 15:33:24   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/08/17 15:33:24   meters:      charge ✓
[lp-2  ] INFO 2023/08/17 15:33:24     charge:    power ✓ energy ✓ currents ✗`

Vor Anschalten des Loadpoints mit Tasmota Steckdose:
`[site  ] DEBUG 2023/08/17 15:36:07 ----
[lp-1  ] DEBUG 2023/08/17 15:36:07 charge power: 0W
[lp-2  ] DEBUG 2023/08/17 15:36:07 charge power: 0W
[site  ] DEBUG 2023/08/17 15:36:09 pv power: 1256W
[site  ] DEBUG 2023/08/17 15:36:11 battery soc: 70%
[site  ] DEBUG 2023/08/17 15:36:11 battery power: -765W
[site  ] DEBUG 2023/08/17 15:36:12 grid power: -0W
[site  ] DEBUG 2023/08/17 15:36:13 grid currents: [-0.75 -0.57 1.61]A
[site  ] DEBUG 2023/08/17 15:36:14 site power: -765W
[lp-1  ] DEBUG 2023/08/17 15:36:14 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/17 15:36:14 charger status: A
[site  ] DEBUG 2023/08/17 15:36:14 ----`

Nach Anschalten (Modus "Schnell") des Loadpoints mit Tasmota Steckdose:
`[site  ] DEBUG 2023/08/17 15:38:20 ----
[lp-1  ] DEBUG 2023/08/17 15:38:20 charge power: 0W
[lp-2  ] DEBUG 2023/08/17 15:38:20 charge power: 2937W
[site  ] DEBUG 2023/08/17 15:38:22 pv power: 2987W
[site  ] DEBUG 2023/08/17 15:38:26 battery soc: 71%
[site  ] DEBUG 2023/08/17 15:38:26 battery power: -397W
[site  ] DEBUG 2023/08/17 15:38:27 grid power: -0W
[site  ] DEBUG 2023/08/17 15:38:29 grid currents: [-3.73 -3.74 7.61]A
[site  ] DEBUG 2023/08/17 15:38:30 site power: -397W
[lp-1  ] DEBUG 2023/08/17 15:38:30 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/17 15:38:30 charger status: A
[site  ] DEBUG 2023/08/17 15:38:30 ----`

thierolm, 2023-08-17:
Das hat nichts mit dem Tasmota-Charger zu tun. Wie gesagt, weiss ich nicht was der Sun2000 Dongle misst. Wenn ich mir die `site power` , `pv`und `grid power` Werte in deinen 3 Zuands-Logs anschaue, kann ich mit keinen Reim draus machen.
Passen denn die Werte, die dein `evcc -l debug meter` Aufruf ausgibt, zu dem, was in deinem Wechselricher angezeigt wird?

flole123, 2023-08-17:
Ja, die Werte passen gut übereinander.

thierolm, 2023-08-17:
Teste mal, wenn die Sonne nicht mehr scheint und dein WR keinen PV Strom liefert. Irgendwas wird da falsch gemessen. Wie gesagt, ich kenne den Sun2000 nicht...
Aufruf an alle SUN2000 Nutzer hier in der evcc Community: Kann jemand supporten?

flole123, 2023-08-17:
Okay, ich teste und melde mich dann. Nehme gerne weitere Hinweise aus der Community an 😊.

Danke @thierolm 💚

VolkerK62, 2023-08-17:
Vor dem Einschalten
`DEBUG 2023/08/17 15:36:09 pv power: 1256W`

nach dem Einschalten
`DEBUG 2023/08/17 15:38:22 pv power: 2987W `

Ist der Unterschied von 1700W realistisch (Bewölkung?)

flole123, 2023-08-17:
> Vor dem Einschalten
> 
> `DEBUG 2023/08/17 15:36:09 pv power: 1256W`
> 
> 
> 
> nach dem Einschalten
> 
> `DEBUG 2023/08/17 15:38:22 pv power: 2987W `
> 
> 
> 
> Ist der Unterschied von 1700W realistisch (Bewölkung?)

Ich habe in dem Moment nicht genau drauf geachtet. Aber ja, der Unterschied ist durchaus realistisch bei freiem Himmel vs. Bewölkung. Habe ich so auch an anderen Tagen schon gesehen.