evcc-io/evcc #9427: Go-e Charger lädt mit schwankender und zu hoher Ladeleistung

McGrummel, 2023-08-18:
### Describe the bug

Hi,

ich verwende einen Go-e Charger (V2) an einer Renault Zoe. Ich habe sie in das evcc eingebunden. Wenn ich bei der Ladeeinstellung auf "PV" also das reine Überschussladen stelle, dann schwankt die Ladeleistung von 300 Watt bis 6,8 Watt im Takt von wenigen Sekunden (ca. 10 - 20 sec.). Und das obwohl die PV (Daten kommen von der Tesla Powerwall) aktuell einen Überschuss von 3,8 kw bringt. Der Überschuss ist nahezu konstant.

Leider habe ich keine Ahnung woran es liegen könnte, jedoch scheint das Verhalten so nicht gewollt und auch nicht sonderlich gut für den Akku :-)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070
  
sponsortoken: XXXXXXXXXXXXX

log: debug
levels:
  cache: error

# unique installation id
plant: XXXXXXXXXXXXX

interval: 10s # control cycle interval

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: XXXXXXXXXXXXX  
  password: XXXXXXXXXXXXX 
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: XXXXXXXXXXXXX  
  password: XXXXXXXXXXXXX  
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: XXXXXXXXXXXXX  
  password: XXXXXXXXXXXXX  
  name: battery3

chargers:
- type: template
  template: twc3 
  host: XXXXXXXXXXXXX  
  name: wallbox5
- type: template
  template: go-e 
  host: XXXXXXXXXXXXX  
  name: wallbox4

vehicles:
- type: template
  template: renault 
  title: Zoe  
  user: XXXXXXXXXXXXX 
  password: XXXXXXXXXXXXX  
  vin: XXXXXXXXXXXXX  
  capacity: 50  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
- title: Sonnenlader
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Version

0.118.11

closed 2023-08-18 by andig

McGrummel, 2023-08-18:
Warum sofort geschlossen? Ein Tipp wie ich es gelöst bekomme, wäre toll gewesen.

andig, 2023-08-18:
Weil a) die angefragten Informationen fehlen (Logfile) und b) die Suchfunktion dich bei Zoe sofort auf das Stichwort `minCurrent` führt. Es ist kein offenes Issue für die Entwicklung.