evcc-io/evcc #9431: UI permanently shows guard duration, resets on reload

GrimmiMeloni, 2023-08-18:
### Describe the bug

Running the latest nightly I am seeing a guard duration countdown on the UI, even though no guard duration is running according to the logs. After the duration (4 mins) has passed, the status on the UI changes to "loading". 
However, the guard duration also restarts at 4mins on every reload of the UI.

cc @naltatis

### Steps to reproduce

1. Attach vehicle and start charging
2. Guard duration appears
3. reload UI
4. guard duration is reset and countdown on UI starts anew

### Configuration details

```yaml
skipping for now
```

### Log details

```text
example log snippet of an interval while the UI shows the guard duration interval counting down. No mention of the guard timer.

[site  ] DEBUG 2023/08/18 13:47:19 ----
[lp-1  ] DEBUG 2023/08/18 13:47:19 charge power: 5000W
[site  ] DEBUG 2023/08/18 13:47:19 pv power: 5521W
[site  ] DEBUG 2023/08/18 13:47:19 battery soc: 100%
[site  ] DEBUG 2023/08/18 13:47:19 battery power: -10W
[site  ] DEBUG 2023/08/18 13:47:19 grid power: -246W
[site  ] DEBUG 2023/08/18 13:47:19 site power: -156W
[lp-1  ] DEBUG 2023/08/18 13:47:19 charge currents: [20.7 0.014 0.014]A
[lp-1  ] DEBUG 2023/08/18 13:47:19 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/18 13:47:19 charge total import: 1612.050kWh
[lp-1  ] DEBUG 2023/08/18 13:47:19 charger status: C
[lp-1  ] DEBUG 2023/08/18 13:47:19 soc estimated: 67.00% (vehicle: 67.00%)
[lp-1  ] DEBUG 2023/08/18 13:47:19 vehicle soc: 67%
[lp-1  ] DEBUG 2023/08/18 13:47:19 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/08/18 13:47:19 vehicle range: 332km
[lp-1  ] DEBUG 2023/08/18 13:47:19 pv charge current: 21.7A = 21A + 0.678A (-156W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

nightly 0.118.11 (52bbc4e5)

andig, 2023-08-18:
Findest Du raus, welcher Commit schuld ist?

VolkerK62, 2023-08-18:
Das hab ich auch gelegentlich. Ich habe noch nicht herausgefunden, wie ich das reproduzieren kann.
Siehe auch https://github.com/evcc-io/evcc/issues/8375

GrimmiMeloni, 2023-08-18:
Danke für den Hinweis @VolkerK62 !
Ich habe nicht neugestartet, aber seid dem das Auto ab- und später wieder angesteckt wurde ist die UI wieder Ok. Auf die schnelle sieht auch das Log in Bezug auf den Guard in beiden Fällen gleich aus. Ich schaue mir das morgen mal in Ruhe an.

jheinitz, 2023-08-18:
Moin. Ich glaube, dass ich das gleiche Phänomen habe. Alles hat perfekt geladen und nun ist das Auto bei 80%. In evcc habe ich als Grenze 85% eingestellt. Kann es damit zusammen hängen?

Ansonsten keine Fehler. Danke! 

Version: 0.118.11 (Nightly)

GrimmiMeloni, 2023-08-19:
Hey Jens, ich glaube nein. Soweit ich es überblicken kann ist es ein rein optisches Problem. Die Funktionalität scheint nicht betroffen.

andig, 2023-08-19:
Helfen würde rauszufinden welcher commit das gebracht hat.

jheinitz, 2023-08-19:
> Hey Jens, ich glaube nein. Soweit ich es überblicken kann ist es ein rein optisches Problem. Die Funktionalität scheint nicht betroffen. 

Ich glaube, ich habe es etwas falsch ausgedrückt. Funktion ist wie gewollt. Mit 80% meine ich, dass das Fahrzeug Limit erreicht ist und deshalb die Ladung wie erwartet endet. Also auch von meiner Seite nur ein "optisches" Phänomen.

GrimmiMeloni, 2023-08-19:
> Helfen würde rauszufinden welcher commit das gebracht hat.

Das war mein letzter `loadpoint.syncCharger()` change.
Wenn wir `Loadpoint.enabled` mit dem Charger syncen führt dies dazu das wir in `setLimit()` den Block mit enabled (Zeile 721++) Skipper. Somit publishen wir keine updates mehr für den Guard Timer. PR erstellt.

closed 2023-08-20 by andig

jheinitz, 2023-08-28:
Moin!
Ich war gestern etwas verwundet, denn ich habe das Auto mit `0.119.1` geladen, im Auto das Limit auf 90% gestellt und in evcc das Limit auf 85%. Als dann die 85% erreicht waren, stoppte die Ladung auch (0kWh Leistung), aber der Timer lief. Nach einem Refresh startet er erneut bei 2:20 und nach Ablauf auf 0:00 wurde "Verbunden" angezeigt. 

Ich dachte, dass das Problem bereits in `0.119.0` gefixt ist. 

Und nun? Braucht ihr noch mehr Logs oder Traces?

GrimmiMeloni, 2023-08-28:
Moin Jens - ja, ein Log wäre super. Debug müßte vermutlich schon ausreichen. Wirf einfach mal etwas über den Zaun, ich schaue es mir an.

jheinitz, 2023-08-28:
Moin Michael,

habe das Log etwas gekürzt und die ganzen **Idle** Passagen herausgeschnitten. Am 26.8. war alles okay, aber das lag auch daran, dass ich vor Erreichen von *targetSoc* den Enyaq abgezogen habe. Am 27.8. hatte ich dann das Problem.

Vielleicht findest Du ja etwas.

Danke und Gruß
Jens

[evcc_0.119.1_guard_active.log](https://github.com/evcc-io/evcc/files/12451400/evcc_0.119.1_guard_active.log)

GrimmiMeloni, 2023-08-28:
Habe mir das Log mal angeschaut und auch den Code darauf hin untersucht. Kann erstmal keinen Fehler sehen. Der "charger disable" bedeutet daß der Guard Timer definitiv als abgelaufen an die UI gesendet wird. Ich finde auch keine Variante wie er wieder gestartet werden würde.

Kannst Du das reproduzieren, oder war das ein one-off?

jheinitz, 2023-08-28:
Ich habe das gesehen, als Du auch den Issue hier aufgemacht hast. Dann erst wieder am Sonntag. Ich gehe jetzt auch mal auf 0.119.2 und werde weiter beobachten.

PR #9450 (2023-08-20): elapse guard when syncing lp to enabled
fix #9431