evcc-io/evcc #9435: Anzeige Ladeenergie Energiepreis passt nicht

AUSMAUS, 2023-08-18:
### Describe the bug

Der Energiepreis kann nie niedriger als Preis für Einspeisung!

[image]

### Steps to reproduce

Ich hatte von Anfang an die aktuellen Preise in der yaml-Datei.

...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

interval: 10s # control cycle interval

sponsortoken: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: grid1
  type: custom
  power:
    source: http
   uri: http://192.168.178.20/cm?cmnd=Status%208
    jq: .StatusSNS.SML.Power_Import_W
- name: pv2
  type: custom
  power:
    source: http
    uri: http://192.168.178.21/cm?cmnd=Status%208
    jq: .StatusSNS.SML.Power_Import_W

chargers:
- type: template
  template: go-e
  host: 192.168.178.22
  name: wallbox4

vehicles:
- type: template
  template: XXXX
 template: XXX
  title: XXXXX
  user: XXXXXX
  password: XXXXXX
  vin: XXXXXXXX
  capacity: 50
  phases: 1
  icon: car
  cache: 15m
  minSoc: 20
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 20
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox4
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: true

tariffs:
  currency: EUR
 grid:
    type: fixed
    price: 0.281

  feedin:
    type: fixed
    price: 0.192

site:
  title: West
  meters:
    grid: grid1
    pv:
    - pv2
  residualpower: 70
```

### Log details

```text
irrelevant
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11

andig, 2023-08-18:
> Der Energiepreis kann nie niedriger als Preis für Einspeisung!

Der "Preis" für Einspeisung ist negativ.

closed 2023-08-18 by andig

AUSMAUS, 2023-08-18:
Wo steht in "feedin" ein negativer Wert?????

AUSMAUS, 2023-08-18:
Wo steht hier ein negativer Wert? Oder muss der negativ sein? Schaue doch meine Konfig an

AUSMAUS, 2023-08-18:
> > Der Energiepreis kann nie niedriger als Preis für Einspeisung!
> 
> Der "Preis" für Einspeisung ist negativ.

Wo siehst du das?