evcc-io/evcc #9438: EVCC does not start on SE Home Charger when no 1p3p Switch is installed

snowball77, 2023-08-19:
### Describe the bug

I tried EVCC on my Solar Edge Home Charger wallbox which does not have a 1p3p switch and is connected to a SE hybrid inverter. I use the keba modbus template to integrate it.

However, EVCC does not start in this setting. The box fails on wb.conn.ReadHoldingRegisters(kebaRegPhaseSource, 2). I observed two types of failure. Either it says IllegalAccess or it says i/o timeout.

### Steps to reproduce

1. Configure SE wallbox without 1p3p switch
2. Start EVCC (in my case under macos)

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ...

interval: 10s # control cycle interval

sponsortoken: ...

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-inverter
  id: 1
  host: solaredgeinverter.fritz.box
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solaredge-inverter
  id: 1
  host: solaredgeinverter.fritz.box
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv2

chargers:
- type: template
  template: keba-modbus
  id: 255
  host: p30.fritz.box
  port: 502
  modbus: tcpip
  name: wallbox4

vehicles:
- type: template
  template: kia
  title: EV6
  ...
  capacity: 77
  phases: 3
  icon: car
  cache: 15m
  language: de
  mode: off
  minSoc: 20
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 16
  name: ev3

loadpoints:
- title: Stellplatz
  charger: wallbox4
  vehicle: ev3
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: site
  meters:
    grid: grid1
    pv:
    - pv2%
```

### Log details

```text
[main  ] INFO 2023/08/19 07:15:40 evcc 0.118.11
[main  ] INFO 2023/08/19 07:15:40 using config file: /Users/snowball/evcc.yaml
[main  ] INFO 2023/08/19 07:15:40 starting ui and api at :7070
[db    ] INFO 2023/08/19 07:15:40 using sqlite database: /Users/snowball/.evcc/evcc.db
[main  ] FATAL 2023/08/19 07:15:52 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'keba-modbus': read tcp 192.168.1.149:60190->192.168.1.156:502: i/o timeout
[main  ] FATAL 2023/08/19 07:15:52 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11

snowball77, 2023-08-19:
I propose the following patch. It tries to read the modbus register. On failure it assumes the number of phases in the config but not setting the phases variable.

===================================================================
diff --git a/charger/keba-modbus.go b/charger/keba-modbus.go
--- a/charger/keba-modbus.go	(revision 97aef0cf7599d1ff37be47c3eb4067e8c4e85c17)
+++ b/charger/keba-modbus.go	(date 1692404516032)
@@ -104,14 +104,12 @@
 	}
 
 	// phases
-	b, err = wb.conn.ReadHoldingRegisters(kebaRegPhaseSource, 2)
-	if err != nil {
-		return nil, err
-	}
-
 	var phases func(int) error
-	if source := binary.BigEndian.Uint32(b); source == 3 {
-		phases = wb.phases1p3p
+	b, err = wb.conn.ReadHoldingRegisters(kebaRegPhaseSource, 2)
+	if err == nil {
+		if source := binary.BigEndian.Uint32(b); source == 3 {
+			phases = wb.phases1p3p
+		}
 	}
 
 	// failsafe

snowball77, 2023-08-19:
Maybe a warning message could be added as well that defaults are used due to the inability to read the 1p3p feature from the box's modbus interface.

premultiply, 2023-08-19:
Could you create a PR for further discussion/solution for this? Thanks.

snowball77, 2023-08-19:
PR is at https://github.com/evcc-io/evcc/pull/9440

closed 2023-08-19 by andig