evcc-io/evcc #9439: Konfiguration UI: länderspezifische Nummerneingabe

StevieC121176, 2023-08-19:
Ich habe heute einige meiner Fahrzeuge aus der .yaml entfernt und über die UI Config neu angelegt. Das hat sehr gut funktioniert. Look & Feel ist hervorragend.
Eine Kleinigkeit fand ich allerdings etwas irritierend: Mein E-Bike hat eine Kapazität von 0,750 kWh. Überall in EVCC wird auch "," angezeigt. Nur bei der Eingabe der Kapazität muss ich 0.75 eingeben. Warum hier mit "." anstatt "," Zumal der "." auch bei 1000er genutzt wird "4.760 km"
Was noch schön wäre, diese Zusatzinfos wie im 1. Post.

_Originally posted by @StevieC121176 in https://github.com/evcc-io/evcc/issues/6199#issuecomment-1684389267_

andig, 2023-08-19:
Wir sollten hier nichts neues erfinden sondern einfach so funktionieren wie jeder andere Webseite auch. Mir fehlt die Info nach den Spracheinstellungen Deines Rechners- Server wie auch Client. Bin noch nicht überzeugt, dass hier etwas zu tun ist.

StevieC121176, 2023-08-19:
Sprache ist deutsch eingestellt bei mir, sowohl auf dem Rechner als auch in EVCC.
Nur die UI Config Seite in EVCC wird bei mir in englisch angezeigt, der Rest ist alles auf deutsch.

naltatis, 2023-08-19:
> Bin noch nicht überzeugt, dass hier etwas zu tun ist.

@andig Doch das müssen wir auf jeden Fall, damit Anzeige und Eingabe nach dem gleichen Schema funktioniert. Ist aber ein reines UI Thema. Richtung API ists immer gleich.

andig, 2023-08-19:
Ich verstehe das Problem noch nicht. Parsen des Userinputs in entsprechender Locale?

naltatis, 2023-08-19:
Ja, Parsen und(!) Formatieren. In "de" möchte ich "56,5" ins kWh Akkukapazität Feld eingeben können. In der DB muss natürlich `56.5` gespeichert werden. Wenn ich mir den Eintrag aber später wieder im Formular anschaue muss das in der aktuellen Locale angezeigt werden (also "56,5" in "de").

naltatis, 2023-08-20:
@andig sehe gerade, dass wir die Typinformationen an den Feldern nur als nen Ende bekommen:

```# /api/config/templates/vehicle?lang=de&name=offline```
```json
            {
                "Name": "capacity",
                "Description": "Akkukapazität in kWh",
                "Help": "Akkukapazität in kWh",
                "Example": "50",
                "Usages": [
                    "vehicle",
                    "battery"
                ],
              "Type": 5
            },
```

`Type: 5` entspricht in diesem Fall Float:
https://github.com/evcc-io/evcc/blob/6c4af90b7e53ec83ae6d82981c8452d4f76bd5d5/util/templates/paramtype.go#L14-L23

Bekommst du das einfach in einen aussagekräftigen String ausgegeben (`Type: "float`) oder soll ich vorerst mal ein paralleles Mapping der Typen in der UI aufbauen? Vermutlich ändern wir da ja nicht zu häufig.

andig, 2023-08-20:
Dann braucht der `ParamType` noch eine `MarshalJSON` Methode um sich als String auszugeben.

andig, 2023-09-09:
Gibt es eine Webseite die das unterstützen würde? Beispiel?

andig, 2023-09-12:
Ein Beispiel wie andere Seiten das lösen wäre wirklich super.

PR #9846 (2023-09-16): Config UI: localized number input
fixes: #9439

Das hier ist erstmal die einfache Lösung. Durch die Verwendung von `input[type=number]` wird die jeweilige Dezimalschreibweise des Browsers akzeptiert und ein bestehender Wert auch in dieser ausgegeben.

Leider ist hier das Locale des Browsers ausschlaggeben und es kann nicht bspw. durch die Sprachauswahl, übersteuert werden. Also keine perfekte Lösung, sollte aber in den meisten Fällen passen. Zudem gibts durch diesen Ansatz auch ordentliche Fehlermeldungen bei falscher Eingabe.

[image]
closed 2023-09-16 by andig