evcc-io/evcc #9444: MQTT via. TLS

LeoWinterDE, 2023-08-19:
**Is your feature request related to a problem? Please describe.**
Connecting EVCC against an MQTT Broker via. TLS or with an TLS Auth.

**Describe the solution you'd like**
Provide and document a option for the CA/Cert file in the MQTT config to use TLS for the MQTT Broker connection or to even auth against the Broker with keys.

andig, 2023-08-19:
Probier mal bitte tls:// in der Broker uri.

LeoWinterDE, 2023-08-19:
Dann fehlt ihm leider das CA Root cert, bei mir letsencrypt. Oder kann man die validation abschalten?

andig, 2023-08-19:
> Dann fehlt ihm leider das CA Root cert. Oder kann man die validation abschalten?

Es gibt keine (mutual) Validation, alle Root Certs sollten vorhanden sein. Falls nein bitte vollständiges Issue.

closed 2023-08-19 by andig

LeoWinterDE, 2023-08-19:
Nutze euren Docker Build in Homeassistent und letsencrypt für die certs

andig, 2023-08-19:
Dann bitte komplettes Issue. LE muss funktionieren.