evcc-io/evcc #9447: Fehler im Charger Template "myStrom Switch"

VolkerK62, 2023-08-19:
### Describe the bug

kann es sein, dass sich ein Fehler in die Dokumentation eingeschlichen hat?
Ich spiele mit dem Gedanken, über einen myStrom Switch zu laden, wenn nur wenig Überschuss zur Verfügung steht. 
Bei der Steckdose beginnt die Konfiguration als Wallbox in der Doku mit „meters“. Müsste dort nicht „chargers“ stehen?

per Mail

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

VolkerK62, 2023-08-19:
Hab mir das Template angesehen. Kann da nix entdecken.
Kommt das übers Template, oder direkt aus der Doku?

andig, 2023-08-19:
Fixed in https://github.com/evcc-io/evcc/commit/396b0ff6c56992656cc88002ee614031cdab82ed

closed 2023-08-19 by andig

VolkerK62, 2023-08-19:
ah. `usage` führt dann dazu, dass es als meter angezeigt wird?  Wieder was gelernt :)

andig, 2023-08-19:
Sieht so aus. Ist wohl irgendwo hartcodiert.