evcc-io/evcc #9463: Durchschnittliche Ladeleistung bei Unterbrechung

lstefan123, 2023-08-20:
### Describe the bug

Im Ladelog wird die durchschnittliche Ladeleistung angezeigt. Diese beruht bei Unterbrechungen aber auf der letzte Zeitspanne.

[image]

[image]

### Steps to reproduce

1. Eine Stunde laden
2. Ladevorgang stoppen 
3. Ladelogeintrag im Detail ansehen
4. Ladevorgang starten
5. Kurz warten
6. Ladevorgang stoppen
7. Ladelogeintrag im Detail ansehen
Version 118

### Configuration details

```yaml
Nicht relevant.
```

### Log details

```text
Nicht relevant
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11 (97aef0cf)

andig, 2023-08-20:
Was wäre denn das gewünschte Verhalten?

naltatis, 2023-08-20:
Was genau meinst du mit "Ladevorgang stoppen/starten"? Über die UI Moduswechsel oder Abstecken/Anstecken?

naltatis, 2023-08-20:
@lstefan123 Welche Wallbox verwendest du?

lstefan123, 2023-08-20:
Hallo Michael,

eine SmartWB/efseWifi

Viele Grüße, Stefan

lstefan123, 2023-08-20:
Dass dort nicht steht, dass mein Auto mit durchschnittlich 4000kW geladen wurde. 😉 Meine Wallbox kann nur 11kW.

lstefan123, 2023-08-20:
Im UI auf von "Schnell" auf "Aus" und dann wieder auf "schnell"

naltatis, 2023-08-20:
@premultiply kann es sein, dass einige Charger (hier SmartWB) die Ladezeit nach einer Pause wieder zurücksetzen so dass Geladene Energiemenge und Zeit nicht zusammen passen?

https://github.com/evcc-io/evcc/blob/6e88810fa9bf68a598bb04297fcead6312b73773/core/loadpoint.go#L1326

premultiply, 2023-08-20:
Das ist durchaus möglich.
ChargingTime() und ChargedEnergy() sind immer Fehleranfällig weil sie direkt von den Boxen kommen.
Ich empfehle da immer nur TotalEnergy() und kein ChargingTime() zu verwenden wenn man nicht 100% sicher ist dass die Boxen das korrekt(=wie von evcc erwartet) handhaben.

PR #9484 (2023-08-25): evsewifi: fix ChargedTime() and autodetect meter
Fixes #9463 by removal of ChargedTime()
Fixes #9266

- [x] Fixed power detection
- [x] Removed ChargedTime()
- [x] Add phase voltages
closed 2023-08-25 by andig