evcc-io/evcc #9473: Charging with identified vehicle and PV mode does not regulate power on connect

DerAndereAndi, 2023-08-21:
### Describe the bug

When connecting a vehicle, that is identified by the charger (PMCC EEBUS), and the default charge mode is set to `pv`, it will start charging immediately. Due to EEBUS it will start with full power, as the logic to pause charging right away was removed and there is no limit set when connecting the vehicle. This is fine.

But in PV mode, it should start regulating the currents or active PV disable timer, which does not happen.

The log file is cut off, as below the end it is repeating the same until I manually change the charge mode in the UI, e.g. to MinPV

### Steps to reproduce

1. Setup an vehicle that can be identified by some means
2. Set the vehicle to default mode PV
3. Use an EEBUS charger, or maybe other charger, that has no limits set on connection and charging is enabled by default
4. Connect the vehicle

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # host: 0.0.0.0
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

# unique installation id
plant: xxxxx08eed430aefb76f5b3a4684b69b708xxxxx

interval: 5s # control cycle interval

log: debug
levels:
  core: trace
  lp-1: trace
  lp-2: trace
  semp: info
  eebus: trace
  sma: debug

# HEMS
hems:
  type: sma
  VendorID: 111081111
  DeviceID: 11156b3ddaaa
  AllowControl: false

eebus:
  shipid: bbb84c490507aaaf
  interfaces:
  - eth0
  # local signed certificate, required for eebus chargers and generated via `evcc eebus-cert`
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      ...
      -----END CERTIFICATE-----
    private: |
      -----BEGIN EC PRIVATE KEY-----
      ...
      -----END EC PRIVATE KEY-----

# mqtt message broker
mqtt:
  broker: 192.168.1.59:1883
[... 328 lines cut ...]
[lp-1  ] DEBUG 2023/08/21 10:49:55 charger status: A
[site  ] DEBUG 2023/08/21 10:50:00 ----
[lp-1  ] DEBUG 2023/08/21 10:50:00 charge power: 0W
[lp-2  ] DEBUG 2023/08/21 10:50:00 charge power: 11240W
[site  ] DEBUG 2023/08/21 10:50:00 pv power: 4060W
[site  ] DEBUG 2023/08/21 10:50:00 battery soc: 60%
[site  ] DEBUG 2023/08/21 10:50:00 battery power: 4565W
[site  ] DEBUG 2023/08/21 10:50:00 grid power: 3062W
[site  ] DEBUG 2023/08/21 10:50:00 grid powers: [-3990 3481 3571]W
[site  ] DEBUG 2023/08/21 10:50:00 grid currents: [-16.7 15.1 15.2]A
[site  ] DEBUG 2023/08/21 10:50:00 site power: 7677W
[lp-2  ] DEBUG 2023/08/21 10:50:00 charge currents: [16 16.2 16.2]A
[lp-2  ] DEBUG 2023/08/21 10:50:00 detected active phases: 3p
[lp-2  ] DEBUG 2023/08/21 10:50:00 charger status: C
[lp-2  ] DEBUG 2023/08/21 10:50:00 pv charge current: 0A = 0A + -11.1A (7677W @ 3p)
[site  ] DEBUG 2023/08/21 10:50:05 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11 (d3914a3c)

andig, 2023-08-21:
Ich vermute Du kommst hier einfach nie über die 6A default Minimum.

DerAndereAndi, 2023-08-21:
Da bei

```
[lp-2  ] DEBUG 2023/08/21 10:49:20 charger status: C
```

der Status auf C ist, sollte dann nicht der PV disable Timer anspringen? charger enable bewirkt auf EEBUS sowieso nichts.

GrimmiMeloni, 2023-08-23:
Ich denke, das haben wir in https://github.com/evcc-io/evcc/pull/9369 gelöst. Bitte mal mit der 0.119.0 probieren.

DerAndereAndi, 2023-08-25:
Das Verhalten ist identisch.

andig, 2023-08-25:
@DerAndereAndi falls Du das immer reproduzieren kannst: bekommst Du den Commit oder die Range identifiziert?

DerAndereAndi, 2023-08-25:
> @DerAndereAndi falls Du das immer reproduzieren kannst: bekommst Du den Commit oder die Range identifiziert?

Bauchgefühl: Seit beim Start nicht mehr direkt pausiert wird und dann je nach Lademodus freigegeben wird.

infinus, 2023-08-27:
I’m having the same issue. I went from version 0.118.11 using Tesla with WC3 and everything was working perfectly. 

Upgraded to  0.119.1 and now vehicle starts charging immediately when plugged in using whatever the last charge setting was. Turning off works, min+solar sets it to minimum charge but pv only does not regulate charging. 

Ich habe dasselbe Problem. Ich bin von Version 0.118.11 auf Version 0.119.1 mit Tesla und WC3 umgestiegen, und alles hat perfekt funktioniert. Jetzt startet das Fahrzeug sofort mit dem Laden, sobald es eingesteckt wird, und verwendet dabei die zuletzt verwendeten Ladeeinstellungen. Das Ausschalten funktioniert, "min+solar" setzt es auf die minimale Ladung, aber "pv only" reguliert das Laden nicht.

GrimmiMeloni, 2023-08-27:
OK, das Problem scheint mit Chargern aufzutreten die automatisch Starten und bei denen die Implementierung den `Enabled()` Zustand nicht korrekt setzen. Zumindest für die TWC kann ich das nachvollziehen - die speichert lediglich einen internen State. Dieser sagt aber nichts über die Realität aus.

Ich vermute einen Zusammenhang mit #9408 oder #9407.

@infinus kannst Du mal ein Debug Log teilen?

GrimmiMeloni, 2023-08-27:
Just as a thought. Maybe we could (also) check charger status in syncCharger(). Based on the log shared by @DerAndereAndi this would have detected the ongoing charge in his scenario.

Generally speaking, a disabled loadpoint with a charger in status C can never be right.

GrimmiMeloni, 2023-08-27:
Just for context why I am focussing on enabled states of charger and loadpoint. Based on the log shared originally, I can conclude that in lp.pvMaxCurrent() here:

https://github.com/evcc-io/evcc/blob/57384e461b179d55ebb194b3b2c98cedb66f3795/core/loadpoint.go#L1136

both the `mode` as well as the `targetCurrent` condition are met. Therefore I conclude that `lp.enabled` was false (as otherwise the pv disable timer would have started).

Next, as in the same log `lp.syncCharger()` did _**not**_ log the "charger out of sync" message, I also am positive that `charger.enabled` matches `lp.enabled`, i.e. is also `false`.

GrimmiMeloni, 2023-08-27:
One more thing to add. The reason why also no adjustment of the current happens is because evcc correctly calculates a targetCurrent of 0A, however lp.setLimit() is effectively a NoOp in this scenario. TargetCurrent matches lp.current (which is still 0A as well) and the load point also deems itself disabled.

TL;DR - the solution is to either get `charger.Enabled()` right, or if not possible at least to ensure `lp.syncCharger()` correctly detects a charge and updates `lp.enabled` accordingly.

infinus, 2023-08-27:
Here are my log details:

`

```
[main  ] INFO 2023/08/27 16:30:20 evcc 0.119.1
--
[main  ] INFO 2023/08/27 16:30:20 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/27 16:30:20 starting ui and api at :7070
[db    ] INFO 2023/08/27 16:30:20 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2023/08/27 16:30:21 site config:
[site  ] INFO 2023/08/27 16:30:21   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/08/27 16:30:21     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/27 16:30:21     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/08/27 16:30:21   vehicles:
[site  ] INFO 2023/08/27 16:30:21     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/08/27 16:30:21 loadpoint 1:
[lp-1  ] INFO 2023/08/27 16:30:21   mode:        pv
[lp-1  ] INFO 2023/08/27 16:30:21   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/08/27 16:30:21   meters:      charge ✓
[lp-1  ] INFO 2023/08/27 16:30:21     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/08/27 16:30:21 phase timer inactive
[lp-1  ] DEBUG 2023/08/27 16:30:21 pv timer inactive
[lp-1  ] DEBUG 2023/08/27 16:30:21 guard timer inactive
[lp-1  ] INFO 2023/08/27 16:30:21 vehicle updated: unknown -> Infinus
[site  ] WARN 2023/08/27 16:30:21 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/08/27 16:30:21 ----
[lp-1  ] DEBUG 2023/08/27 16:30:21 charge power: 1493W
[site  ] DEBUG 2023/08/27 16:30:23 pv power: 516W
[site  ] DEBUG 2023/08/27 16:30:24 grid power: 1876W
[site  ] DEBUG 2023/08/27 16:30:24 site power: 1876W
[lp-1  ] DEBUG 2023/08/27 16:30:24 vehicle odometer: 62767km
[lp-1  ] DEBUG 2023/08/27 16:30:24 charge voltages: [245 0 0]V
[lp-1  ] DEBUG 2023/08/27 16:30:24 detected connected phases: 1p
[lp-1  ] DEBUG 2023/08/27 16:30:24 charge currents: [6.1 0 0]A
[lp-1  ] DEBUG 2023/08/27 16:30:24 charger status: C
[lp-1  ] INFO 2023/08/27 16:30:24 car connected
[lp-1  ] DEBUG 2023/08/27 16:30:24 pv timer elapse
[lp-1  ] DEBUG 2023/08/27 16:30:24 pv timer inactive
[lp-1  ] DEBUG 2023/08/27 16:30:24 charger: guard elapse
[lp-1  ] DEBUG 2023/08/27 16:30:24 guard timer inactive
[lp-1  ] INFO 2023/08/27 16:30:24 start charging ->
[lp-1  ] DEBUG 2023/08/27 16:30:24 wake-up timer: stop
[lp-1  ] DEBUG 2023/08/27 16:30:24 vehicle soc: 89%
[lp-1  ] DEBUG 2023/08/27 16:30:24 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/27 16:30:24 vehicle range: 391km
[lp-1  ] DEBUG 2023/08/27 16:30:24 pv charge current: 0A = 0A + -8.15A (1876W @ 1p)
[lp-1  ] DEBUG 2023/08/27 16:30:24 pv enable timer reset
[lp-1  ] DEBUG 2023/08/27 16:30:24 pv timer inactive
[site  ] DEBUG 2023/08/27 16:30:31 ----
[lp-1  ] DEBUG 2023/08/27 16:30:31 charge power: 1494W
[site  ] DEBUG 2023/08/27 16:30:33 pv power: 519W
[site  ] DEBUG 2023/08/27 16:30:34 grid power: 1879W
[site  ] DEBUG 2023/08/27 16:30:34 site power: 1879W
[lp-1  ] DEBUG 2023/08/27 16:30:34 vehicle odometer: 62767km
[lp-1  ] DEBUG 2023/08/27 16:30:34 charge voltages: [245 0 0]V
[lp-1  ] DEBUG 2023/08/27 16:30:34 detected connected phases: 1p
[lp-1  ] DEBUG 2023/08/27 16:30:34 charge currents: [6.1 0 0]A
[lp-1  ] DEBUG 2023/08/27 16:30:34 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/27 16:30:34 charger status: C
[lp-1  ] DEBUG 2023/08/27 16:30:34 vehicle soc: 89%
[lp-1  ] DEBUG 2023/08/27 16:30:34 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/27 16:30:34 vehicle range: 391km
[lp-1  ] DEBUG 2023/08/27 16:30:34 pv charge current: 0A = 0A + -8.17A (1879W @ 1p)
[site  ] DEBUG 2023/08/27 16:30:41 ----
[lp-1  ] DEBUG 2023/08/27 16:30:41 charge power: 1495W
[site  ] DEBUG 2023/08/27 16:30:42 pv power: 520W
[site  ] DEBUG 2023/08/27 16:30:44 grid power: 1901W
[site  ] DEBUG 2023/08/27 16:30:44 site power: 1901W
[lp-1  ] DEBUG 2023/08/27 16:30:44 charge voltages: [245 0 0]V
[... 29 lines cut ...]
[lp-1  ] DEBUG 2023/08/27 16:31:06 charge currents: [6.1 0 0]A
[lp-1  ] DEBUG 2023/08/27 16:31:06 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/27 16:31:06 charger status: C
[lp-1  ] DEBUG 2023/08/27 16:31:06 vehicle soc: 89%
[lp-1  ] DEBUG 2023/08/27 16:31:06 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/27 16:31:06 vehicle range: 391km
[lp-1  ] DEBUG 2023/08/27 16:31:06 pv charge current: 0A = 0A + -8.43A (1940W @ 1p)
[site  ] DEBUG 2023/08/27 16:31:11 ----
[lp-1  ] DEBUG 2023/08/27 16:31:11 charge power: 1495W
[site  ] DEBUG 2023/08/27 16:31:13 pv power: 521W
[site  ] DEBUG 2023/08/27 16:31:14 grid power: 1916W
[site  ] DEBUG 2023/08/27 16:31:14 site power: 1916W
[lp-1  ] DEBUG 2023/08/27 16:31:14 charge voltages: [245 0 0]V
[lp-1  ] DEBUG 2023/08/27 16:31:14 detected connected phases: 1p
[lp-1  ] DEBUG 2023/08/27 16:31:14 charge currents: [6.1 0 0]A
[lp-1  ] DEBUG 2023/08/27 16:31:14 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/27 16:31:14 charger status: C
[lp-1  ] DEBUG 2023/08/27 16:31:14 vehicle soc: 89%
[lp-1  ] DEBUG 2023/08/27 16:31:14 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/27 16:31:14 vehicle range: 391km
[lp-1  ] DEBUG 2023/08/27 16:31:14 pv charge current: 0A = 0A + -8.33A (1916W @ 1p)
[site  ] DEBUG 2023/08/27 16:31:21 ----
```

`

andig, 2023-08-28:
Could you please try 0.119.2 in a few minutes?

@DerAndereAndi falls @GrimmiMeloni richtig liegt wäre das ein Bug im Charger der durch die Änderungen am Loadpoint jetzt sichtbar wird. Ein "enabled=true" darf der LP eigentlich nicht zurück geben wenn er gleichzeitig lädt!

GrimmiMeloni, 2023-08-28:
> "enabled=true"

Das sollte `enabled=false` heißen, oder?

DerAndereAndi, 2023-08-28:
Danke für die Analyse @GrimmiMeloni Ich werde einen PR machen.

closed 2023-08-28 by andig

GrimmiMeloni, 2023-08-28:
We probably also need a fix for TWC, which we should track in a separate issue.
That said @infinus can you please check if 0.119.2 changes anything for you? Thanks!

infinus, 2023-08-28:
I tested 0.119.2 and the TWC is unaffected it will still charge as soon as plugged in. I can start a new issue.

PR #9570 (2023-08-29): Always treat charger as enabled when charging 
Fix https://github.com/evcc-io/evcc/issues/9540. Possibly fixes #9473.

- sync current only on enabled lp
- charger logic error detection now corrects enabled state of lp

/cc @DerAndereAndi hiermit werden die Logikfehler im Charger wieder vom Loadpoint abgefangen. Ich würde daher drauf verzichten, das im TWC zusätzlich zu korrigieren.