evcc-io/evcc #9476: Volvo XC90 Recharge zegt "conected" gibt aber keine Daten aus

rssprinter, 2023-08-21:
### Describe the bug

Hallo Zusammen,

ich habe erfolgreich evcc aufgesetzt und habe nur ein Problem.

Die Daten von meinem Volvo XC90 BJ 2022 werden nicht angezeigt.
Ich habe einen vccapikey erstellt und eingetragen damit wird das Auto auch als "online" angezeigt.
Leider werden die Daten für Ladestand und Reichweite .....   nicht angezeigt.

Beim BMW klappt das problemlos.

Muss ich noch etwas einstellen? Oder freigeben. ?

Beste Grüße

Matthias

Im Anhang meine Config:

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: be.......

interval: 10s # control cycle interval

sponsortoken: eyJhb......

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.36 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.07 # [currency]/kWh

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.x  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
[... 197 lines cut ...]
[site  ] DEBUG 2023/08/21 11:01:56 battery soc: 0%
[site  ] DEBUG 2023/08/21 11:01:56 battery power: 0W
[site  ] DEBUG 2023/08/21 11:01:56 grid power: -6777W
[site  ] DEBUG 2023/08/21 11:01:56 giving loadpoint priority for additional: 913W
[site  ] DEBUG 2023/08/21 11:01:56 site power: -7690W
[lp-1  ] DEBUG 2023/08/21 11:01:56 charge currents: [0.494 0 0.076]A
[lp-1  ] DEBUG 2023/08/21 11:01:56 charge total import: 81.840kWh
[lp-1  ] DEBUG 2023/08/21 11:01:56 charger status: C
[lp-1  ] DEBUG 2023/08/21 11:01:56 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1  ] DEBUG 2023/08/21 11:01:56 vehicle soc: 0%
[lp-1  ] DEBUG 2023/08/21 11:01:56 vehicle range: 0km
[lp-1  ] DEBUG 2023/08/21 11:01:56 available power 7683W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/08/21 11:01:56 start phase scale3p timer
[lp-1  ] DEBUG 2023/08/21 11:01:56 phase scale3p in 1m0s
[lp-1  ] DEBUG 2023/08/21 11:01:56 pv charge current: 13.6A = 2.49A + 11.1A (-7690W @ 3p)
[lp-1  ] DEBUG 2023/08/21 11:01:57 max charge current: 13A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11

VolkerK62, 2023-08-21:
Schau dir doch mal die Ausgabe von `evcc vehicle --log trace` an, ob in der Antwort diese Daten enthalten sind.

andig, 2023-08-21:
> lp Garage at prio 3 gets additional 913W from Poolpumpe at prio 1, total 913W

Die Priorisierung der Ladepunkte funktioniert- schön zu sehen :)

andig, 2023-08-21:
Gelöscht- da waren Deine Tokens enthalten!

rssprinter, 2023-08-21:
Also Fehlermeldungen zum Volvo kommen keine:

Der Status wird mit 200 angegeben, was I.O. ist nach der Volvo Beschreibung

Status code | Description
200 OK        | Indicates a successful request.

[volvo-cc] TRACE 2023/08/21 20:29:19 {
  "vehicles" : [ {
    "id" : "***"
  } ]
}

ev4
---
[volvo-cc] TRACE 2023/08/21 20:29:23 GET https://api.volvocars.com/energy/v1/vehicles/***/recharge-status
[volvo-cc] TRACE 2023/08/21 20:29:23 {
  "status" : 200,
  "operationId" : "731.....................................",
  "data" : { }
}
Soc:           0%
Capacity:      12.6kWh
Charge status: A
Range:         0km
Finish time:   0001-01-01 00:00:00 +0000 UTC
Features:      []

VolkerK62, 2023-08-22:
Das ist weniger als nichts. 
Der Volvo wird nur deshalb als "verbunden" angezeigt, weil du ihn dem Loadpoint fest zugewiesen hast `vehicle: ev4`.
Und "online" ist er, weil der Abruf keinen Fehler bringt. Darin sind aber keine Daten enthalten.

andig, 2023-08-22:
Gerne mal Zugangsdaten an info@evcc.io. Hier kommen wir anscheinend nicht weiter, die Logs sind unvollständig.

closed 2023-08-22 by andig

andig, 2023-08-22:
Leider kommt da überhaupt nix:

```
{
  "status" : 200,
  "operationId" : "d3984e4b-83db-4eea-bf95-afb15a5efd24",
  "data" : { }
}
```

Jetzt ist guter Rat teuer. Ich würde vorschlagen, mal von Deinem Konto aus den Volvo Entwicklersupport zu kontaktieren warum das so ist.

m-kloeckner, 2023-11-23:
Ich habe leider exakt das gleiche Problem. Auch ein XC90 Recharge MJ 2022. Die Volvo API liefert keine Daten.
Gibt es hier neue Erkenntnisse?

````
[volvo-cc] TRACE 2023/11/23 14:39:37 GET https://api.volvocars.com/energy/v1/vehicles/xxxx/recharge-status
[volvo-cc] TRACE 2023/11/23 14:39:37 {
  "status" : 200,
  "operationId" : "e2a80ed7-xxxx-xxxx-xxxx-626a459850f2",
  "data" : { }
}
Soc:           0%
Capacity:      11.6kWh
Charge status: A
Range:         0km
Finish time:   0001-01-01 00:53:28 +0053 LMT
Features:      []
````

andig, 2023-11-23:
Am Vorschlag hat sich nichts geändert:

> https://github.com/evcc-io/evcc/issues/9476#issuecomment-1688064356

m-kloeckner, 2023-11-23:
OK. Scheinbar ist die Sache ist ziemlich eindeutig. Laut Volvo unterstützen nur PHEVs ab BJ2023 (die mit Google) die Energy API.

> This API works for all plug-in hybrid and electric vehicles which are equipped with Google Automotive System (Google built-in).
> 
> List of supported cars:
> 
> XC40 and C40 BEV.
> XC60 PHEV, S90 PHEV, V90 PHEV from model year 2022.
> XC90 PHEV, S60 PHEV, V60 PHEV from model year 2023. 
https://developer.volvocars.com/apis/energy/v1/overview/#availability

Was bedeutet, dass ältere Modelle mit der Legacy Implementierung arbeiten müssen.