evcc-io/evcc #9478: Aktivierung enphase Live-Daten über evcc

JAK0721, 2023-08-21:
@premultiply , bb7674 und ich haben beim Testen festgestellt, dass beim Aktivieren des "Live-Status" in der enphase app der 15min Timer nach dem Ablauf  den "sc_stream" wieder auf  "disabled" setzt und somit keine Datenaktualisierung mehr erfolgt. 

Folgende Idee: 
Bei der Abfrage der Daten mittels `https://192.168.xxx.xxx/ivp/livedata/status` den Status von "sc_stream" auswerten. Sollte "sc_stream": "disabled" sein, mittels `curl -f -k -H "Authorization:bearer {token}" -H "Content-Type:application/json" -d "{\"enable\":1}" https://192.168.xxx.xxx/ivp/livedata/stream` "sc_stream": "enable" setzen.
Bei einem Abfrage-Intervall von 5s wären die Daten max. 10s veraltet.

Ich habe versucht im Code herauszufinden, wie die enphase.yaml weiterverarbeitet wird, habe es aber nicht gefunden.
Gibt es eine grobe Erklärung wie die vielen Code ineinander greifen?

_Originally posted by @JAK0721 in https://github.com/evcc-io/evcc/issues/9332#issuecomment-1684902637_

premultiply, 2023-08-21:
Ich würde dann wieder auf den alten Stand zurückdrehen
Batterie kann man dann bei Bedarf individuell mit sowas als Custom-Meter einbinden.

JAK0721, 2023-08-21:
@premultiply, mit dem alten Template gibt es 2 Probleme:

**battery power über http://{{ .host }}/production.json?details=1** 
1. Leistung für das Laden / Entladen der Hausbatterie wird nicht bedient (Custom-Meter Einbindung müßte dann mit eine gesonderten Meßtechnik, z.B. shelly 3EM umsetzen, richtig?)

**battery soc: über http://{{ .host }}/ivp/ensemble/inventory**
2. liefert keinen Mittelwert aller installierten Batteriemodule, sondern nur Einzelwert. Bei meinem Bekannten sind 4 Batteriemodule verbaut und die SoC's weichen zum Teil deutlich von einander ab.

Frage: Ist mein Vorschlag zu individuell für die Umsetzung in der evcc Software? 

Ich persönlich würde das neue Template aus den oben genannten beiden Punkten weiter nutzen wollen.
Wenn es keine generische Lösung durch die evcc Software geben wird, würde ich mittels einem kleinem python script die Abfrage http://{{ .host }}/ivp/livedata/status und Auswertung "sc_stream" machen. Bei disabled dann den oben beschriebenen Befehl raus schreiben. Nachteil: ggf. Kollisionen bei  parallelen Abfragen über evcc und das python script.

Kann man ggf. den Inhalt von "sc_stream" mit in die influxDB schreiben? Dann kann sich das python scirpt den Status dort abgreifen.

Weitere Frage: Gibt es die Möglichkeit beide Templates zu behalten? Oder verwirrt das die Anwender zu viel? (beide arbeiten so nicht Plug & Play und in beiden Fällen braucht der User einen Fix, der eine kostet Geld der andere benötigt Software Kenntnisse)

Besten Dank schon mal vorab für Deine Rückmeldung

premultiply, 2023-08-21:
Ja, dann am besten via Custom-Meter die Daten ziehen und verarbeiten bis Enphase mal brauchbare APIs anbietet bzw. die entsprechend erweitert.

JAK0721, 2023-08-21:
das Custom-Meter löst das SoC Problem ja nicht
Daher nochmal die Frage. Beide Templates anbieten?

Für einen Fork: In welchen Softwaremodulen wird das Template verarbeitet und die influxDB?

premultiply, 2023-08-21:
> das Custom-Meter löst das SoC Problem ja nicht

Warum?

closed 2023-08-21 by premultiply

premultiply, 2023-08-21:
Einfach mittels custom-Meter z.B. vom Typ script selbst was zusammendengeln was per curl die nötigen Requests macht.

JAK0721, 2023-08-21:
jetzt habe durch das Lesen der Doku verstanden, was Du meinst. Werde ich mal ausprobieren

Ch10De, 2023-08-22:
An einer Lösung für die Enphase Batterien wäre ich auch sehr interessiert. Gerne kann ich was testen.

bb7674, 2023-08-23:
ich weiß grad absolut nicht wovon ihr redet, würde mich aber schon interessieren was so ein "Custom-Meter" ist bzw. was heisst das? Ich teste auch wieder gerne mit wenn ich weiß was ich tun soll

JAK0721, 2023-08-23:
@Ch10De, @bb7674, leider wird mein Vorschlag oben nicht umgesetzt. Auch wurde das enphase Template auf die alte Version umgestellt.  Daten kommen jetzt wieder über http://{{ .host }}/production.json. Da wird aber die Batterieleistung nicht bedient. Bei Laden und Entladen kommt 0W. Das Überschussladen funktioniert, aber die Balken Anzeige ist mist.
Der Vorschlag von premultiply war nun, die fehlenden Batteriedaten über ein Custom-Meter wie in https://docs.evcc.io/docs/reference/configuration/meters/#custom & https://docs.evcc.io/docs/reference/plugins beschrieben.
Ich denke ich bekomme die Abfrage schon so an laufen und könnte dann die Lösung hier einstellen. Das löst aber weiterhin nicht das Problem mit den 15min Timer welcher durch die Enphase app getriggert wird wenn man sich die Live-Daten anschaut.

JAK0721, 2023-08-23:
Für das Timer-Problem habe ich jetzt folgende Idee:
Bei den Plugins besteht die Möglichkeit bash-scripte einzubinden. Das Bash-Script fragt den Status von "sc_stream" ab. Wenn es auf disabled steht, wird es mit dem oben geannten Befehl auf enabled gesetzt. So kann evcc alle Daten abgreifen und man kann auch die Live-Daten der enphase App nutzen.

premultiply, 2023-08-23:
Die Erklärung dass es nur mit "AC Battery" funktioniert steht im Template.
Die wirkliche Lösung wäre wenn Enphase die API auch für die aktuellen Produkte anpassen bzw. reparieren würde.

JAK0721, 2023-08-23:
@premultiply, ich gebe Dir recht, habe da aber wenig Hoffnung. Ich hatte mit den amerikanischen Support Kontakt über 2 Kanäle Kontakt. Von beiden habe ich eine negative Rückmeldung bekommen. Teilweise verstehen die auch das Problem nicht. Man kommt leider nicht zum 3rd Level support durch, sondern scheitert bereits am 1st.
Einen Angriff starte ich noch, aber...

"Die Erklärung dass es nur mit "AC Battery" funktioniert steht im Template." --> hat doch niemand was anderes behauptet. Bei meinem Freund und auch bei @bb7674 sind "AC Batteries" verbaut.

JAK0721, 2023-08-23:
@premultiply, noch eine Frage zur Umsetzung mit einem Bash. 
Wenn in der evcc.yaml bei # control cycle interval 10s eingetragen sind, wird dann zeitsynchron mit der Abfrage der Daten (Grid, PV,...) auch das Script ausgeführt?
Wenn dem so ist, würde ich mit einem 5s delay im Script beim Enphase Gateway den Status "sc_stream" abfragen und bei Bedarf auf enable stellen. Ich hoffe somit Konflikte durch parallelen Zugriff vermeiden zu können. 

Danke!

bb7674, 2023-08-23:
also beitragen kann ich hier jetzt momentan grad mal gar nichts dazu, verfolgen tu ich es natürlich......ich belass jetzt mal alles so wie es ist momentan, soll heissen die Batterie wird in % angegeben und der Live- Zustand wird auch erfasst.
Wenn ich dann mal bei der Enphase APP die Live Ansicht gewählt habe weiß ich ja was ich zu tun habe damit es wieder funktioniert.
Ist für mich immer noch besser als dass mir die Live Daten komplett fehlen von der Batterie

Ch10De, 2023-08-23:
@JAK0721 Die Einbindung als Custom Meter traue ich mir nicht zu, da müsste ich mich einarbeiten. Aber aus den Changes der enphase.yaml in #9332 kann man sicher was gebrauchen. Deinen curl Command fürs sc_stream enablen hätte ich versucht über cron laufen zu lassen, aber wenn das als Script Plugin geht, umso besser.
Das Nightly hatte ich am Laufen, halt mit der 15min Problematik.

Ch10De, 2023-08-23:
@JAK0721 danke für den Hinweis auf #9501. Das Nightly, welches bei mir die Batterieleistung geliefert hat, müsste die 0.118.11+1692341876 gewesen sein.

JAK0721, 2023-08-23:
@Ch10De, Danke! Werde ich ausprobieren.

JAK0721, 2023-08-25:
@bb7674 / @Ch10De, ich teste gerade das automatische einschalten der Live-Daten über das Label "sc_stream". Mein bash script funktioniert schon mal. Jetzt hoffe ich, dass die Integration über die evcc.yaml auch noch funktioniert.
[image]

Ch10De, 2023-08-25:
@JAK0721 habe die enphase als custom meter eingebunden bekommen (inklusive battery und soc) und halte den sc_stream mittels deines curl Befehls als cronjob auf enabled. Das läuft seit gestern Abend super :-).
Ich poste später hier die Details zum custom meter in der evcc.yaml

JAK0721, 2023-08-25:
@Ch10De, danke für update. Hört sich gut an. Ja poste mal das custom meter. Man kann ja nur lernen.
In welchem Intervall läßt Du den Cronjob schreiben?

Ch10De, 2023-08-25:
Der Cronjob läuft jede Minute, aber im Prinzip sollte ja auch alle 15 Minuten reichen. Bislang habe ich keine Aussetzer oder Probleme beobachtet.
In der evcc.yaml habe ich es nicht geschafft nur die battery als custom einzubinden und pv und grid bei template zu belassen. Ich musste alle drei als custom definieren, aber das ganze Thema ist neu für mich:

`meters:
  - name: Enphase_pv
    type: custom
    power:
      source: http
      uri: http://192.168.x.x/ivp/livedata/status
      method: GET
      auth:
        type: bearer
        password: <token>
      insecure: true
      jq: .meters.pv.agg_p_mw
      scale: 0.001
  - name: Enphase_grid
    type: custom
    power:
      source: http
      uri: http://192.168.x.x/ivp/livedata/status
      method: GET
      auth:
        type: bearer
        password: <token>
      insecure: true
      jq: .meters.grid.agg_p_mw
      scale: 0.001
  - name: Enphase_battery
    type: custom
    power:
      source: http
      uri: http://192.168.x.x/ivp/livedata/status
      method: GET
      auth:
        type: bearer
        password: <token>
      insecure: true
      jq: .meters.storage.agg_p_mw
      scale: 0.001
    soc:
      source: http
      uri: http://192.168.x.x/ivp/livedata/status
      method: GET
      auth:
        type: bearer
        password: <token>
      insecure: true
      jq: .meters.soc
    capacity: 7 # kWh`

Die usage wird bei custom metern über die site festgelegt, das hat mich auch 'ne Weile gekostet.
Viel Erfolg.

JAK0721, 2023-08-25:
@Ch10De, top Danke! Mit Deiner Variante hast Du den Vorteil, dass Aktualisierungen von evcc genutzt werden können. 
Bei meinem Freund läuft noch die 118.11 Version mit http://192.168.x.x/ivp/livedata/status im enphase Template.
Wenn die Kombination aus Template und Custom nicht funktioniert, hat aus meiner Sicht hat das Zurückdrehen auf die alte Abfrage über http://{{ .host }}/production.json somit keinen Mehrwert gebracht, es sei man legt keinen Wert auf die Batterieleistung. Der Rest funktioniert ja dauerhaft.

JAK0721, 2023-08-25:
@Ch10De, ich habe es jetzt auch gelöst, allerdings mittels meines Scripts, welches alle 30s den Status prüft und nur einen Schreibbefehl an "sc_stream" auslöst, wenn dieses Label auf disabled steht. 
Wie oben bereits angemerkt, hatte ich Sorge wegen der hohen Anzahl Schreibvorgänge. Wenn Du jede Minute schreibst, kommst Du im Jahr auf 525.600 Schreibvorgänge. 
Ich bin kein IT / ET Spezialist, aber wenn man nach max. Schreibzyklen eines EEPROM googelt kommt da häufig 1.000.000. Dann wäre die Lebensdauer theoretisch nach 2 Jahren erreicht. Kann sein, das keine Alterung auftritt wenn man die Speicherzellen immer mit den gleichen Wert beschreibt. Aber wie gesagt, ich bin da kein Experte.

fano0001, 2023-11-03:
Vielen Dank @Ch10De für das Template.
Bei mir klappt es jetzt auch mit der Enphase IQ Battery 3T

JAK0721, 2023-11-03:
https://github.com/evcc-io/evcc/discussions/9536#discussioncomment-6823465

PR #9479 (2023-08-21): Enphase: revert API change
Fixes #9478
Reverts #9332