evcc-io/evcc #9487: Alfen: Autodetect 1p3p setting

premultiply, 2023-08-22:
So the error: "modus exception 3 illegal data value" I was able to solve by enabling 1 and 3 phase charging on the Alfen Eve App.

Everything is working, thanks all for assisting me.

_Originally posted by @koen81 in https://github.com/evcc-io/evcc/discussions/9178#discussioncomment-6788431_

andig, 2023-08-22:
...und wie sollen wir das detektieren? Ein direkter Schreibversuch wäre blöd. Könnten wir lesen was drin steht? Müsste man mal mit einer echten Alfen ausprobieren @koen81

BDBAfH, 2023-08-23:
@andig:
Es gibt zwar ein Modbus-Register, aber das liefert mir (neueste Firmware 6.4.0-4210) unabhängig von der tatsächlichen Einstellung immer den Wert "3".
In der Alfen-Modbus-Doku habe ich folgendes gefunden:

> Note: the checkbox for ‘Allow 1- and 3-phased charging’. This option must be active to allow an Energy
Management System to control switching between single-phase and three-phases while charging. **In the
current implementation this option must be checked locally using the Service Installer Application.**

Ich vermute mal, der Wert wird derzeit irgendwo gespeichert, aber nicht in dem angegebenen Modbus-Register.

HTH

premultiply, 2023-09-03:
https://github.com/evcc-io/evcc/discussions/9214#discussioncomment-6895960

Macrisu, 2023-09-03:
Is this related with my problem, here?

https://github.com/evcc-io/evcc/discussions/9691

andig, 2023-09-09:
> Es gibt zwar ein Modbus-Register, aber das liefert mir (neueste Firmware 6.4.0-4210) unabhängig von der tatsächlichen Einstellung immer den Wert "3".

Können wir hier zu machen oder gibts noch was zu tun?

closed 2023-09-14 by andig