evcc-io/evcc #9488: SENEC.Home, Umstellung von HTTP auf HTTPS durch SENEC

Mogl403, 2023-08-22:
### Discussed in https://github.com/evcc-io/evcc/discussions/9457

<div type='discussions-op-text'>

<sup>Originally posted by **Mogl403** August 20, 2023</sup>
Hallo zusammen, ich hoffe das ich ein "New Issue" erstellen darf.

Bis letzte Woche Donnestag (17.08.) lief EVCC mit meinem Speicher SENEC Hybrid duo, Wallbox Webasto und dem VW e-UP ziemlich "geschmeidig.
Doch nun bringt mir EVCC die Fehlermeldung "Error: Post "http://192.xxx.x.xxx/lala.cgi": dial tcp 192.xxx.x.xxx:80: connect: connection refused".
SENEC hat letzte Woche am 17. August eine neue Software auf die Speicher gespielt und wohl HTTP durch HTTPS abgelöst.
Ich denke das nun die senec-home.yaml angepasst werden müsste.

Danke schon mal vorab.

andig, 2023-08-22:
Betrifft das alle senecs?

Mogl403, 2023-08-22:
Ich kann "nur" von meinem "SENEC.Home V3 hybrid duo" sprechen.
Revision MCU: 3825
Revision MCU-BL: 2307
Revision NPU-REGS: 10
Revision-Image: 2307
Revision GUI: 2966
(wie von phorbi https://gist.github.com/smashnet/82ad0b9d7f0ba2e5098e6649ba08f88a?permalink_comment_id=4664821#gistcomment-4664821 beschrieben).

ledermann, 2023-08-22:
Das Knifflige ist, dass erst "Revision MCU 3825" `https` verlangt, die Vorgängerversion 3824 aber ausschließlich `http`. Es gibt kein Redirect. Und es haben längst noch nicht alle SENEC-Kunden die neue Version.

Bei SOLECTRUS habe ich das gleiche Problem und es erst mal so gelöst, dass man das Protokoll (`http`/`https`) konfigurieren kann:

- https://github.com/solectrus/senec-collector/issues/224
- https://solectrus.de/blog/2023-08-18-unterstuetzung-fuer-neue-senec-firmware/

Die perfekte Lösung wäre, erst `http` zu versuchen und wenn das fehlschlägt, auf `https` zu wechseln.

closed 2023-08-29 by andig

dscho, 2023-09-01:
> Ich kann "nur" von meinem "SENEC.Home V3 hybrid duo" sprechen.

Und ich "nur" von dem meinen (selbes Modell), aber andere Revisionen:

REVISION MCU: 1824
REVISION MCU-BL: 954
REVISION NPU-REGS: 09
REVISION NPU-IMAGE: 2011
REVISION GUI: 0964

Und _ich_ bekomme jetzt so was:

[image]

Im Log steht auch schön und brav:

```
Sep 01 08:53:22 raspberrypi evcc[19816]: [site  ] ERROR 2023/09/01 08:53:22 pv 1 power: Post "https://192.168.178.84/lala.cgi": dial tcp 192.168.178.84:443: connect: connection refused
Sep 01 08:53:22 raspberrypi evcc[19816]: [site  ] DEBUG 2023/09/01 08:53:22 pv power: 0W
```

Weil halt https _nicht_ geht.

Wat nu?

neubi4, 2023-09-01:
@dscho wenn du das Update mit https noch nicht bekommen hast, dann müsstest du in der evcc config vom Senec `schema: http` setzen können, damit er dann http statt https nimmt.

https://docs.evcc.io/docs/devices/meters#senec-home

andig, 2023-09-01:
Da gabs leider keine gute Lösung. Hätten wir http default gelassen wärs bei den Leuten mit https kaputt gegangen. Schlecht gemacht von Senec- best practice wäre ein redirect gewesen…

dscho, 2023-09-01:
> wenn du das Update mit https noch nicht bekommen hast, dann müsstest du in der evcc config vom Senec `schema: http` setzen können, damit er dann http statt https nimmt.

@neubi4 danke Dir! Ich habe einfach mal das hier in der `evcc.yml` probiert, und es klappt:

```yaml
meters:
- type: template
  template: senec-home
  usage: grid
  schema: http
  host: 192.168.178.84
  name: grid1
- type: template
  template: senec-home
  usage: pv
  schema: http
  host: 192.168.178.84
  name: bsh
- type: template
  template: senec-home
  usage: battery
  schema: http
  host: 192.168.178.84
  name: senec
```

> Da gabs leider keine gute Lösung. Hätten wir http default gelassen wärs bei den Leuten mit https kaputt gegangen. Schlecht gemacht von Senec- best practice wäre ein redirect gewesen…

Stimmt schon. Aber könnten wir nicht eine lecker "versuch doch HTTPS und wenn das nicht tut, HTTP" Stellschraube in EVCC einführen, @andig? Wäre ein pragmatischer Weg, die Benutzer vor SENECs Entscheidungen zu schützen.

neubi4, 2023-09-01:
Ich glaube dass sich dafür der Aufwand nicht lohnt. Die neue Software kommt eh zwangsweise für alle, und damit auch das https für alle, der Rollout läuft nur gestaffelt aktuell. 

Ichs schätze in spätestens 4 Wochen haben fast alle das Update, und die wenigen die dann noch auf einer alten Version hängen können dann das `schema` Setting nutzen.

PR #9492 (2023-08-29): Senec: fix HTTPS and add energy
Fixes #9488