evcc-io/evcc #9502: Dockerfile incorrect COPY commands

JeremyMahieu, 2023-08-22:
### Describe the bug

Building dockerfile results in this error.
```
Step 7/45 : COPY .*.js .
ERROR: Service 'evcc' failed to build : When using COPY with more than one source file, the destination must be a directory and end with a /
```

### Steps to reproduce

1. Build docker file

### Configuration details

```yaml
Irrelevant
```

### Log details

```text
No logs before running
```

### What type of operating system are you running?

Linux

StefanSchoof, 2023-08-22:
Which version of docker do you use?

andig, 2023-08-22:
Closed in https://github.com/evcc-io/evcc/commit/748d2f5d3dbf77bce3ccf7e6e15282b586f3bd95. War nie ein Problem, aber zwei Zeilen drüber machen wir es ja auch "richtig".

closed 2023-08-22 by andig