evcc-io/evcc #9504: Guard against implausible soc values

andig, 2023-08-22:
bufferStartSoc: 90
Wie dem auch sei. Störungen, nicht plausible temporäre falsche Werte sind, aus meiner Erfahrung, Tagesgeschäft.
Mein Ansatz wäre auch alles Soc <0 > 100,  in meinem Fall 5838% ignorieren. Der nächste Zyklus, wie auch in meinem Fall, wird es hoffentlich korrigieren.
Alles abdecken und lösen wird niemals möglich sein. Die Frage die sich mir stellt ist: können wir damit leben?
Ich kann das, so lange es erklärbar ist. Ist ja offensichtlich nur temporär.
Sollte es jedoch regelmäßig oder öfters auftreten so zitiere ich gerne meinen ehemaligen Chef: "Stellen sie das ab"
Da ihr einen super Job macht liegt es an euch, ich habe, wie gesagt, damit kein Problem.

_Originally posted by @JHammerschmid in https://github.com/evcc-io/evcc/discussions/9460#discussioncomment-6795545_

naltatis, 2023-08-23:
Kleiner Hinweis: SoC verwenden wir inzwischen auch für Temperaturen (`heating`). Die Range ist im Celsius Bereich idR. auch von 0-100, wenn wir da aber mal Fahrenheit raus machen (noch zu diskutieren wie) könnte es zu komischen Effekten kommen wenn wir zu viele Annahmen treffen.

andig, 2023-08-23:
> Wir könnten bei Fahrenheit das gleiche machen wie bei Meilen und es rein auf der Ausgabeseite umrechnen. 

Unbedingt. Der Temperaturhinweis ist gut- im Moment würde es ja dankbarerweise passen, müssen wir im Hinterkopf behalten oder je nach "features" unterschiedlich validieren.

closed 2023-08-25 by andig