evcc-io/evcc #9516: Einstellungen Phasenupdate funktioniert nicht

Klausmd5, 2023-08-23:
### Describe the bug

Der Phasenwechsel funktioniert über das UI nicht.
[image]

In der Console wird es bis zur Methode [hier](https://github.com/evcc-io/evcc/blob/91de2a00e3dc14e66d11c9c7a5d369d6d38f71b2/core/loadpoint_api.go#L184) durchgereicht.
Weiter passiert jedoch zumindest mit dem Go E charger v3 http nichts, also kein Aufruf für 1p3p change.

Ich vermute, dass die Methode aufgrund des decorators nicht ausgeführt wird. Weil es im testmodus auf der CLI funktioniert. `./evcc charger -p 3`

Weiters ist fraglich, ob evcc die Phasenumschaltung über den PV Modus macht – testen kann ich das aktuell noch nicht.

### Steps to reproduce

1. add go e charger
2. try to change phases in settings ui
3. look console  -  DEBUG 2023/08/24 00:45:39 set phases: 1
4. check charger api http://192.168.0.53/api/status?filter=rfb,alw,acu,adi,amp,psm

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xx

interval: 10s # control cycle interval

sponsortoken: xx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

chargers:
- type: template
  template: go-e-v3
  host: 192.168.0.53
  name: wallbox1

loadpoints:
- title: Garage
  charger: wallbox1
  vehicle: eGolf
  mode: minpv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

meters:
  - name: grid
    type: custom
    power:
      source: http
      auth:
[... 143 lines cut ...]
[site  ] ERROR 2023/08/24 01:13:26 battery 1 soc: Get "http://localhost:7072/api/state": dial tcp [::1]:7072: connect: connection refused
[site  ] DEBUG 2023/08/24 01:13:26 battery soc: 0%
[site  ] DEBUG 2023/08/24 01:13:26 battery power: 0W
[site  ] ERROR 2023/08/24 01:13:27 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2023/08/24 01:13:27 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
 

./evcc charger -p 3
[main  ] INFO 2023/08/24 01:13:53 evcc 0.118.11 (9c99a2e8)
[main  ] INFO 2023/08/24 01:13:53 using config file: /home/klaus/Downloads/test/evcc/evcc.yaml
[db    ] INFO 2023/08/24 01:13:53 using sqlite database: /home/klaus/.evcc/evcc.db
phase switcher
change
amp=16
change
psm=2
```

### What type of operating system are you running?

Linux

### Version

0.118.11

Klausmd5, 2023-08-23:
Zur lösung müsste man 
`lp.charger.(api.PhaseSwitcher).Phases1p3p(phases)`

[hier](https://github.com/evcc-io/evcc/blob/91de2a00e3dc14e66d11c9c7a5d369d6d38f71b2/core/loadpoint_api.go#L184) mit einbauen.

andig, 2023-08-24:
Woher kommen die Logmeldungen? Ist das tatsächlich ein offizielles evcc Release?

andig, 2023-08-24:
Davon unabhängig: schon dein Netzzähler liefert keine Werte- ab da wird auch nix mehr gesteuert…

closed 2023-08-24 by andig

Klausmd5, 2023-08-24:
> Woher kommen die Logmeldungen? Ist das tatsächlich ein offizielles evcc Release?

Nein, hab im go e Controller bei den ampere etwas Customizing betrieben.

> Davon unabhängig: schon dein Netzzähler liefert keine Werte- ab da wird auch nix mehr gesteuert…

Das ist mir bewusst, aber das web ui sollte doch trotzdem den Befehl an wie wallbox schicken können, wenn ich das manuell möchte.

andig, 2023-08-24:
Nein, tut es nicht weil das erst im nachgelagerten Regelzyklus passiert.