evcc-io/evcc #9522: 0.119.0 - runtime error: slice bounds out of range [:1] with length 0

mdkeil, 2023-08-24:
### Describe the bug

seitdem ich heute morgen auf v0.119.0 geupdated habe, ist mir bereits 5mal der Docker-Container mit folgender Meldung im Log abgeschmiert:

```
panic: runtime error: slice bounds out of range [:1] with length 0
goroutine 263 [running]:
github.com/evcc-io/evcc/api.ChargeStatusString({0x0, 0x0})
	github.com/evcc-io/evcc/api/api.go:50 +0x185
github.com/evcc-io/evcc/charger.(*Salia).Status(0xc001387a28?)
	github.com/evcc-io/evcc/charger/hardybarth-salia.go:161 +0xc5
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0xc0009d0c00)
	github.com/evcc-io/evcc/core/loadpoint.go:869 +0x3c
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc0009d0c00, 0x3ff0000000000000?, 0x0, 0x0?, 0x0?, 0x3ff0000000000000, 0xc0006bb0d8, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1494 +0x212
github.com/evcc-io/evcc/core.(*Site).update(0xc001734000, {0x311d738?, 0xc0009d0c00})
	github.com/evcc-io/evcc/core/site.go:754 +0x5ba
github.com/evcc-io/evcc/core.(*Site).Run(0xc001734000, 0xc00174a180, 0x2540be400)
	github.com/evcc-io/evcc/core/site.go:854 +0x2e5
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:282 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:281 +0x1a13
```

### Steps to reproduce

interner Fahrzeugwechsel von unknown -> Fahrzeug und umgekehrt; also wenn ein Fahrzeug an die WB connected/disconnected.

Was ich noch nicht getestet habe, ob es nur beim Kia passiert und/oder auch beim Peugeot. Bei Bedarf kann ich noch ein Log mit debug/trace-Informationen liefern.

### Configuration details

```yaml
network:
  schema: http
  host: evcc
  port: 7070
interval: 10s

influx:
  url: http://192.168.1.2:8086
  database: evcc
  user: evcc
  password: **********

mqtt:
  broker: 192.168.1.2:1883
  topic: evcc
  user: mosquitto
  password: **********

log: info
levels:
  #site: trace
  #salia: trace
  #lp-1: trace

vehicles:
  - name: my_peugeot
    type: template
    template: flobz
    title: e-208
    url: http://192.168.1.2:5050
    vin: **********
    capacity: 46
    mode: pv
    #minSoC: 40
    targetSoC: 80
  - name: my_kia
    type: custom
    title: Niro-EV
    soc:
      source: mqtt
      topic: nodered/evnotify/all
      jq: .soc_display
    status:
      source: mqtt
      topic: nodered/evnotify/status
    range:
      source: mqtt
      topic: nodered/evnotify/all
      jq: .range
    #akey: c53705
    #token: **********
    capacity: 64.8
    onIdentify:
      mode: pv
      #minSoC: 40
      targetSoC: 80

chargers:
  - name: my_charger
    type: template
    template: hardybarth-salia
    host: 192.168.1.11

meters:
  - name: my_grid
    type: template
    template: victron-energy
    usage: grid
    host: 192.168.1.3
  - name: my_pv
    type: template
    template: victron-energy
    usage: pv
    host: 192.168.1.3
  - name: my_battery
    type: template
    template: victron-energy
    usage: battery
    host: 192.168.1.3
    capacity: 14.34

site:
  title: Home
  meters:
    grid: my_grid
    pv:  my_pv
    battery: my_battery
  prioritySoc: 60
  bufferSoc: 75
  bufferStartSoc: 90
  residualPower: 100

loadpoints:
  - title: Carport
    charger: my_charger
    mode: pv
    phases: 3
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true
    #enable:
    #  threshold: -2500
    #  delay: 3m
    #disable:
    #  threshold: 2000
    #  delay: 5m

tariffs:
  grid:
    #type: fixed
    #price: 0.3355
    type: tibber
    token: "**********"
  feedin:
    type: fixed
    price: 0.0736

plant: **********
sponsortoken: **********
```

### Log details

```text
[site  ] WARN 2023/08/24 10:47:49 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2023/08/24 10:47:49 car connected
[lp-1  ] INFO 2023/08/24 10:47:49 start charging ->
[lp-1  ] INFO 2023/08/24 10:47:49 vehicle updated: unknown -> Niro-EV
[lp-1  ] INFO 2023/08/24 10:47:59 stop charging <-
[lp-1  ] INFO 2023/08/24 10:49:19 start charging ->
[lp-1  ] INFO 2023/08/24 10:58:49 stop charging <-
[lp-1  ] INFO 2023/08/24 11:03:49 start charging ->
[lp-1  ] INFO 2023/08/24 11:23:49 stop charging <-
[lp-1  ] INFO 2023/08/24 13:53:39 car disconnected
[lp-1  ] INFO 2023/08/24 13:53:39 vehicle updated: Niro-EV -> unknown
panic: runtime error: slice bounds out of range [:1] with length 0
goroutine 263 [running]:
github.com/evcc-io/evcc/api.ChargeStatusString({0x0, 0x0})
	github.com/evcc-io/evcc/api/api.go:50 +0x185
github.com/evcc-io/evcc/charger.(*Salia).Status(0xc001387a28?)
	github.com/evcc-io/evcc/charger/hardybarth-salia.go:161 +0xc5
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargerStatus(0xc0009d0c00)
	github.com/evcc-io/evcc/core/loadpoint.go:869 +0x3c
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc0009d0c00, 0x3ff0000000000000?, 0x0, 0x0?, 0x0?, 0x3ff0000000000000, 0xc0006bb0d8, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1494 +0x212
github.com/evcc-io/evcc/core.(*Site).update(0xc001734000, {0x311d738?, 0xc0009d0c00})
	github.com/evcc-io/evcc/core/site.go:754 +0x5ba
github.com/evcc-io/evcc/core.(*Site).Run(0xc001734000, 0xc00174a180, 0x2540be400)
	github.com/evcc-io/evcc/core/site.go:854 +0x2e5
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:282 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:281 +0x1a13
```

### What type of operating system are you running?

Docker container

### Version

0.119.0

andig, 2023-08-24:
Im Master behoben, fraglich ist allerdings, wie Du da hin gekommen bist. Falls es erneut auftritt bitte trace log für den Charger.

closed 2023-08-24 by andig

mdkeil, 2023-08-24:
Kann ich beantworten, das liegt am Verhalten der WB, das einige Dienste unter gewissen Umständen in mehr oder minder regelmäßigen Abständen intern neugestartet werden. In den pre 0.119.0 Versionen wurde das im Log immer mit 'invalid charger status' beantwortet. Warum es nun zum Absturz führt hat, werdet ihr besser wissen als ich.., 😉

andig, 2023-08-24:
Der Grund ist hier: https://github.com/evcc-io/evcc/commit/0f32a4386a0c63f6846465ca978afb0680e95f7e#diff-bf98d5fab5bcfabded9769069e7318683f549ec91cff84c5f87a9ae3a3ca3d8dR51. Länge Null wurde nicht abgefangen.

mdkeil, 2023-08-24:
#9142 wird dann vermutlich bei mir das Problem verursacht haben, da vorher dieser Fall zumindest gewollt oder ungewollt abgefangen wurde.

troem, 2024-03-13:
bekomme auch den Fehler 

`panic: runtime error: slice bounds out of range [1:0]`

Config details:
`network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s

log: info

meters:
  - name: grid
    type: template
    template: janitza
    usage: grid   
   
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.231 # Hostname
    port: 502 # Port 

  - name: pv1
    type: template
    template: sunspec-inverter
    usage: pv 
    modbus: tcpip
    id: 1
    host: 192.168.178.232 #WR1
    port: 502 # Port    

  - name: pv2
    type: template
    template: sma-webbox
    usage: pv 
    modbus: tcpip
    id: 13
    host: 192.168.178.233 #WR2
    port: 502 # Port  

  - name: battery1
    type: template
    template: sma-sbs-modbus
    usage: battery
    modbus: tcpip
    id: 3
    host: 192.168.178.235 # Hostname
    port: 502 # Port   
    capacity: 81

  - name: battery2
    type: template
    template: sma-sbs-modbus
    usage: battery
    modbus: tcpip
    id: 2    
    host: 192.168.178.236 # Hostname
    port: 502 # Port
    capacity: 161     

chargers:
  - name: my_dummy
    type: custom
    status: # charger status A..F
      source: script
      cmd: /bin/sh -c 'echo b'
    enabled: # charger enabled state (true/false or 0/1)
      source: script
      cmd: /bin/sh -c 'echo false'
    enable: # set charger enabled state (true/false or 0/1)
      source: script
      cmd: /bin/sh -c 'echo false'
    maxcurrent: # set charger max current (A)
      source: script

site:
  title: Testkunde
  meters:
    grid: grid
    pv: 
      - pv1
      - pv2
    battery: 
      - battery1
      - battery2      
tariffs:
  co2:
    type: grünstromindex
    zip: 79111
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.05 # [currency]/kWh

loadpoints:
  - title: loadpoint # display name for UI
    charger: my_dummy # Wallbox Referenz`

andig, 2024-03-13:
@troem da das Issue hier seit 6 Monaten gelöst ist mach bitte ein neues auf mit komplettem Logfile.