evcc-io/evcc #9527: Switch to single phase charging not working

richterolaf, 2023-08-24:
### Describe the bug

I have in the settings single phase only as presetting.
In PV Mode EVCC is ignoring this and is charging with 3~ by using my home battery 
With battery setting at 100% house only EVCC is not charging even a surplus of 3.2kW is available 

### Steps to reproduce

1. bug fix is needed 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 3b655175b6e260572ead533565bc46fb08b9a84cbb239c92cd80afe97da1b749

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3Nzk0MzU1OTUsImlhdCI6MTY4NDgyNzU5NSwiaXNzIjoiZXZjYy5pbyIsInN1YiI6InJpY2h0ZXJvbGFmIn0.7VikR9iRxDFRSmz4vRFDJJgA3DNlHk69W7EpYEbkb4Q

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.59  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.59  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.59  
  port: 8080  
  name: battery3

chargers:
- type: template
  template: easee 
  user: olaf.richter@me.com  
  password: PKaZ6fc9{\&6  
  charger: EH9768ST  
  name: wallbox4
- type: template
  template: easee 
  user: olaf.richter@me.com  
  password: PKaZ6fc9{\&6  
  charger: EHWWLQS6  
  name: wallbox5

loadpoints:
- title: Innen
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: Aussen
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Villa Sunshine
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 3b655175b6e260572ead533565bc46fb08b9a84cbb239c92cd80afe97da1b749

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3Nzk0MzU1OTUsImlhdCI6MTY4NDgyNzU5NSwiaXNzIjoiZXZjYy5pbyIsInN1YiI6InJpY2h0ZXJvbGFmIn0.7VikR9iRxDFRSmz4vRFDJJgA3DNlHk69W7EpYEbkb4Q

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.59  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.59  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.59  
  port: 8080  
  name: battery3

chargers:
- type: template
  template: easee 
  user: olaf.richter@me.com  
  password: PKaZ6fc9{\&6  
  charger: EH9768ST  
  name: wallbox4
- type: template
  template: easee 
  user: olaf.richter@me.com  
  password: PKaZ6fc9{\&6  
  charger: EHWWLQS6  
  name: wallbox5

loadpoints:
- title: Innen
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: Aussen
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Villa Sunshine
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### What type of operating system are you running?

Linux

### Version

Latest

VolkerK62, 2023-08-24:
Ich hoffe, die Login Daten sind ein Fake. Ansonsten sind sie verbrannt.

VolkerK62, 2023-08-24:
`phases: 3` am Loadpoint = fest auf 3 Phasen konfiguriert. 
Das sollte `phases: 1` stehen.

andig, 2023-08-24:
Log fehlt

closed 2023-08-24 by andig

naltatis, 2023-08-24:
@richterolaf Magst du noch mal genau beschreiben was du machst und wie sich die Box verhält? Ich hab bei mir mit dem aktuellen Softwarestand und meiner Easee gerade folgendes probiert:

- Phaseneinstellung UI: automatischer Wechsel
- Modus: Sofort
- Auto lädt 3-phasig
- Phaseneinstellung UI: 1-phasig
- Ladung stoppt und startet 1-phasig

Verhält sich das bei dir anders? Bitte häng mal ein Logfile für den Zeitraum mit an wo das Fehlverhalten auftritt:
https://docs.evcc.io/docs/guides/setup/#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen