evcc-io/evcc #9540: Tesla does not stop charging in 0.119.0

yipikaye, 2023-08-25:
### Describe the bug

Hi All,
I've been using EVCC with tesla wall charger for a while now. Often my Tesla would start charging for no reason (probably schedule or off peak or whatever) but EVCC was always great and would stop charging. EVCC would throw an error like "logic error: disabled but charging" but it would anyway stop the charging. Now it has stopped doing this. I also noticed yesterday it kept charging into the night but I don't know if EVCC stopped the charging and it started again or if it never stopped. 

### Steps to reproduce

1. Plug in Tesla
2. EVCC status PV or off, no sun
3. Wait
4. Car starts charging EVCC does not stop 
...

### Configuration details

```yaml
No change since working several months
```

### Log details

```text
I will collect tonight, on my phone sorry
```

### What type of operating system are you running?

Linux

### Version

0.119.0

andig, 2023-08-26:
Bitte trace log für Tesla und debug für den Rest. Ohne ist keine Analyse möglich.

yipikaye, 2023-08-26:
I have updated the logs for a few minutes where the carger was charging when it shouldn't. 

I changed to min+solar, then to off and it stopped charging, then back to solar

yipikaye, 2023-08-28:
Yes sure thing but my car isn't plugged in right now. I can give you results in 24 hrs :)

alexonbandit, 2023-08-28:
I can confirm the issue. With the mode being set to PV the Tesla starts to charge (using TWC3) immediately even is the is no excessive power from the PV available. 
In addition if I manually change the EVCC mode to OFF the charging continues to as if the command did not make its way to the car.
Stopping it from the Tesla app worked right away.
Tesla SW Version updated to 2023.26.8. Could this update may have an influence?

yipikaye, 2023-08-28:
Same here

`Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 ----
Aug 29 09:37:10 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:10 charge power: 0W
Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 pv power: 5875W
Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 battery soc: 72%
Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 battery power: -5300W
Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 grid power: 0W
Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 giving priority to battery charging at soc: 72%
Aug 29 09:37:10 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:10 site power: 500W
Aug 29 09:37:10 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:10 charge voltages: [0.7 0 0]V
Aug 29 09:37:10 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:10 charge currents: [0.3 0.3 0.2]A
Aug 29 09:37:10 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:10 charger status: A
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 ----
Aug 29 09:37:20 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:20 charge power: 0W
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 pv power: 5672W
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 battery soc: 72%
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 battery power: -5060W
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 grid power: 0W
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 giving priority to battery charging at soc: 72%
Aug 29 09:37:20 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:20 site power: 500W
Aug 29 09:37:20 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:20 charge voltages: [0.7 0 0]V
Aug 29 09:37:20 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:20 charge currents: [0.3 0.3 0.2]A
Aug 29 09:37:20 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:20 charger status: A
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 ----
Aug 29 09:37:30 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:30 charge power: 0W
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 pv power: 5498W
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 battery soc: 72%
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 battery power: -4910W
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 grid power: 0W
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 giving priority to battery charging at soc: 72%
Aug 29 09:37:30 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:30 site power: 500W
Aug 29 09:37:30 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:30 charge voltages: [0.7 0 0]V
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 charge currents: [0.3 0.3 0.2]A
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 charger status: B
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] INFO 2023/08/29 09:37:31 car connected
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 vehicle api refresh
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 pv timer elapse
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 pv timer inactive
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 charger: guard elapse
Aug 29 09:37:31 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:31 guard timer inactive
Aug 29 09:37:33 raspberrypi evcc[5346]: [main  ] ERROR 2023/08/29 09:37:33 vehicle status: asleep
Aug 29 09:37:33 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:33 pv charge current: 0A = 0A + -2.17A (500W @ 1p)
Aug 29 09:37:33 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:33 pv enable timer reset
Aug 29 09:37:33 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:33 pv timer inactive
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 ----
Aug 29 09:37:40 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:40 charge power: 0W
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 pv power: 5466W
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 battery soc: 72%
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 battery power: -4920W
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 grid power: 0W
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 giving priority to battery charging at soc: 72%
Aug 29 09:37:40 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:37:40 site power: 500W
Aug 29 09:37:40 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:40 charge voltages: [244 0 0]V
Aug 29 09:37:40 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:40 detected connected phases: 1p
Aug 29 09:37:40 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:40 charge currents: [5.1 0.3 0.2]A
Aug 29 09:37:40 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:37:40 charger status: B
Aug 29 09:37:41 raspberrypi evcc[5346]: [main  ] DEBUG 2023/08/29 09:37:41 vehicle status: C (Pikachu)
[... cut ...]
Aug 29 09:40:40 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:40:40 vehicle soc: 71%
Aug 29 09:40:41 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:40:41 vehicle soc limit: 80%
Aug 29 09:40:41 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:40:41 vehicle range: 374km
Aug 29 09:40:41 raspberrypi evcc[5346]: [lp-1  ] DEBUG 2023/08/29 09:40:41 pv charge current: 0A = 0A + -2.17A (500W @ 1p)
Aug 29 09:40:50 raspberrypi evcc[5346]: [site  ] DEBUG 2023/08/29 09:40:50 ----
Aug 29 09:40:50 raspberrypi evcc[5346`

andig, 2023-08-29:
What's interesting here is this:

    set charge mode: off

Even that does not stop charging. Could I ask you to set the Tesla car to log trace to see if evcc even attempts to stop it?

I've also hears rumors that the TWC api might have changed- could that potentially mean that Tesla is now controlled by the TWC and ignores the app api in that case?

@GrimmiMeloni oder siehst Du irgendeine Möglichkeit, dass hier das geänderte `syncCharger` etwas bewirken sollte? Ich würde mindestens bei `off` immer noch eindeutigen Effekt erwarten?

yipikaye, 2023-08-29:
Hi @andig, 

I will try and collect more logs. The app does stop the car sometimes so it can work. E.g. when it is charging on solar (as it should) then cloud moves over or you set the % SOC of home battery higher then EVCC does stop charging. But then the car starts again sometime later. 

There was a major update to the TWC firmware and app. The car can still be controlled to change the charge rate but there is something going on with the wall chager. There is also a [hardware refresh coming](https://www.notateslaapp.com/news/1559/tesla-adds-universal-wall-connector-with-magic-dock-type-solution).  There are some people reverse engineering the app now. 

It doesn't appear to be well documented at all yet
https://community.home-assistant.io/t/tesla-wall-connector-gen-3-restful/311670/68

GrimmiMeloni, 2023-08-29:
> When I set EVCC to off, but its already in PV state where it should be in the off state but isn't, then nothing happens and the car continues to charge.

From the top of my head, I think it still matches the analysis.
Very likely both charger and loadpoint think they are disabled.
Given the restart, also the loadpoint chargeCurrent is default at 0.

mode.Off calls setLimit(0), but given the above situation it is NoOp.

To the TWC owners here - is the behavior that charging starts immediately on plugging in the car something new (maybe due to the mentioned firmware update), or was it always like this?

andig, 2023-08-29:
> But then the car starts again sometime later.

If that happens without evcc sending the startup according to detailed log then I'm afraid we're out of luck for time being.

yipikaye, 2023-08-29:
> To the TWC owners here - is the behavior that charging starts immediately on plugging in the car something new (maybe due to the mentioned firmware update), or was it always like this?

Charging always started immediately, but then EVCC stopped it within 1 control loop.

andig, 2023-08-29:
> Charging always started immediately, but then EVCC stopped it within 1 control loop.

Ah, that makes it (a little) clearer. We've refactored the way that evcc forces its will upon the charger. Since we cannot enable/disable the TWC we're just returning the last set state. We do _not_ take into account if the charger is actually charging (if so, it cannot be disabled).

@GrimmiMeloni das können wir im TWC fixen aber genauso auch im Loadpoint (Dein Vorschlag). Bei mir macht es aber noch nicht klick, woher diese Änderung des Verhaltens kommt? Im Modus `Off` müsste doch ein `enable`ter Charger doch wieder gestoppt werden?

GrimmiMeloni, 2023-08-29:
> Im Modus `Off` müsste doch ein `enable`ter Charger doch wieder gestoppt werden?

Korrekt. Aber die TWC Implementierung von Enabled() gibt lediglich einen Boolean zurück, der nur durch den Loadpoint via Enabled() verändert wird. Da der Loadpoint aber gar nicht enabled (hat) gibt der Charger auch bei laufender Ladung ein disabled zurück.
Generell sollten die Charger Implementierungen den physikalischen Ist-Zustand abbilden. Insbesondere bei Chargers mit Eigenleben muß Enabled() auch entsprechend intelligent sein um das zu erkennen.

andig, 2023-08-29:
> Aber die TWC Implementierung von Enabled() gibt lediglich einen Boolean zurück, der nur durch den Loadpoint via Enabled() verändert wird. Da der Loadpoint aber gar nicht enabled (hat) gibt der Charger auch bei laufender Ladung ein disabled zurück.

@GrimmiMeloni genau so- aber warum sollte das erst ab 0.119.0 ein Problem sein?

GrimmiMeloni, 2023-08-29:
> aber warum sollte das erst ab 0.119.0 ein Problem sein?

Weil bis 0.118.11 noch die Erkennung über lp.charging() angesprungen ist, und dann der Charger disabled wurde. (Das ist der case mit "Charger Logic error").

Das würde durch meinen Fix aus #9570 wieder hergestellt.

andig, 2023-08-29:
@yipikaye any chance you could try the nightly version in 20 minutes?

infinus, 2023-08-29:
I tried the nightly and we have good and bad. Good: charger timer starts to stop charging. The bad is charging will not resume when there is excess solar. The off button will stop charging but going back to solar will not resume. Here is the log:

`

[main  ] INFO 2023/08/29 22:56:42 evcc 0.119.2 (b28c7b51)
--
[main  ] INFO 2023/08/29 22:56:42 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/29 22:56:42 starting ui and api at :7070
[main  ] INFO 2023/08/29 22:56:48 evcc 0.119.2 (b28c7b51)
[main  ] INFO 2023/08/29 22:56:48 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/29 22:56:48 starting ui and api at :7070
[db    ] INFO 2023/08/29 22:56:48 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2023/08/29 22:56:49 site config:
[site  ] INFO 2023/08/29 22:56:49   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/08/29 22:56:49     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/29 22:56:49     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/08/29 22:56:49   vehicles:
[site  ] INFO 2023/08/29 22:56:49     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/08/29 22:56:49 loadpoint 1:
[lp-1  ] INFO 2023/08/29 22:56:49   mode:        pv
[lp-1  ] INFO 2023/08/29 22:56:49   charger:     power ✓ energy ✗ currents ✓ phases ✗wakeup ✗
[lp-1  ] INFO 2023/08/29 22:56:49   meters:      charge ✓
[lp-1  ] INFO 2023/08/29 22:56:49     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/08/29 22:56:49 phase timer inactive
[lp-1  ] DEBUG 2023/08/29 22:56:49 pv timer inactive
[lp-1  ] DEBUG 2023/08/29 22:56:49 guard timer inactive
[lp-1  ] INFO 2023/08/29 22:56:49 vehicle updated: unknown -> Infinus
[site  ] WARN 2023/08/29 22:56:49 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/08/29 22:56:49 ----
[lp-1  ] DEBUG 2023/08/29 22:56:49 charge power: 0W
[site  ] DEBUG 2023/08/29 22:56:52 pv power: 9150W
[site  ] DEBUG 2023/08/29 22:56:54 grid power: 368W
[site  ] DEBUG 2023/08/29 22:56:54 site power: 368W
[lp-1  ] DEBUG 2023/08/29 22:56:54 vehicle odometer: 63008km
[lp-1  ] DEBUG 2023/08/29 22:56:54 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/08/29 22:56:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/29 22:56:54 charger status: B
[lp-1  ] INFO 2023/08/29 22:56:54 car connected
[lp-1  ] DEBUG 2023/08/29 22:56:54 pv timer elapse
[lp-1  ] DEBUG 2023/08/29 22:56:54 pv timer inactive
[lp-1  ] DEBUG 2023/08/29 22:56:54 charger: guard elapse
[lp-1  ] DEBUG 2023/08/29 22:56:54 guard timer inactive
[lp-1  ] DEBUG 2023/08/29 22:56:54 vehicle soc: 47%
[lp-1  ] DEBUG 2023/08/29 22:56:54 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/29 22:56:54 vehicle range: 205km
[lp-1  ] DEBUG 2023/08/29 22:56:54 pv charge current: 0A = 0A + -1.6A (368W @ 1p)
[lp-1  ] DEBUG 2023/08/29 22:56:54 pv enable timer reset
[lp-1  ] DEBUG 2023/08/29 22:56:54 pv timer inactive
[site  ] DEBUG 2023/08/29 22:56:59 ----
[lp-1  ] DEBUG 2023/08/29 22:56:59 charge power: 0W
[site  ] DEBUG 2023/08/29 22:57:01 pv power: 9128W
[site  ] DEBUG 2023/08/29 22:57:03 grid power: 380W
[site  ] DEBUG 2023/08/29 22:57:03 site power: 380W
[lp-1  ] DEBUG 2023/08/29 22:57:03 vehicle odometer: 63008km
[lp-1  ] DEBUG 2023/08/29 22:57:03 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/08/29 22:57:03 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/29 22:57:03 charger status: B
[lp-1  ] DEBUG 2023/08/29 22:57:03 pv charge current: 0A = 0A + -1.65A (380W @ 1p)
[site  ] DEBUG 2023/08/29 22:57:09 ----
[lp-1  ] DEBUG 2023/08/29 22:57:11 charge power: 0W
[site  ] DEBUG 2023/08/29 22:57:13 pv power: 9141W
[site  ] DEBUG 2023/08/29 22:57:15 grid power: 355W
[site  ] DEBUG 2023/08/29 22:57:15 site power: 355W
[lp-1  ] DEBUG 2023/08/29 22:57:16 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/08/29 22:57:16 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/29 22:57:16 charger status: B
[lp-1  ] DEBUG 2023/08/29 22:57:16 pv charge current: 0A = 0A + -1.54A (355W @ 1p)
[site  ] DEBUG 2023/08/29 22:57:19 ----
[lp-1  ] DEBUG 2023/08/29 22:57:19 charge power: 0W
[site  ] DEBUG 2023/08/29 22:57:22 pv power: 9151W
[... 949 lines cut ...]
[lp-1  ] DEBUG 2023/08/29 23:08:04 charger status: C
[lp-1  ] DEBUG 2023/08/29 23:08:04 vehicle soc: 47%
[lp-1  ] DEBUG 2023/08/29 23:08:04 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/29 23:08:04 vehicle range: 205km
[lp-1  ] WARN 2023/08/29 23:08:04 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/08/29 23:08:04 pv charge current: 20.9A = 3.5A + 17.4A (-4013W @ 1p)
[lp-1  ] DEBUG 2023/08/29 23:08:04 max charge current: 20A
[site  ] DEBUG 2023/08/29 23:08:09 ----
[lp-1  ] DEBUG 2023/08/29 23:08:09 charge power: 336W
[site  ] DEBUG 2023/08/29 23:08:12 pv power: 9117W
[site  ] DEBUG 2023/08/29 23:08:15 grid power: -4078W
[site  ] DEBUG 2023/08/29 23:08:15 site power: -4078W
[lp-1  ] DEBUG 2023/08/29 23:08:15 charge voltages: [240 0 0]V
[lp-1  ] DEBUG 2023/08/29 23:08:15 detected connected phases: 1p
[lp-1  ] DEBUG 2023/08/29 23:08:15 charge currents: [1.3 0 0]A
[lp-1  ] DEBUG 2023/08/29 23:08:15 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/29 23:08:15 charger status: C
[lp-1  ] DEBUG 2023/08/29 23:08:15 vehicle soc: 47%
[lp-1  ] DEBUG 2023/08/29 23:08:15 vehicle soc limit: 91%
[lp-1  ] DEBUG 2023/08/29 23:08:15 vehicle range: 205km
[lp-1  ] WARN 2023/08/29 23:08:15 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/08/29 23:08:15 pv charge current: 21A = 3.3A + 17.7A (-4078W @ 1p)
[lp-1  ] DEBUG 2023/08/29 23:08:15 max charge current: 21A

`

GrimmiMeloni, 2023-08-30:
@infinus thanks for sharing the log. I have multiple questions.

1. What happens at 23:00:34 that makes the charger start? Seems like the vehicle was connected from the start, so the EV connection is unlikely the trigger here. This is interesting, because also after you manually stopped the charger later, it also starts again without a clear reason. Is another controlling entity apart from evcc talking to the charger?
2. PV Power is at ~9kW - still your setup seems to be drawing ~350W grid power even before the charge. Once the charge kicks in, it gets worse. Assuming your timestamps are local time, I am wondering if the PV power can be correct.
3. @andig Charger logic error does not go away after one cycle. Go lang special question: Does the defer I introduced keep a local copy of the boolean var, so that the change of enable to true is not effective and the defer always sets it to false no matter what? It would explain why the charger logic error does not get corrected.

infinus, 2023-08-30:
@GrimmiMeloni 
1. I experimented with multiple scenarios to see how it would handle them. I unplugged and plugged back in physically and I also tried to imitate charging via the Tesla app to see if it would trigger the timer to stop charging.  What you are seeing there is probably me starting charging via the app. 

2.  It is correct. I live in a hot climate and have two air conditioning systems that pull 7kw plus pool pump was running at 2kw and minor appliances.  The time stamp itself might be incorrect because I don’t think I set the TZ in docker as it was my test machine.

yipikaye, 2023-08-30:
@GrimmiMeloni @andig Sorry I didn't get a chance today and cannot test tomorrow either as I'm away. If im following the thread though you have what you need and all seems OK thanks to logs kindly provided by @infinus ?

GrimmiMeloni, 2023-08-30:
Yes, for now. Please make sure to still capture a log when you test though.

andig, 2023-08-30:
Gents, I've hidden the excess logs. Lets please sort this:

>  I reran a test on the nightly build. It looks like it is working.

@infinus so finally confirmed ok? Thank you!

> Charger logic error does not go away after one cycle. 

Seems we're in PV mode and the PV timer is still running. In this case, we should expire the PV timer, too to make this an immediate action?

/cc @MarkusGH seems this is also the case for https://github.com/evcc-io/evcc/issues/9586#issuecomment-1699561620 which is why you're getting the error messages time and again.

closed 2023-08-30 by andig

MarkusGH, 2023-08-30:
Das wird nicht helfen, denn wenn der Loadpoint enabled ist (und das ist er ab dem Verlassen von syncCharger) dann wird er nicht mehr enabled.
Gegenvorschlag: https://github.com/evcc-io/evcc/issues/9586

MarkusGH, 2023-08-30:
@andig: Alternativ könnte man vielleicht auch Deine Lösung nehmen, allerdings muss dann das Konstrukt:

	defer func() {
		lp.enabled = enabled
		lp.publish("enabled", lp.enabled)
	}()

weiter nach unten, denn wie geschrieben, wenn lp.enabled == true ist wird das Laden nicht gestartet, dann bleibt der Loadpoint weiterhin disabled und lädt.

andig, 2023-08-30:
@MarkusGH hier gehts doch um den disable timer?!

MarkusGH, 2023-08-30:
Dir schon - aber mein Problem ist ein anderes.
Solange im MinPV Modus genug PV Strom da ist, spielt der disable timer keine Rolle.
Dann kommt die Meldung "charger logic error: disabled but charging." wieder und wieder.
Man braucht deswegen beides.
Mein Vorschlag in https://github.com/evcc-io/evcc/pull/9629

andig, 2023-08-30:
Verstehe. Das ist hässlich. Bitte neues Issue!

yipikaye, 2023-09-01:
I have deployed 0.119.3 and confirm this is working perfectly, some patience is needed for the timer but it does the correct operation :)

andig, 2023-09-01:
@GrimmiMeloni with the latest update we should not have the timer issue anymore. Or do you think we need to expire the pv timer in all places?

@yipikaye could you once more share a log? We should not wait for timer here.

yipikaye, 2023-09-01:
Hi @andig 

For a reason I don't know the car started charging itself at 16:20. Rather than let the timer finish I did stop it myself but not before letting it run for a minute or so. 

```

Sep  1 16:20:09 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:09 ----
Sep  1 16:20:09 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:09 charge power: 0W
Sep  1 16:20:09 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:09 pv power: 390W
Sep  1 16:20:09 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:09 battery soc: 90%
Sep  1 16:20:09 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:09 battery power: 520W
Sep  1 16:20:09 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:09 grid power: 0W
Sep  1 16:20:09 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:09 site power: 1020W
Sep  1 16:20:09 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:09 charge voltages: [0.7 0 0]V
Sep  1 16:20:09 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:09 charge currents: [0.3 0.3 0.2]A
Sep  1 16:20:09 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:09 charger status: B
Sep  1 16:20:09 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:09 pv charge current: 0A = 0A + -4.43A (1020W @ 1p)
Sep  1 16:20:19 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:19 ----
Sep  1 16:20:19 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:19 charge power: 0W
Sep  1 16:20:19 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:19 pv power: 383W
Sep  1 16:20:19 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:19 battery soc: 90%
Sep  1 16:20:19 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:19 battery power: 550W
Sep  1 16:20:19 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:19 grid power: 0W
Sep  1 16:20:19 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:19 site power: 1050W
Sep  1 16:20:19 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:19 charge voltages: [0.7 0 0]V
Sep  1 16:20:19 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:19 charge currents: [0.3 0.3 0.2]A
Sep  1 16:20:19 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:19 charger status: B
Sep  1 16:20:19 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:19 pv charge current: 0A = 0A + -4.57A (1050W @ 1p)
Sep  1 16:20:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:29 ----
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:29 charge power: 98W
Sep  1 16:20:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:29 pv power: 378W
Sep  1 16:20:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:29 battery soc: 90%
Sep  1 16:20:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:29 battery power: 560W
Sep  1 16:20:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:29 grid power: 0W
Sep  1 16:20:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:29 site power: 1060W
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:29 charge voltages: [245 0 0]V
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:29 detected connected phases: 1p
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:29 charge currents: [0.4 0.3 0.2]A
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:29 charger status: C
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] INFO 2023/09/01 16:20:29 start charging ->
Sep  1 16:20:29 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:29 wake-up timer: stop
Sep  1 16:20:29 raspberrypi evcc[5348]: [pushover] DEBUG 2023/09/01 16:20:29 sending to ***
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 soc estimated: 77.07% (vehicle: 77.00%)
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 vehicle soc: 77%
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 vehicle soc limit: 100%
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 vehicle range: 403km
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] WARN 2023/09/01 16:20:30 charger out of sync: expected disabled, got enabled
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 charger: guard elapse
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 guard timer inactive
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 pv charge current: 0A = 2.4A + -4.61A (1060W @ 1p)
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 site power 1060W >= 0W disable threshold
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 pv disable timer start: 3m0s
Sep  1 16:20:30 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:30 pv disable in 3m0s
Sep  1 16:20:39 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:39 ----
Sep  1 16:20:39 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:20:39 charge power: 1270W
Sep  1 16:20:39 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:39 pv power: 379W
Sep  1 16:20:39 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:39 battery soc: 90%
Sep  1 16:20:39 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:39 battery power: 580W
Sep  1 16:20:39 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:39 grid power: 0W
Sep  1 16:20:39 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:20:39 site power: 1080W
[... cut ...]
Sep  1 16:22:21 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:22:21 pv charge current: 0A = 0A + -10.3A (2360W @ 1p)
Sep  1 16:22:21 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:22:21 site power 2360W >= 0W disable threshold
Sep  1 16:22:21 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:22:21 pv disable timer start: 3m0s
Sep  1 16:22:21 raspberrypi evcc[5348]: [lp-1  ] DEBUG 2023/09/01 16:22:21 pv disable in 3m0s
Sep  1 16:22:29 raspberrypi evcc[5348]: [site  ] DEBUG 2023/09/01 16:22:29 ----

```

andig, 2023-09-03:
> For a reason I don't know the car started charging itself at 16:20.

No idea. If we did start it we should see that in the log. To be sure set `tesla: trace` as that's done via vehicle.

Re timer: which version are you running? The disable timer should already be expired in 119.3, but please confirm your version and commit.