evcc-io/evcc #9542: Emphase envoy without https

pdonovan, 2023-08-26:
### Describe the bug

My Enphase Envoy-S does not listen on port 443, only port 80. The recent commit 866da9f that changed the enphase.yaml template to use https means that it no longer works.

What was the reason for switching to https? It's local traffic on a local network, there is nothing secret there.

### Steps to reproduce

1. Add an enphase grid or pv meter.
2. Note that evcc can't connect

My Envoy is running firmware D5.0.62 (5e57a9)

### Configuration details

```yaml
meters:
- type: template
  template: enphase 
  usage: grid  
  host: 192.168.0.128  
  name: grid1
- type: template
  template: enphase 
  usage: pv  
  host: 192.168.0.128  
  name: pv2
```

### Log details

```text
ERROR 2023/08/26 09:08:39 pv 1 power: Get "https://192.168.0.128/production.json": dial tcp 192.168.0.128:443: connect: connection refused
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.1

andig, 2023-08-26:
@premultiply wir dürfen solche Änderungen nicht als breaking change machen.

premultiply, 2023-08-26:
Aus dem letzten Tracelog ging hervor dass von Enphase ohnehin HTTP auf HTTPS umgeleitet wird. Daher bin ich jetzt einigermaßen verwundert.
Daher hatte ich das gleich mit vereinfacht.

Gut, aber das kann man ja schnell wieder ändern...

premultiply, 2023-08-26:
Fixed in https://github.com/evcc-io/evcc/commit/8adc1f480eddd5d3026ab348b017298f036c860d

closed 2023-08-26 by premultiply

lilstevie, 2024-01-03:
> What was the reason for switching to https? It's local traffic on a local network, there is nothing secret there.
Enphase Envoy Firmwares beyond 7.x do not listen to traffic on http and are https only.

This means https://github.com/evcc-io/evcc/commit/8adc1f480eddd5d3026ab348b017298f036c860d breaks support for latest firmwares