evcc-io/evcc #9543: missing phase scale down, blocks 1p PV start

GrimmiMeloni, 2023-08-26:
### Describe the bug

When charger is in 3p mode prior to charge start, charging does not start in PV mode until enough surplus for 3p is available.
Expectation is for evcc to down scale to 1p once 1,4kW are available, and start charging.

Workaround:
Forcing the charger to 1p in this situation gets evcc to start charging.
Switching to automatic phases afterwards also is fine.

### Steps to reproduce

1. Missing surplus situation (i.e. Sunrise)
2. Force charger to 3p via UI.
3. Set charger back to automatic.
4. Wait for surplus to reach 1,4kW
5. evcc does not start charging

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  easee: trace
#  enyaq: trace

# unique installation id
plant: xxx
sponsortoken: xxx

interval: 15s # control cycle interval

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.3.38  
  password: xxx  
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.3.38  
  password: xxx  
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.3.38  
  password: xxx  
  name: battery3

site:
  title: Zu Hause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100 # im PV modus wird 100W für Laden des Hausakkus übrig gelassen, damit Regelkreise Tesla/EVCC nicht konkurrieren
  prioritySoc: 70 # Hausbatterie bekommt bis zum SoC Priorität beim laden
  bufferSoc: 80 # bis zu diesem SoC wird Batterie zur kurzfristigen Überbrückung bei PV Ladung genutzt

chargers:
  - name: Easee
    type: template
    template: easee
    user: xxx
    password: xxx
    charger: xxx
    authorize: true

loadpoints:
  - title: Carport # display name for UI
    charger: Easee
    mode: off
    resetOnDisconnect: true
    maxcurrent: 32

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.3796 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.082 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: "xxx"

vehicles:
  - name: Enyaq
    type: template
    template: enyaq
    title: Enyaq
    user: xxx
    password: xxx
    vin: xxx 
    capacity: 77 
    mode: pv 

# mqtt message broker
mqtt:
  broker: ds218.fritz.box:1883
  topic: evcc
```

### Log details

```text
too large, will attach
```

### What type of operating system are you running?

Linux

### Version

0.119.1

GrimmiMeloni, 2023-08-26:
[evcc.log](https://github.com/evcc-io/evcc/files/12444832/evcc.log)

GrimmiMeloni, 2023-08-26:
current "suspect" is sitePower > 0 in

https://github.com/evcc-io/evcc/blob/8adc1f480eddd5d3026ab348b017298f036c860d/core/loadpoint.go#L1028

GrimmiMeloni, 2023-08-27:
I considered this some more - right now I don't see any reason why grid import is a requirement for enabling the scale down. The remaining logic of targetCurrent being below minA while we have more than a single phase active seems sufficient.

andig, 2023-08-27:
Ahh, good catch!

GrimmiMeloni, 2023-08-27:
> Could you test

Will do and post an update.

GrimmiMeloni, 2023-08-27:
I can confirm your suggested change works as well, @andig

andig, 2023-08-28:
@GrimmiMeloni ich tendiere dazu, https://github.com/evcc-io/evcc/pull/9462 nochmal zu reverten und einschl. Fix dann nach Release erneut ins Nightly zu nehmen. Scheint mir noch nicht ganz ausgereift. Was meinst Du?

andig, 2023-08-28:
Reverted per https://github.com/evcc-io/evcc/commit/1faa666265675d75e14738182dafd83867306286.

GrimmiMeloni, 2023-08-28:
Ja klingt gut. Zumindest laufen dann nicht noch mehr Leute in das Problem mit dem fehlenden Ladestart.

andig, 2023-08-28:
Ich lasse hier mal offen. Wenn sich das Release beruhigt hat probieren wir das nochmal im Nightly.

RodeoRoland, 2023-09-29:
I have a similar/related issue with my manual phase switching charger. When connecting the car, evcc assumes always 3 phases, although only 1 phase is connected (and was used during the last charging). As soon as the charge process starts (pv triggered, manual started, or by a timer) evcc realizes that only 1p is connected.

For example, when having a timer set for charging to 8am, it will simply start too late with charging (as 3p is expected, but 1p is connected only).

Idea: would it make sense to simply save (non-persistent) the active number of phases, instead of assuming 3p in every situation?

andig, 2023-09-29:
Bitte neues Thema!

closed 2023-10-08 by andig