evcc-io/evcc #9548: Add Nordpool or Entso-e tariff or MQTT Tariff

djfanatix, 2023-08-26:
**Is your feature request related to a problem? Please describe.**
Is it possible to add Nordpool or Entso-e tariff for Belgium? (or other EU countries)

**Describe the solution you'd like**
Or use the Nordpool scraper as mentioned before in other issues.
Or use the Entso-e api, by letting the user provide an API key and country to retrieve data.
Home Assistant integration:
https://github.com/JaccoR/hass-entso-e/tree/main

API documentation: https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html

Python: https://github.com/EnergieID/entsoe-py

Or if too complicated, provide a MQTT listener, to retrieve tariff information.
For example send from Home Assistant or other every hour the feed-in and the feed-out tariff

Thanks!

andig, 2023-08-27:
Happy to take a PR. As discussed before this is currently not a priority.

closed 2023-08-27 by andig

andig, 2023-09-03:
See https://github.com/evcc-io/evcc/issues/5291#issuecomment-1704070656

/cc @duckfullstop if you want to jump in.

duckfullstop, 2023-09-03:
I'd say let's keep discussion of Entsoe to #5291, and open a new issue for a potential future MQTT tariff provider? The latter would be a little more challenging / potentially out of band, as we wouldn't be able to (easily) do look-ahead forecasting - MQTT is a messaging protocol after all.

duckfullstop, 2023-09-08:
@djfanatix Please see #5291 where your feedback would be greatly appreciated 🙏