evcc-io/evcc #9550: Target charge not reached at set time

bolisaf, 2023-08-27:
### Describe the bug

Since upgrading to 0.119.x the target charge that I set does not get reached in time. 
I notice that it only uses 3-phase charging when emergency charging, but not when getting charging as part of a plan. that's fine, but not if the target charge doesn't get reached at the set time. 
just this morning I wanted the car to be charge from 50% (emergency minimum charge) to 70% (the charge I needed today) at 7am. it started charging at 5, but wasn't finished at 7. that's a real problem! to me it looks like it just starts charging too late. The maxcurrent of my charger is 32A and 3phase, but my VW has a max of 16A. it should be able to calculate when to start charging based on the fact that it knows it's max 16A 3phase, but it seems like that part is failing.

### Steps to reproduce

1.set the target charge plan to anything more than 20% of the battery
2.wake up next morning and see that it's not charged in time

### Configuration details

```yaml
sponsortoken: xxxxxxxxxx
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:

- name: SMA
  type: template
  template: sma-inverter
  usage: pv
  host: 10.10.10.47 # IP-Adresse oder Hostname
  password: xxxxxxxx # Passwort für Benutzergruppe Benutzer (Optional) 

- name: slimmelezer
  type: custom
[... 398 lines cut ...]
[lp-1  ] DEBUG 2023/08/27 08:32:36 charge total import: 584.096kWh
[lp-1  ] DEBUG 2023/08/27 08:32:37 charger status: B
[lp-1  ] DEBUG 2023/08/27 08:32:37 planned 1h14m22s until 2023-08-27 10:00:00 +0200 CEST at 11040W: total plan duration: 1h14m22s, avg cost: 0.207
[lp-1  ] DEBUG 2023/08/27 08:32:37 pv charge current: 4.62A = 0A + 4.62A (-1063W @ 1p)
[site  ] DEBUG 2023/08/27 08:32:46 ----
[lp-1  ] DEBUG 2023/08/27 08:32:46 charge power: 0W
[site  ] DEBUG 2023/08/27 08:32:46 pv power: 1302W
[site  ] DEBUG 2023/08/27 08:32:46 grid power: -1085W
[site  ] DEBUG 2023/08/27 08:32:46 grid currents: [1.51 1.62 1.57]A
[site  ] DEBUG 2023/08/27 08:32:46 site power: -1085W
[lp-1  ] DEBUG 2023/08/27 08:32:46 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/27 08:32:46 charge total import: 584.096kWh
[lp-1  ] DEBUG 2023/08/27 08:32:47 charger status: B
[lp-1  ] DEBUG 2023/08/27 08:32:47 planned 1h14m22s until 2023-08-27 10:00:00 +0200 CEST at 11040W: total plan duration: 1h14m22s, avg cost: 0.207
[lp-1  ] DEBUG 2023/08/27 08:32:47 pv charge current: 4.72A = 0A + 4.72A (-1085W @ 1p)
^C[lp-1  ] DEBUG 2023/08/27 08:32:51 charge total import: 584.096kWh
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.119.1

bolisaf, 2023-08-29:
Hey,That doesn’t seem to have fixed it I’m afraid. But when I charge use the ‘fast’ setting 3 phases are used again. This did not work on 0.119.1, but is fixed in .2Thanks!L

Could you please try 0.119.2 in a few minutes?

cschlipf, 2023-08-30:
I experienced a similar issue just a few days ago. Could it be that the charging mode is no longer set to 'Fast' in case of planned target SoCs and that it just keeps the current charging mode? In my case it was still set to PV even though the target charge was planned in the early morning when the sun still was down.

bolisaf, 2023-08-30:
But if it has the whole night to reach the charging goal, why should it use ‘fast’? I try to avoid this mode as much as possible for the capacity tariff. I would rather it charges the whole night at the lowest possible power.Thanks!L

andig, 2023-08-30:
That‘s just how it works.

andig, 2023-08-30:
@bolisaf are you able to compile evcc? I've looked hard but cannot find a problem. 

Could you kindly change the following in loadpoint.go:

```go
// scalePhasesIfAvailable scales if api.PhaseSwitcher is available
func (lp *Loadpoint) scalePhasesIfAvailable(phases int) error {
	want := phases
	if lp.ConfiguredPhases != 0 {
		phases = lp.ConfiguredPhases
	}

	if _, ok := lp.charger.(api.PhaseSwitcher); ok {
		lp.log.DEBUG.Printf("!! scalePhasesIfAvailable: %dp -> %dp", want, phases)
		return lp.scalePhases(phases)
	}

	return nil
}

// scalePhases adjusts the number of active phases and returns the appropriate charging current.
// Returns api.ErrNotAvailable if api.PhaseSwitcher is not available.
func (lp *Loadpoint) scalePhases(phases int) error {
	cp, ok := lp.charger.(api.PhaseSwitcher)
	if !ok {
		panic("charger does not implement api.PhaseSwitcher")
	}

	lp.log.DEBUG.Printf("!! scalePhases: GetPhases %dp <> phases %dp", lp.GetPhases(), phases)
	if lp.GetPhases() != phases {
		// switch phases
		if err := cp.Phases1p3p(phases); err != nil {
			return fmt.Errorf("switch phases: %w", err)
		}

		// prevent premature measurement of active phases
		lp.phasesSwitched = lp.clock.Now()

		// update setting and reset timer
		lp.setPhases(phases)
	}

	return nil
}
```

If not I'll add this to the nightly build.

andig, 2023-08-30:
Please try new nightly version tomorrow plus change above- I've also improved planner logging.

andig, 2023-08-31:
Please retry 0.119.3.

bolisaf, 2023-09-01:
Hey,It didn’t work with 0.119.3 I’m afraid. Would it help you if I ran it in debug the whole night and provided you the logs? It would be a pretty lengthy file thoughThanks!L

Closed #9550 as completed.

andig, 2023-09-01:
Yes please, new improved log would be helpful.

bolisaf, 2023-09-02:
here you go. hope it helps you to find the bug!

[debug.txt](https://github.com/evcc-io/evcc/files/12503302/debug.txt)

andig, 2023-09-03:
This is the plan:

```
[lp-1  ] DEBUG 2023/09/02 05:08:17 plan: charge 1h51m36s starting at 2023-09-02 05:08:24 +0200 CEST until 2023-09-02 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)
[lp-1  ] DEBUG 2023/09/02 05:08:17 pv charge current: 0A = 0A + -1.22A (281W @ 1p)
```

Then charging starts:

```
[lp-1  ] DEBUG 2023/09/02 05:08:26 charge power: 0W
[site  ] DEBUG 2023/09/02 05:08:27 pv power: 0W
[site  ] DEBUG 2023/09/02 05:08:27 grid power: 282W
[site  ] DEBUG 2023/09/02 05:08:27 grid currents: [0.76 0.72 0.58]A
[site  ] DEBUG 2023/09/02 05:08:27 site power: 282W
[lp-1  ] DEBUG 2023/09/02 05:08:27 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/02 05:08:27 charge total import: 676.534kWh
[lp-1  ] DEBUG 2023/09/02 05:08:27 charger status: B
[lp-1  ] DEBUG 2023/09/02 05:08:27 plan: charge 1h51m32s (required: 1h51m36s) starting at 2023-09-02 05:08:28 +0200 CEST until 2023-09-02 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)
[lp-1  ] DEBUG 2023/09/02 05:08:28 max charge current: 16A
[lp-1  ] DEBUG 2023/09/02 05:08:28 charger enable
[lp-1  ] DEBUG 2023/09/02 05:08:28 wake-up timer: start
[lp-1  ] DEBUG 2023/09/02 05:08:28 pv timer elapse
[lp-1  ] DEBUG 2023/09/02 05:08:28 pv timer inactive
[lp-1  ] DEBUG 2023/09/02 05:08:28 charger: guard elapse
[lp-1  ] DEBUG 2023/09/02 05:08:28 guard timer inactive
```

and

```
[lp-1  ] DEBUG 2023/09/02 05:08:46 charge power: 8702W
[site  ] DEBUG 2023/09/02 05:08:47 pv power: 0W
[site  ] DEBUG 2023/09/02 05:08:47 grid power: 9903W
[site  ] DEBUG 2023/09/02 05:08:47 grid currents: [14.3 14.3 14.4]A
[site  ] DEBUG 2023/09/02 05:08:47 site power: 9903W
[lp-1  ] DEBUG 2023/09/02 05:08:47 charge currents: [15.5 15.5 15.5]A
[lp-1  ] DEBUG 2023/09/02 05:08:47 charge total import: 676.558kWh
[lp-1  ] DEBUG 2023/09/02 05:08:47 charger status: C
[lp-1  ] DEBUG 2023/09/02 05:08:47 soc estimated: 51.02% (vehicle: 51.00%)
[lp-1  ] DEBUG 2023/09/02 05:08:47 vehicle soc: 51%
[lp-1  ] DEBUG 2023/09/02 05:08:47 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/02 05:08:47 vehicle range: 200km
[lp-1  ] DEBUG 2023/09/02 05:08:47 plan: charge 1h51m12s (required: 1h51m30s) starting at 2023-09-02 05:08:48 +0200 CEST until 2023-09-02 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)
```

And at 7:00

```
[lp-1  ] DEBUG 2023/09/02 07:00:06 charge power: 10775W
[site  ] DEBUG 2023/09/02 07:00:07 pv power: 0W
[site  ] DEBUG 2023/09/02 07:00:07 grid power: 11108W
[site  ] DEBUG 2023/09/02 07:00:07 grid currents: [16 15.9 16]A
[site  ] DEBUG 2023/09/02 07:00:07 site power: 11108W
[lp-1  ] DEBUG 2023/09/02 07:00:07 charge currents: [15.5 15.5 15.6]A
[lp-1  ] DEBUG 2023/09/02 07:00:07 detected active phases: 3p
[lp-1  ] DEBUG 2023/09/02 07:00:07 charge total import: 696.587kWh
[lp-1  ] DEBUG 2023/09/02 07:00:07 charger status: C
[lp-1  ] DEBUG 2023/09/02 07:00:07 soc estimated: 74.10% (vehicle: 73.00%)
[lp-1  ] DEBUG 2023/09/02 07:00:07 vehicle soc: 74%
[lp-1  ] DEBUG 2023/09/02 07:00:07 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/02 07:00:07 vehicle range: 288km
[lp-1  ] DEBUG 2023/09/02 07:00:08 plan: charge 0s (required: 4m14s) starting at 0001-01-01 01:00:00 +0100 CET until 2023-09-02 07:00:00 +0200 CEST (power: 11040W, avg cost: NaN)
[lp-1  ] DEBUG 2023/09/02 07:00:08 plan: continuing after target time
```

Everything is just perfect- or what is the bug that you're seeing?

bolisaf, 2023-09-03:
It looks like it did charge it in time this time around. The others times it just wasn’t charged at the target time. Let me do that again tomorrow night.

bolisaf, 2023-09-08:
Hey @andig i ran another debug this night. i asked evcc to charge the car to 70% by 7AM, but in the morning it was only charged to about 56%

at line 75 the car soc is 51%
`[lp-1  ] DEBUG 2023/09/07 23:18:25 vehicle soc: 51%`

at line 134 i set the plan:
`[lp-1  ] DEBUG 2023/09/07 23:18:52 plan: charge 1h28m21s starting at 2023-09-08 05:31:39 +0200 CEST until 
2023-09-08 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)`

at line 24756 the charging starts at the planned time:
```
[lp-1  ] DEBUG 2023/09/08 05:31:41 charge power: 0W
[site  ] DEBUG 2023/09/08 05:31:41 pv power: 0W
[site  ] DEBUG 2023/09/08 05:31:41 grid power: 293W
[site  ] DEBUG 2023/09/08 05:31:41 grid currents: [0.76 0.74 0.66]A
[site  ] DEBUG 2023/09/08 05:31:41 site power: 293W
[lp-1  ] DEBUG 2023/09/08 05:31:41 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/08 05:31:42 charge total import: 784.906kWh
[lp-1  ] DEBUG 2023/09/08 05:31:42 charger status: B
[lp-1  ] DEBUG 2023/09/08 05:31:42 plan: charge 1h28m18s (required: 1h28m19s) starting at 2023-09-08 05:31:42 +0200 CEST until 2023-09-08 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)
[lp-1  ] DEBUG 2023/09/08 05:31:42 max charge current: 16A
[lp-1  ] DEBUG 2023/09/08 05:31:42 charger enable
[lp-1  ] DEBUG 2023/09/08 05:31:42 wake-up timer: start
[lp-1  ] DEBUG 2023/09/08 05:31:42 pv timer elapse
[lp-1  ] DEBUG 2023/09/08 05:31:42 pv timer inactive
[lp-1  ] DEBUG 2023/09/08 05:31:42 charger: guard elapse
[lp-1  ] DEBUG 2023/09/08 05:31:42 guard timer inactive
[lp-1  ] DEBUG 2023/09/08 05:31:42 wake-up vehicle
[site  ] DEBUG 2023/09/08 05:31:51 ----
[lp-1  ] DEBUG 2023/09/08 05:31:51 charge power: 1839W
[site  ] DEBUG 2023/09/08 05:31:51 pv power: 0W
[site  ] DEBUG 2023/09/08 05:31:51 grid power: 2143W
[site  ] DEBUG 2023/09/08 05:31:51 grid currents: [8.42 0.75 0.67]A
[site  ] DEBUG 2023/09/08 05:31:51 site power: 2143W
[lp-1  ] DEBUG 2023/09/08 05:31:51 charge currents: [7.94 0 0]A
[lp-1  ] DEBUG 2023/09/08 05:31:51 charge total import: 784.909kWh
[lp-1  ] DEBUG 2023/09/08 05:31:52 charger status: C
[lp-1  ] DEBUG 2023/09/08 05:31:52 charge start energy: 785kWh
[lp-1  ] INFO 2023/09/08 05:31:52 start charging ->
[lp-1  ] DEBUG 2023/09/08 05:31:52 wake-up timer: stop
[lp-1  ] DEBUG 2023/09/08 05:31:53 soc estimated: 51.01% (vehicle: 51.00%)
[lp-1  ] DEBUG 2023/09/08 05:31:53 vehicle soc: 51%
[lp-1  ] DEBUG 2023/09/08 05:31:53 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/08 05:31:53 vehicle range: 205km
[lp-1  ] DEBUG 2023/09/08 05:31:53 plan: charge 1h28m7s (required: 1h28m19s) starting at 2023-09-08 05:31:53 +0200 CEST until 2023-09-08 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)

at line 32748 at 7am the charging is only 56% instead of the 70% i asked for
[lp-1  ] DEBUG 2023/09/08 07:00:01 charge power: 3581W
[site  ] DEBUG 2023/09/08 07:00:01 pv power: 0W
[site  ] DEBUG 2023/09/08 07:00:01 grid power: 3984W
[site  ] DEBUG 2023/09/08 07:00:01 grid currents: [16.2 0.93 0.68]A
[site  ] DEBUG 2023/09/08 07:00:01 site power: 3984W
[lp-1  ] DEBUG 2023/09/08 07:00:01 charge currents: [15.7 0 0]A
[lp-1  ] DEBUG 2023/09/08 07:00:01 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/08 07:00:02 charge total import: 790.152kWh
[lp-1  ] DEBUG 2023/09/08 07:00:02 charger status: C
[lp-1  ] DEBUG 2023/09/08 07:00:02 soc estimated: 55.86% (vehicle: 55.00%)
[lp-1  ] DEBUG 2023/09/08 07:00:02 vehicle soc: 56%
[lp-1  ] DEBUG 2023/09/08 07:00:02 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/08 07:00:02 vehicle range: 222km
[lp-1  ] DEBUG 2023/09/08 07:00:02 plan: charge 0s (required: 3h17m15s) starting at 0001-01-01 01:00:00 +0100 CET until 2023-09-08 07:00:00 +0200 CEST (power: 3680W, avg cost: NaN)
[lp-1  ] DEBUG 2023/09/08 07:00:02 plan: continuing after target time
```

all the time it says charge x amount of time but the requried time is higher than the planned time, for example: 
`plan: charge 1m48s (required: 3h19m3s)`

here's the full file:
[debug.txt](https://github.com/evcc-io/evcc/files/12556324/debug.txt)

andig, 2023-09-08:
It's charging 1p only:

    charge currents: [15.6 0 0]A

We'll need to find out why.

bolisaf, 2023-09-08:
> It's charging 1p only:
> 
> ```
> charge currents: [15.6 0 0]A
> ```
> 
> We'll need to find out why.

isn't it supposed to use 1p, so the power is kept to a minimum and spread throughout the night like you said earlier in this thread?

cschlipf, 2023-09-08:
> isn't it supposed to use 1p, so the power is kept to a minimum and spread throughout the night like you said earlier in this thread?

This does not make sense... While charging your car has a constant self-consumption. So if you charge from the Grid anyway you should charge as fast as possible. Except you want to limit charging from your home battery.

bolisaf, 2023-09-08:
Here in BE we have a capacity tariff, we get punished for drawing too much power from the grid. So it makes a lot of sense to spread the charging as much as possible. Also the car has a minimum charging current, which is set taking the self-consumption of the car into account. As long as the power is equal or larger than that minimum, you’re fine.Thanks!L

cschlipf, 2023-09-08:
Thanks @bolisaf - didn't know that something like this exists. Learned something new.

andig, 2023-09-09:
@bolisaf we're still missing the switch to 3p. You're charging at 1p and we're failing (or not executing) to switch. Seems you did not add the additional logging requested?

andig, 2023-09-09:
I've added the additional logging to the nightly release as of tomorrow- it would be great if you could provide another logfile with that version.

bolisaf, 2023-09-09:
I just did a —log debug with the latest version..Anyhow, I’ll run it again with the nightly.But again, I’d really prefer it didn’t switch to 3p but just started earlier, for reasons I mentioned earlier.Thanks!L

andig, 2023-09-09:
> But again, I’d really prefer it didn’t switch to 3p but just started earlier, for reasons I mentioned earlier.

Fair, but that is a completely different topic.

For sake of completeness: is it possible that you've set the charger to 1p intentionally (ui settings)? I'd need to double-check, but that might explain why it didn't switch to 3p but the planner still used 3p for calculation. I'll need to verify that.

Anyway please share the new log.

bolisaf, 2023-09-09:
Here’s a screen from my UI:It switches to 3p fine when I select ‘fast’ charging. I can do that as well as part of the debug.Do I create an enhancement request for the 1p low power charging?Thanks!L

bolisaf, 2023-09-11:
> I've added the additional logging to the nightly release as of tomorrow- it would be great if you could provide another logfile with that version.

hey andig, sory for the stupid question, where can i download the nightly build? is it compiled as exe? can i just run it as i did before?

andig, 2023-09-15:
You cannot for HA but latest release should contain the logging.

> Here’s a screen from my UI

No screenshot.

>  I'll need to verify that.

Lets reopen when we have a new log.

bolisaf, 2023-09-16:
heuy @andig here's the new log
[debug.txt](https://github.com/evcc-io/evcc/files/12640442/debug.txt)

take a look at line 21402, it says 'active phases detected: 1p' but doesn't change to 3p

`[site  ] DEBUG 2023/09/16 05:32:47 ----
[lp-1  ] DEBUG 2023/09/16 05:32:47 charge power: 3566W
[site  ] DEBUG 2023/09/16 05:32:47 pv power: 0W
[site  ] DEBUG 2023/09/16 05:32:47 grid power: 3824W
[site  ] DEBUG 2023/09/16 05:32:47 grid currents: [15.9 0.72 0.41]A
[site  ] DEBUG 2023/09/16 05:32:47 site power: 3824W
[lp-1  ] DEBUG 2023/09/16 05:32:47 charge currents: [15.5 0 0]A
[lp-1  ] DEBUG 2023/09/16 05:32:47 charge total import: 933.988kWh
[lp-1  ] DEBUG 2023/09/16 05:32:48 charger status: C
[lp-1  ] DEBUG 2023/09/16 05:32:48 soc estimated: 51.06% (vehicle: 51.00%)
[lp-1  ] DEBUG 2023/09/16 05:32:48 vehicle soc: 51%
[lp-1  ] DEBUG 2023/09/16 05:32:48 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/16 05:32:48 vehicle range: 230km
[lp-1  ] DEBUG 2023/09/16 05:32:48 plan: charge 1h27m12s (required: 1h28m4s) starting at 2023-09-16 05:32:48 +0200 CEST until 2023-09-16 07:00:00 +0200 CEST (power: 11040W, avg cost: 0.207)
[site  ] DEBUG 2023/09/16 05:32:57 ----
[lp-1  ] DEBUG 2023/09/16 05:32:57 charge power: 3567W
[site  ] DEBUG 2023/09/16 05:32:57 pv power: 0W
[site  ] DEBUG 2023/09/16 05:32:57 grid power: 3827W
[site  ] DEBUG 2023/09/16 05:32:57 grid currents: [15.9 0.72 0.43]A
[site  ] DEBUG 2023/09/16 05:32:57 site power: 3827W
[lp-1  ] DEBUG 2023/09/16 05:32:57 charge currents: [15.5 0 0]A
[lp-1  ] DEBUG 2023/09/16 05:32:57 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/16 05:32:57 charge total import: 933.997kWh
[lp-1  ] DEBUG 2023/09/16 05:32:58 charger status: C
[lp-1  ] DEBUG 2023/09/16 05:32:58 soc estimated: 51.07% (vehicle: 51.00%)
[lp-1  ] DEBUG 2023/09/16 05:32:58 vehicle soc: 51%
[lp-1  ] DEBUG 2023/09/16 05:32:58 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/09/16 05:32:58 vehicle range: 230km
[lp-1  ] DEBUG 2023/09/16 05:32:58 plan: charge 1h27m2s (required: 4h24m2s) starting at 2023-09-16 05:32:58 +0200 CEST until 2023-09-16 07:00:00 +0200 CEST (power: 3680W, avg cost: 0.207)
`

bolisaf, 2023-09-16:
here's the screenshot from the UI again to show it's set to auto switch phases:

[image]

bolisaf, 2023-09-16:
me:
> But if it has the whole night to reach the charging goal, why should it use ‘fast’? I try to avoid this mode as much as possible for the capacity tariff. I would rather it charges the whole night at the lowest possible power.Thanks!L
@andig: 
> That‘s just how it works.
me:
> But again, I’d really prefer it didn’t switch to 3p but just started earlier, for reasons I mentioned earlier.
@andig 
> Fair, but that is a completely different topic.

am i reading this wrong or are these contradictory statements? if it doesn't spread out the charging as much as possible, I'll create an enhancement request, i just want to be sure

andig, 2023-09-16:
Thats just how it works. Plan charging tages as much PV as possible and covers the remainder at full speed.

bolisaf, 2023-09-16:
@andig 

i did some more testing. i logged the successful phase switch when i set the mode from 'pv' to 'fast'.

evcc debug:

[phaseswitch_evcc_debug.txt](https://github.com/evcc-io/evcc/files/12640518/phaseswitch_evcc_debug.txt)

alfeneve debug:

[phaseswitch_alfeneve_debug.txt](https://github.com/evcc-io/evcc/files/12640521/phaseswitch_alfeneve_debug.txt)

don't forget about the 2hour time difference :)

andig, 2023-09-16:
I'm really sorry. The additional logging is only in the nightly. I'll release .6 now and we'll need to do this again :(

bolisaf, 2023-09-18:
1. first start planned charge at 09:38 => only 1 phase
2. start fast charging after plan => also only 1p!
3. disconnect car and reconnect => switch to 3p worksµ
4. switch back to pv mode => switch to 1p works
5. switch back to fast mode => switch to 3p works
6. switch back to pv mode => switch to 1p works
7. reactivate plan => switych to 3p doesn't work
8. disconnect car and reconnect => now the switch to 3p does work as part of the plan !
9. cancel plan => switch to 1p works
10. switch to fast => switch to 3P doesnt work anymore

here's the logs. testing (more or less in the order that i state above) starts at 09h38 and ends at 10h31

[evcc_debug.txt](https://github.com/evcc-io/evcc/files/12645959/evcc_debug.txt)
[alfeneve_debug.txt](https://github.com/evcc-io/evcc/files/12645993/alfeneve_debug.txt)

andig, 2023-09-24:
That is really helpful, lets take a look:

We start in PV mode and successfully switch to 1p:

    [lp-1  ] DEBUG 2023/09/18 09:17:23 set charge mode: pv
    [lp-1  ] DEBUG 2023/09/18 09:20:32 phase scale1p timer elapsed
    [lp-1  ] DEBUG 2023/09/18 09:20:32 !! scalePhases: GetPhases 0p <> phases 1p
    [lp-1  ] DEBUG 2023/09/18 09:20:32 phase timer inactive
    [lp-1  ] DEBUG 2023/09/18 09:20:32 switched phases: 1p @ 4W

We switch to Now mode and successfully switch to 3p:

    [lp-1  ] DEBUG 2023/09/18 09:20:47 set charge mode: now
    [lp-1  ] DEBUG 2023/09/18 09:20:48 charger status: B
    [lp-1  ] DEBUG 2023/09/18 09:20:48 !! scalePhasesIfAvailable: 3p -> 3p
    [lp-1  ] DEBUG 2023/09/18 09:20:48 !! scalePhases: GetPhases 1p <> phases 3p
    [lp-1  ] DEBUG 2023/09/18 09:20:48 charger enable
    [lp-1  ] DEBUG 2023/09/18 09:20:48 wake-up timer: start
    [lp-1  ] DEBUG 2023/09/18 09:20:52 charger status: C
    [lp-1  ] INFO 2023/09/18 09:20:52 start charging ->
    [lp-1  ] DEBUG 2023/09/18 09:20:52 wake-up timer: stop
    [lp-1  ] DEBUG 2023/09/18 09:20:52 !! scalePhasesIfAvailable: 3p -> 3p
    [lp-1  ] DEBUG 2023/09/18 09:20:52 !! scalePhases: GetPhases 3p <> phases 3p
    [site  ] DEBUG 2023/09/18 09:21:01 grid currents: [15.6 15.5 15.8]A

Note: we do not see the phase switch in the log (bad- seems only PV mode is actually logging this) but we can see 3p charging.

We switch back to PV mode and successfully switch to 1p:

    [lp-1  ] DEBUG 2023/09/18 09:22:56 set charge mode: pv
    [lp-1  ] DEBUG 2023/09/18 09:26:01 charge power: 4159W
    [site  ] DEBUG 2023/09/18 09:26:01 grid power: 3936W
    [lp-1  ] DEBUG 2023/09/18 09:26:01 detected active phases: 3p
    [lp-1  ] DEBUG 2023/09/18 09:26:02 charger status: C
    [lp-1  ] DEBUG 2023/09/18 09:26:02 available power 223W < 4140W min 3p threshold
    [lp-1  ] DEBUG 2023/09/18 09:26:02 phase scale1p in 0s
    [lp-1  ] DEBUG 2023/09/18 09:26:02 phase scale1p timer elapsed
    [lp-1  ] DEBUG 2023/09/18 09:26:02 !! scalePhases: GetPhases 3p <> phases 1p
    [lp-1  ] DEBUG 2023/09/18 09:26:02 phase timer inactive
    [lp-1  ] DEBUG 2023/09/18 09:26:02 switched phases: 1p @ 223W

Sofar, so good. Now the plan starts, we're in PV mode:

    [lp-1  ] DEBUG 2023/09/18 09:39:02 plan: charge 45m58s (required: 46m7s) starting at 2023-09-18 09:39:02 +0200 CEST until 2023-09-18 10:25:00 +0200 CEST (power: 11040W, avg cost: 0.207)
    [lp-1  ] DEBUG 2023/09/18 09:39:02 !! scalePhasesIfAvailable: 3p -> 3p
    [lp-1  ] DEBUG 2023/09/18 09:39:02 !! scalePhases: GetPhases 1p <> phases 3p
    [lp-1  ] DEBUG 2023/09/18 09:39:02 max charge current: 16A
    [lp-1  ] DEBUG 2023/09/18 09:39:02 charger: guard elapse
    [lp-1  ] DEBUG 2023/09/18 09:39:02 guard timer inactive
    [lp-1  ] DEBUG 2023/09/18 09:39:02 charger enable
    [lp-1  ] DEBUG 2023/09/18 09:39:02 pv timer elapse
    [lp-1  ] DEBUG 2023/09/18 09:39:02 pv timer inactive
    [lp-1  ] DEBUG 2023/09/18 09:39:02 charger: guard elapse
    [lp-1  ] DEBUG 2023/09/18 09:39:02 guard timer inactive
    [lp-1  ] DEBUG 2023/09/18 09:39:11 charge currents: [8.14 0 0]A
    [lp-1  ] DEBUG 2023/09/18 09:39:11 charger status: C
    [lp-1  ] INFO 2023/09/18 09:39:12 start charging ->

Again we see that we've apparently trying to switch phases but we don't. Similar to Now mode we don't see if any switching actually happened (bad).

/cc @GrimmiMeloni hast Du eine Idee, was hier los ist? In jedem Fall ein hässlicher Bug.

andig, 2023-09-24:
@bolisaf I've improved logging in master branch. Could you please confirm if the Alfen log shows phase switching at 09:20:32, 09:20:48, 09:26:02 und most importantly at 09:39:02?

Once done please upgrade to master and also set `alfen: trace`.

andig, 2023-09-24:
Here is the phases setting from the Alfen log:

```
2023-09-18T07:20:29 #phases is 1 by: Modbus
2023-09-18T07:20:45 #phases is 3
2023-09-18T07:25:59 #phases is 1 by: Modbus
2023-09-18T07:38:59 #phases is 3 <- start of plan
2023-09-18T07:46:59 #phases is 1 by: Modbus
2023-09-18T07:47:19 #phases is 3
2023-09-18T07:50:59 #phases is 1 by: Modbus
2023-09-18T07:59:59 #phases is 3
2023-09-18T08:08:49 #phases is 1 by: Modbus
2023-09-18T08:09:05 #phases is 3
2023-09-18T08:13:19 #phases is 1 by: Modbus
2023-09-18T08:17:49 #phases is 3
2023-09-18T08:23:09 #phases is 1 by: Modbus
2023-09-18T08:26:47 #phases is 3
2023-09-18T08:38:01 #phases is 1 by: Modbus
```

It confirms that with plan start we're switching to 3p. It seems as if this is not working since vehicle continues charging at 1p. Its unclear to me why this happens.

Its also a bit unclear why the `phases=1` is shown as `by: Modbus` while `phases=3` is not, but that seems to be a firmware logging issue.

ATM I would say that evcc works exactly as planned- only the charger does not seem to perform the switch. I'd suggest to contact Alfen with these findings.

GrimmiMeloni, 2023-09-24:
I had a look, @andig.
The Alfen Log looks spurious to me. I have no first hand experience with modBus, so maybe I am misreading.
That said - searching for `#phases` in said log reveals

`2023-09-18T07:38:59.308Z:INFO:taskMain.c:6910:Socket #1: #phases is 3
`

However, 30 seconds later the charge is running on 1p only:

`2023-09-18T07:39:31.292Z:USER:taskMaster.c:6077:Socket #1: I1=15.6A I2=0.0A I3=0.0A P=3.6kW Idc=0.0mA
`

In between these lines quite a few things happen - most importantly charging is stopped and started in less than 2s

```
2023-09-18T07:38:59.484Z:INFO:taskMaster.c:4307:Socket #1: state main  : CHARGING_POWER_OFF
2023-09-18T07:39:01.234Z:INFO:taskMaster.c:4307:Socket #1: state main  : CHARGING_POWER_ON

```

Wild guess: Does the Alfen by any chance reset phases when cycling through charge off/on ?

andig, 2023-09-30:
> It confirms that with plan start we're switching to 3p. It seems as if this is not working since vehicle continues charging at 1p. Its unclear to me why this happens.

@bolisaf seems we're ready to close at this point. evcc log looks as expected, Alfen log is unclear. If you disagree please provide more inside- though I wouldn't see which at this point :(

andig, 2023-09-30:
> If you disagree please provide more inside- though I wouldn't see which at this point :(

Actually: since you're saying that 0.119 introduced the problem I could suggest to have a simple scenario and retest on 118 plus 120. Set `alfen: trace` in both cases and lets see if there is a difference that might make the box so behave differently.

closed 2023-09-30 by andig

bolisaf, 2023-12-29:
as soon as I updated the alfen eve to firmware 6.5.0-4217 phase switching was no longer an issue.