evcc-io/evcc #9552: Ladedauer wird nicht korrekt ermittelt bei Unterbrechung (PV-Überschuss)

lex777777, 2023-08-27:
### Describe the bug

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
sponsortoken: 
plant: 

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

#levels:
#  db: trace
# lp-1: debug

#log: debug
#levels:
#  site: trace

mqtt:
  broker: 192.168.1.100:1883
  topic: hbb33/config/evcc 

interval: 10s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3208  # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0832 # [currency]/kWh
  co2:
    type: grÃ¼nstromindex
    zip: 
meters:
- name: grid1
  type: custom 
  power:
    source: mqtt
    broker: 192.168.1.100:1883
    topic: hbb/raw/smart_meter/0100100700ff
    scale: 1
- name: pv1
  type: custom 
  power:    
    source: mqtt
    broker: 192.168.1.100:1883
    topic: hbb/devices/solar_inverter_sg8ktl-m/5031 - Total active power
    scale: 1
- name: pv2
  type: custom 
  power:    
    source: mqtt
    broker: 192.168.1.100:1883
    topic: hbb/devices/solar_inverter_sg4k6-d/5031 - Total active power
    scale: 1
- name: battery1
  type: template
  template: victron-energy
  usage: battery
  host: 192.168.1.196 # IP-Adresse oder Hostname 

chargers:
- type: template
  template: wallbe-meter
  host: 192.168.1.199  
  port: 502  
  name: wallbeGarage

loadpoints:
- title: Garage
  charger: wallbeGarage
  mode: pv
# phases: 0
  phases: 1
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true
  vehicle: Born

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv: 
     - pv1
     - pv2
    battery:
     - battery1
  residualPower: 100

vehicles:
  - name: Born
    type: template
    template: cupra
    title: Born
    user:
    password: 
    vin: 
    capacity: 58
    mode: pv
    targetSoc: 80
```

### Log details

```text
[site  ] ERROR 2023/08/26 14:41:33 tariff: outdated
[lp-1  ] INFO 2023/08/26 14:41:33 car connected
[lp-1  ] INFO 2023/08/26 14:41:43 start charging ->
[lp-1  ] INFO 2023/08/26 17:46:33 stop charging <-
[lp-1  ] INFO 2023/08/26 17:51:33 start charging ->
[main  ] INFO 2023/08/26 18:33:04 new version available: 0.119.1
[lp-1  ] INFO 2023/08/26 19:12:33 stop charging <-
[gsi   ] ERROR 2023/08/27 07:58:39 unexpected status: 429 (Too Many Requests)
[gsi   ] ERROR 2023/08/27 08:59:18 unexpected status: 429 (Too Many Requests)
[lp-1  ] INFO 2023/08/27 09:13:33 start charging ->
[gsi   ] ERROR 2023/08/27 10:00:05 unexpected status: 429 (Too Many Requests)
[lp-1  ] INFO 2023/08/27 10:17:23 stop charging <-
[lp-1  ] INFO 2023/08/27 10:22:13 start charging ->
[lp-1  ] INFO 2023/08/27 10:22:23 stop charging <-
```

### What type of operating system are you running?

Docker container

### Version

0.119.0

StefanSchoof, 2023-08-27:
Klingt ähnlich zu https://github.com/evcc-io/evcc/issues/9463

lex777777, 2023-08-27:
Jap, sorry nicht gesehen.

StefanSchoof, 2023-08-27:
Der ist auch gecloset, denke daher passt ein neuer Issue. War nur als Context gedacht

lex777777, 2023-08-28:
Okay, ich habe versucht es nachzuvollziehen.
Heißt die Funktion müsste entfernt werden?
https://github.com/evcc-io/evcc/blob/3428a7615a5c29968969cafde8933ed11e080d2e/charger/wallbe.go#L171

andig, 2023-08-30:
> Describe the bug

Hier fehlt erstmal jede Fehlerbeschreibung. Es gibt nichts zu tun!

lex777777, 2023-08-31:
Entschuldige @andig , habe es mal editiert.

andig, 2023-08-31:
@premultiply dann wäre das Proposal hier entweder:
- Wallbe reparieren (entfernen) oder
- Api generell zu entfernen und durch Loadpoint zu ersetzen

premultiply, 2023-08-31:
Ich muss mir das mal bei meiner Box genau ansehen was die da genau zählt.

Je nach Ergebnis dann Variante 1 (reparieren).

premultiply, 2023-08-31:
Ist tatsächlich so dass der Controller immer resettet:

```
pi@autosmatix:~ $ evcc charger
[main  ] INFO 2023/08/31 11:39:18 evcc 0.118.11 (29dbb687)
[main  ] INFO 2023/08/31 11:39:18 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/31 11:39:19 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/08/31 11:39:19 connecting evcc-1029455554 at tcp://localhost:                                                                                                                                                                                                                                             1883
Charge status: C
Enabled:       true
Duration:      9m48s
pi@autosmatix:~ $ evcc charger
[main  ] INFO 2023/08/31 11:39:54 evcc 0.118.11 (29dbb687)
[main  ] INFO 2023/08/31 11:39:54 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/31 11:39:55 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/08/31 11:39:55 connecting evcc-1908092765 at tcp://localhost:                                                                                                                                                                                                                                             1883
Charge status: B
Enabled:       false
Duration:      10m14s
pi@autosmatix:~ $ evcc charger
[main  ] INFO 2023/08/31 11:59:05 evcc 0.118.11 (29dbb687)
[main  ] INFO 2023/08/31 11:59:05 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/31 11:59:06 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/08/31 11:59:06 connecting evcc-1168195473 at tcp://localhost:                                                                                                                                                                                                                                             1883
Charge status: C
Enabled:       true
Duration:      23s
pi@autosmatix:~ $
```

Krass, War mir bislang noch nie aufgefallen.

andig, 2023-08-31:
@premultiply ich bin verwirrt. Die `phoenix-ev-eth` liest gar keine Ladezeit aus. Was überprüft also Dein Test?

premultiply, 2023-08-31:
Dafür gibt es ja derzeit noch unterschiedliche Implementierungen. Hier war es`wallbe`.

andig, 2023-08-31:
Hilf mir bitte- wo finde ich die im Master? Habs vorhin nicht entdecken können 😵

premultiply, 2023-08-31:
https://github.com/evcc-io/evcc/blob/216dcba3de6a241fcdadd0d0d6adebef424d4943/charger/wallbe.go#L171

closed 2023-09-03 by andig