evcc-io/evcc #9556: openWB software 2.0 issue

oreimerEAH, 2023-08-27:
### Describe the bug

After the update of my openWB series 1 charger to the software 2.0 evcc is unable to control or read the wallbox. 
EVCC is outputs: 
charger: openWB/system/Timestamp outdated: 47m42s
charge total import: openWB/system/Timestamp outdated: 47m42s
charge meter: openWB/system/Timestamp outdated: 47m42s

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

influx:
  url: http://localhost:8086
  database: evcc
  user: evcc
  password: evcc

# unique installation id
plant: 0a11b1ecacf65f5c76a1ce0fe1b329114c71df90effc4e8f56763a7e2cd1caed

interval: 10s # control cycle interval

sponsortoken: ####removed

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.33 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.00 # [currency]/kWh

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 10.1.1.21  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 10.1.1.21  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: 10.1.1.21  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid3
- type: custom
  power:
    source: http
    uri: http://10.1.1.24/data.jsn
    jq: .power_ac9
    scale: 0.5
  name: acthor

chargers:
- type: template
  template: openwb 
  host: 10.1.1.23  
  connector: 1  
  phases1p3p: true  
  name: wallbox5
- type: template
  template: go-e-v3 
  host: 10.1.42.23  
  name: wallbox6

vehicles:
- type: template
  template: bmw 
  title: BMW i4  
  user: ####
  password: ####  
  vin: ####  
  capacity: 81  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: off  
  minSoc: 10  
  targetSoc: 90  
  minCurrent: 6  
  maxCurrent: 32  
  priority: 1  
  name: ev4
- name: ev7
  type: template
  template: offline
  title: Fiat 500e
  capacity: 24  

loadpoints:
- title: Garage
  charger: wallbox5
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 24
  resetOnDisconnect: false
- title: Garten
  charger: wallbox6
  mode: off
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2
    aux: acthor
  prioritySoc: 80
  bufferSoc: 80
  bufferStartSoc: 90
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 250
```

### Log details

```text
[site  ] DEBUG 2023/08/27 13:15:37 ----
[lp-1  ] ERROR 2023/08/27 13:15:38 charge meter: openWB/system/Timestamp outdated: 52m21s
[lp-2  ] DEBUG 2023/08/27 13:15:38 charge power: 0W
[site  ] DEBUG 2023/08/27 13:15:38 pv power: 1346W
[site  ] DEBUG 2023/08/27 13:15:38 battery soc: 43%
[site  ] DEBUG 2023/08/27 13:15:38 battery power: -895W
[site  ] DEBUG 2023/08/27 13:15:38 grid power: -2W
[site  ] DEBUG 2023/08/27 13:15:38 giving priority to battery charging at soc: 43%
[site  ] DEBUG 2023/08/27 13:15:39 aux power 1: 0W
[site  ] DEBUG 2023/08/27 13:15:39 aux power: 0W
[site  ] DEBUG 2023/08/27 13:15:39 site power: 98W
[lp-1  ] ERROR 2023/08/27 13:15:39 charge meter: openWB/system/Timestamp outdated: 52m22s
[lp-1  ] ERROR 2023/08/27 13:15:39 charge total import: openWB/system/Timestamp outdated: 52m22s
[lp-1  ] ERROR 2023/08/27 13:15:39 charger: openWB/system/Timestamp outdated: 52m22s
[site  ] DEBUG 2023/08/27 13:15:47 ----
[lp-1  ] ERROR 2023/08/27 13:15:48 charge meter: openWB/system/Timestamp outdated: 52m31s
[lp-2  ] DEBUG 2023/08/27 13:15:48 charge power: 0W
[site  ] DEBUG 2023/08/27 13:15:48 pv power: 1348W
[site  ] DEBUG 2023/08/27 13:15:49 battery soc: 43%
[site  ] DEBUG 2023/08/27 13:15:49 battery power: -894W
[site  ] DEBUG 2023/08/27 13:15:49 grid power: 1W
[site  ] DEBUG 2023/08/27 13:15:49 giving priority to battery charging at soc: 43%
[site  ] DEBUG 2023/08/27 13:15:49 aux power 1: 0W
[site  ] DEBUG 2023/08/27 13:15:49 aux power: 0W
[site  ] DEBUG 2023/08/27 13:15:49 site power: 101W
[lp-2  ] DEBUG 2023/08/27 13:15:49 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/08/27 13:15:49 charge total import: 1370.645kWh
[lp-2  ] DEBUG 2023/08/27 13:15:49 charger status: A
[site  ] DEBUG 2023/08/27 13:15:57 ----
[lp-1  ] ERROR 2023/08/27 13:15:58 charge meter: openWB/system/Timestamp outdated: 52m41s
[lp-2  ] DEBUG 2023/08/27 13:15:58 charge power: 0W
[site  ] DEBUG 2023/08/27 13:15:58 pv power: 1351W
[site  ] DEBUG 2023/08/27 13:15:58 battery soc: 43%
[site  ] DEBUG 2023/08/27 13:15:58 battery power: -894W
[site  ] DEBUG 2023/08/27 13:15:58 grid power: -5W
[site  ] DEBUG 2023/08/27 13:15:58 giving priority to battery charging at soc: 43%
[site  ] DEBUG 2023/08/27 13:15:59 aux power 1: 0W
[site  ] DEBUG 2023/08/27 13:15:59 aux power: 0W
[site  ] DEBUG 2023/08/27 13:15:59 site power: 95W
[lp-1  ] ERROR 2023/08/27 13:15:59 charge meter: openWB/system/Timestamp outdated: 52m42s
[lp-1  ] ERROR 2023/08/27 13:15:59 charge total import: openWB/system/Timestamp outdated: 52m42s
[lp-1  ] ERROR 2023/08/27 13:15:59 charger: openWB/system/Timestamp outdated: 52m42s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.119.1

andig, 2023-08-27:
Hat OpenWB ein neues Api? Mir ist nichts bekannt?

oreimerEAH, 2023-08-28:
Ich bin zurück zur openWB 1.9. Es scheint sich einiges geändert zu haben. 
Auch das HomeAssistent Addon funktioniert nicht mehr. Aus der Diskussion dort habe ich folgendes mitgenommen: 
#https://github.com/a529987659852/openwbmqtt/issues/54#issuecomment-1690404951
_"For implentation it's important to understand that with openWB 2.0 the counters.for vehicle, Chargepoint, ... are not fixed. They increase everytime a new device of this type is added. And they never decrease."_

closed 2023-09-10 by andig

Lich2023, 2023-12-07:
Ich habe die open wb auf Steuerungsmodus: secudary und Steuerung über Modbus als secondary auf An gestellt
Ladepunkte oder sonstiges habe ich dort gar nicht definiert.
Nun nutze ich komplett EVCC mit der open wb 2.0 software

Hätte aber ein Fragen:
Ich habe eine Open wb mit display und dieses zeigt jetzt nurnoch "Die Webseite ist nicht erreichbar" an. Kann man dagegen etwas tun? Evtl sogar die EVCC Seite drauf spiegeln oder so? Überhaupt einfach ladewerte würden ja schon reichen, sieht so iwie sehr doof aus.

[image]