evcc-io/evcc #9560: Goodwe inverter showing home consumption and grip export as 0W   ///   Goodwe-Wechselrichter zeigt den Eigenverbrauch und den Stromexport als 0 W an

dxwil, 2023-08-27:
### Describe the bug

When the Goodwe inverter is added as a meter using Modbus TCP, it shows home consumption and grid export as 0W. The only information it shows is solar production.

Wenn der Goodwe-Wechselrichter über Modbus TCP als Zähler hinzugefügt wird, zeigt er den Eigenverbrauch und den Netzexport als 0 W an. Die einzige Information, die angezeigt wird, ist die Solarproduktion.

`evcc meter` output:

`Power: 4408W
Energy: 3439.4kWh`

### Steps to reproduce

1. Add Goodwe inverter as a meter using Modbus TCP
2.
3.
...

### Configuration details

```yaml
# unique installation id
plant: *****

interval: 10s # control cycle interval

vehicles:
- type: template
  template: tesla
  title: Model Y
  accessToken: *****
  refreshToken: *****
  name: ev1
  capacity: 82
  phases: 1
  mode: pv
  mincurrent: 5
  maxcurrent: 48
  minSoc: 30
  targetSoc: 70
  priority: 5
  icon: car

meters:
    - type: template
      template: goodwe-hybrid
      id: 247
      host: 10.161.13.147
      port: 502
      usage: pv
      modbus: tcpip
      name: pv1

chargers:
  -   type: template
      template: twc3
      host: 10.161.16.51
      name: wallbox1

site:
  title: Namai
  meters:
    pv:
      - pv1

loadpoints:
  - title: Namai
    charger: wallbox1
    vehicle: ev1
    mode: pv

Version: `0.119.0`
```

### Log details

```text
[main  ] INFO 2023/08/27 13:05:18 evcc 0.119.0
[main  ] INFO 2023/08/27 13:05:18 using config file: evcc.yaml
[main  ] INFO 2023/08/27 13:05:18 starting ui and api at :7070
[db    ] INFO 2023/08/27 13:05:19 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/08/27 13:05:19 phases not configured, assuming 3p
[site  ] INFO 2023/08/27 13:05:19 site config:
[site  ] INFO 2023/08/27 13:05:19   meters:      grid ✗ pv ✓ battery ✗
[site  ] INFO 2023/08/27 13:05:19     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/08/27 13:05:19   vehicles:
[site  ] INFO 2023/08/27 13:05:19     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/08/27 13:05:19 loadpoint 1:
[lp-1  ] INFO 2023/08/27 13:05:19   mode:        pv
[lp-1  ] INFO 2023/08/27 13:05:19   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/08/27 13:05:19   meters:      charge ✓
[lp-1  ] INFO 2023/08/27 13:05:19     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/08/27 13:05:19 phase timer inactive
[lp-1  ] DEBUG 2023/08/27 13:05:19 pv timer inactive
[lp-1  ] DEBUG 2023/08/27 13:05:19 guard timer inactive
[lp-1  ] INFO 2023/08/27 13:05:19 vehicle updated: unknown -> Model Y
[lp-1  ] DEBUG 2023/08/27 13:05:19 set charge mode: pv
[lp-1  ] DEBUG 2023/08/27 13:05:19 set priority: 5
[site  ] WARN 2023/08/27 13:05:19 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/08/27 13:05:19 ----
[lp-1  ] DEBUG 2023/08/27 13:05:19 charge power: 0W
[main  ] INFO 2023/08/27 13:05:20 new version available: 0.119.1
[site  ] DEBUG 2023/08/27 13:05:20 pv power: 3864W
[site  ] DEBUG 2023/08/27 13:05:20 site power: -3864W
[lp-1  ] DEBUG 2023/08/27 13:05:21 vehicle odometer: 37610km
[lp-1  ] DEBUG 2023/08/27 13:05:21 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/08/27 13:05:21 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/27 13:05:21 charger status: A
[lp-1  ] INFO 2023/08/27 13:05:21 car disconnected
[lp-1  ] DEBUG 2023/08/27 13:05:21 set charge mode: pv
[lp-1  ] DEBUG 2023/08/27 13:05:21 set priority: 5
```

### What type of operating system are you running?

Linux

### Version

0.119.0

premultiply, 2023-08-27:
Ja, ist ja auch nur von dir ausschließlich als PV-Meter konfiguriert.
Funktioniert also wie gewünscht.

closed 2023-08-27 by premultiply

dxwil, 2023-08-27:
Wie kann ich es so konfigurieren, dass der Netzexport und der Eigenverbrauch angezeigt werden?

dxwil, 2023-08-27:
Thank you, I figured it out