evcc-io/evcc #9564: Solar-mode starts charging directly after phase-switch without check for enough solar-power

Gi2elher, 2023-08-27:
### Describe the bug

Setup:
- KEBA P30 c series
- S10 phase switch
- modbus communication (3)

Situation:
plugged in the cable in the evening with battery fully charged but no solar power from the roof.
after a couple of minutes the phase-switch counter switched from 3 phases to 1 - but started charging.

The log does not show it I think - I will raise the log level and monitor the system...

### Steps to reproduce

1. 3P set from last charge
2. Solar mode is default
3. Cable plugged in in the evening
4. Counter for phase switch 1P starts but without charging
5. Phase switch 3 >> 1 happens after countdown
6. Charging starts immediately even without enough solar power from the roof

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: docker.home
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: *****

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true

plant: 780a6ea792954233cd3fe791a3b2f7a1ebd5493f8ec33787085b0b4326786965

# log settings
log: error
levels:
  site: error
  lp-1: error
  lp-2: trace
  cache: error
  db: error
[... 721 lines cut ...]
[influx] TRACE 2023/08/25 10:36:18 write sessionSolarPercentage=map[value:0] (map[loadpoint:Carport])
[keba  ] TRACE 2023/08/25 10:36:18 modbus: recv 08 f4 00 00 00 07 29 03 04 00 90 78 d8
[lp-1  ] DEBUG 2023/08/25 10:36:18 charge total import: 946.812kWh
[keba  ] TRACE 2023/08/25 10:36:18 modbus: send 08 f5 00 00 00 06 29 03 03 ec 00 02
[influx] TRACE 2023/08/25 10:36:18 write chargeTotalImport=map[value:946.812] (map[loadpoint:Carport])
[keba  ] TRACE 2023/08/25 10:36:18 modbus: recv 08 f5 00 00 00 07 29 03 04 00 00 00 00
[lp-1  ] DEBUG 2023/08/25 10:36:18 charger status: A
[keba  ] TRACE 2023/08/25 10:36:18 modbus: send 08 f6 00 00 00 06 29 03 03 e8 00 02
[keba  ] TRACE 2023/08/25 10:36:18 modbus: recv 08 f6 00 00 00 07 29 03 04 00 00 00 01
[influx] TRACE 2023/08/25 10:36:18 write homePower=map[value:531] (map[])
[influx] TRACE 2023/08/25 10:36:18 write greenShare=map[value:0.979702300405954] (map[])
[influx] TRACE 2023/08/25 10:36:18 write tariffEffectivePrice=map[value:0.08434370771312585] (map[])
[main  ] INFO 2023/08/25 10:36:31 evcc 0.119.0
[main  ] INFO 2023/08/25 10:36:31 using config file: /etc/evcc.yaml
[main  ] ERROR 2023/08/25 10:36:31 creating vehicle audi failed: cannot create vehicle 'template': cannot create vehicle 'etron': missing credentials
[site  ] ERROR 2023/08/27 10:13:12 pv 1 power: add[0]: read failed: dial tcp 10.100.1.190:502: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.119

Hstrohi, 2023-08-27:
Same behaviour here, since updating to versions 18 and higher.

andig, 2023-08-28:
Could you please try 0.119.2 in a few minutes?

Hstrohi, 2023-08-28:
After upgrading the behaviour still exists in my environment. Pluged-in the car and charging startet on one phase while consuming most of the power from battery.
Please send a short note, if you need any details or logs, please let me know.

Gi2elher, 2023-08-28:
Yep, still there - I noticed that I see 2 counters now. First the counter for switching phase 3P >> 1P and then a counter "not enough solar power" but when this 2nd counter ends it still charges with 1.4 KW.

GrimmiMeloni, 2023-08-30:
Just for clarity. The problem is not the battery usage, but the fact that charging starts at all. Correct?

Hstrohi, 2023-08-30:
The problem, as far as I can see, is that evcc is thinking there is still enough solar power and starts and afterwards continues charging.

andig, 2023-08-30:
From what I can tell the logfile does not show anything related to the actual description. Happy to reopen with a meaningful logfile based on 0.119.2- thank you!

closed 2023-08-30 by andig

Gi2elher, 2023-09-01:
[lp-1  ] DEBUG 2023/09/01 08:54:48 charge start energy: 1.02e+03kWh

[lp-1  ] INFO 2023/09/01 08:54:48 car connected

[lp-1  ] DEBUG 2023/09/01 08:54:48 vehicle api refresh

[lp-1  ] DEBUG 2023/09/01 08:54:48 pv timer elapse

[lp-1  ] DEBUG 2023/09/01 08:54:48 pv timer inactive

[lp-1  ] DEBUG 2023/09/01 08:54:48 charger: guard elapse

[lp-1  ] DEBUG 2023/09/01 08:54:48 guard timer inactive

[lp-1  ] DEBUG 2023/09/01 08:54:48 charge total import: 1019.354kWh

[lp-1  ] DEBUG 2023/09/01 08:54:48 available power -197W < 4140W min 3p threshold

[lp-1  ] DEBUG 2023/09/01 08:54:48 start phase scale1p timer

[lp-1  ] DEBUG 2023/09/01 08:54:48 phase scale1p in 5m0s

[lp-1  ] DEBUG 2023/09/01 08:54:48 pv charge current: 0A = 0A + -0.286A (197W @ 3p)

[lp-1  ] DEBUG 2023/09/01 08:54:48 pv enable timer reset

[lp-1  ] DEBUG 2023/09/01 08:54:48 pv timer inactive

[lp-1  ] DEBUG 2023/09/01 08:54:58 charge power: 0W

[lp-1  ] DEBUG 2023/09/01 08:54:58 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2023/09/01 08:54:58 charge total import: 1019.354kWh

[lp-1  ] DEBUG 2023/09/01 08:54:58 charger status: B

[lp-1  ] DEBUG 2023/09/01 08:54:58 available power -220W < 4140W min 3p threshold

[lp-1  ] DEBUG 2023/09/01 08:54:58 phase scale1p in 4m50s

[lp-1  ] DEBUG 2023/09/01 08:54:58 pv charge current: 0A = 0A + -0.319A (220W @ 3p)

[lp-1  ] DEBUG 2023/09/01 08:55:08 charge power: 0W

[lp-1  ] DEBUG 2023/09/01 08:55:08 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2023/09/01 08:55:08 charge total import: 1019.354kWh

[lp-1  ] DEBUG 2023/09/01 08:55:08 charger status: B

[lp-1  ] DEBUG 2023/09/01 08:55:08 available power -228W < 4140W min 3p threshold

[lp-1  ] DEBUG 2023/09/01 08:55:08 phase scale1p in 4m40s

[lp-1  ] DEBUG 2023/09/01 08:55:08 pv charge current: 0A = 0A + -0.33A (228W @ 3p)

[lp-1  ] DEBUG 2023/09/01 08:55:18 charge power: 0W

[lp-1  ] DEBUG 2023/09/01 08:55:18 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2023/09/01 08:55:18 charge total import: 1019.354kWh

[lp-1  ] DEBUG 2023/09/01 08:55:18 charger status: B

[lp-1  ] DEBUG 2023/09/01 08:55:18 available power -241W < 4140W min 3p threshold

[lp-1  ] DEBUG 2023/09/01 08:55:18 phase scale1p in 4m30s

[lp-1  ] DEBUG 2023/09/01 08:55:18 pv charge current: 0A = 0A + -0.349A (241W @ 3p)

[... 986 lines cut ...]
[lp-1  ] DEBUG 2023/09/01 09:05:28 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2023/09/01 09:05:28 charge total import: 1019.476kWh

[lp-1  ] DEBUG 2023/09/01 09:05:28 charger status: B

[lp-1  ] DEBUG 2023/09/01 09:05:28 final charge energy: 0.122kWh

[lp-1  ] INFO 2023/09/01 09:05:28 stop charging <-

[lp-1  ] DEBUG 2023/09/01 09:05:28 pv timer reset

[lp-1  ] DEBUG 2023/09/01 09:05:28 pv timer inactive

[lp-1  ] DEBUG 2023/09/01 09:05:28 charge total import: 1019.476kWh

[lp-1  ] DEBUG 2023/09/01 09:05:28 pv charge current: 0A = 0A + -1.04A (239W @ 1p)

[lp-1  ] DEBUG 2023/09/01 09:05:38 charge power: 0W

[lp-1  ] DEBUG 2023/09/01 09:05:38 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2023/09/01 09:05:38 charge total import: 1019.476kWh

[lp-1  ] DEBUG 2023/09/01 09:05:38 charger status: B

Gi2elher, 2023-09-01:
Behavior with 0.119.2:
- Car connected
- Charge Power 0W while in 3P
- Switch to 1P timer set
- 1P switch timer ends and switches phase from 3 to 1
- Charging starts with minimum 1440W at 1P
- evcc detects missing juice from the panels and starts stop-timer
- stop-timer ends
- charging stops

For about 5 minutes in between the car was charged after switching to one phase.

Gi2elher, 2023-09-01:
> @Gi2elher would be nice to have a readable log with all the unrelated influx or modbus. Makes it _much_ easier to see whats going on. Would also really help to have `site` logged which contains the grid data.

changing the log details now for the site-details. The behavior in 0.119.2 is better but still I think there is just a check for enough power missing after switching from 3 to 1 - just guessing ;)

andig, 2023-09-01:
Still need a full debug log including `site` please.