evcc-io/evcc #9573: Bei Wechselrichter Probleme wird weiter geladen

Crajs20, 2023-08-28:
Hallo,

Anlage: 
Solaredge Wechselrichter
Solaredge Wallbox (Keba)

Problem: Lichtbogenerkennung neu eingeschaltet, die Ärger macht und die Anlage öfters temp. deaktiviert. (Problem wird gelöst)
Dadurch erhält EVCC spontan keine Daten mehr, belässt aber die Wallbox unverändert eingeschaltet – lädt sie das Auto, wird der Strom komplett aus dem Netz bezogen, was vorher PV Strom war => unnötige Kosten.

Soll: Evcc soll die Wallbox sofort (mit eingestelltem Wert wie bei zu  wenig PV Strom, bei mir 2 Minuten) abschalten, wenn keine Werte vom Wechselrichter eintreffen oder bei Verbidungsproblemen.

mdkeil, 2023-08-28:
Wie schaut denn deine Config und das zugehörige Log aus? Denn eigentlich sind nur die Werte vom Grid-Meter relevant.. Bekommst Du diese ebenfalls über das Solaredge-Template? Wenn ja, auf was steht das timeout?

Crajs20, 2023-08-28:
Danke für die Rückmeldung.
In den Logs sehe ich bei einem entsprechenden Event dies Einträge:

[site  ] ERROR 2023/08/28 13:05:30 pv 1 power: add[0]: read tcp >IP>:46258-><IP>:1502: i/o timeout

GridMeter Config:
  - name: grid
    type: template
    template: solaredge-hybrid
    usage: grid

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: <IP> # Hostname
    port: 1502 # Port 

Eine TimeOut Option kann ich in der Config nicht finden - also Default.... :-)

mdkeil, 2023-08-28:
dann füge doch am besten mal unter modbus: tcpip noch timeout: 60s - keine Ahnung was der default-Wert ist.. im template selbst ist keiner festgelegt.

Crajs20, 2023-08-28:
Danke - in jedem Abschnitt zum ModBus?
timeout: 60s
einfügen  - heute Abend direkt ausporbieren....

mdkeil, 2023-08-28:
> Danke - in jedem Abschnitt zum ModBus?

reicht meiner Ansicht nach, wenn es beim usage: grid drinnen steht, da dieser Wert maßgeblich für die Regelung ist.. Schaden wird es aber vermutlich nicht.