evcc-io/evcc #9582: Heartbeat: read tcp Heidelberg Energy Control

redjudge13, 2023-08-28:
### Describe the bug

Version 0.119.1 und 0.119.2 verursacht Verbindungsprobleme zwischen Waveshare RS485 to RJ45 Ethernet Converter  und EVCC:

`[heidel] ERROR 2023/08/26 14:10:44 heartbeat: read tcp 192.168.0.10:45526->192.168.0.11:4196: i/o timeout`

### Steps to reproduce

1. Auto anstecken
2. Laden beginnen
3.
...

### Configuration details

```yaml
[main  ] INFO 2023/08/28 18:01:38 evcc 0.119.0
[main  ] INFO 2023/08/28 18:01:38 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/28 18:01:39 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/08/28 18:01:39 connecting evcc-1463022712 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/08/28 18:01:39 tcp://localhost:1883 connected
[site  ] WARN 2023/08/28 18:01:47 battery configured but residualPower is missing (add residualPower: 100 to site)

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

sponsortoken: *****

log: debug
levels:
  cache: error

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
- type: template
  template: victron-energy
  usage: grid
  host: 192.168.0.*
  port: 502
  name: grid1
- type: template
  template: victron-energy
  usage: pv
  host: 192.168.0.*
  port: 502
  name: pv2
- type: template
  template: victron-energy
  usage: battery
  host: 192.168.0.*
  port: 502
  name: battery3
- name: my_grid
  type: template
[... 142 lines cut ...]
Aug 26 14:10:57 iobroker evcc[278263]: [site  ] DEBUG 2023/08/26 14:10:57 site power: 10939W
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 charge voltages: [236 236 236]V
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 detected connected phases: 3p
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 charge currents: [15.7 15 15.1]A
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 detected active phases: 3p
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 charge total import: 294.332kWh
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 charger status: C
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 soc estimated: 74.38% (vehicle: 72.00%)
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 vehicle soc: 74%
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 vehicle soc limit: 100%
Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 vehicle range: 331km
Aug 26 14:11:07 iobroker evcc[278263]: [site  ] DEBUG 2023/08/26 14:11:07 ----
Aug 26 14:11:14 iobroker evcc[278263]: [heidel] ERROR 2023/08/26 14:11:14 heartbeat: read tcp 192.168.0.10:46370->192.168.0.11:4196: i/o timeout

// 0.119.2 
Aug 28 18:56:31 iobroker evcc[2062]: [heidel] ERROR 2023/08/28 18:56:31 heartbeat: read tcp 192.168.0.10:55306->192.168.0.11:4196: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.119.1 / 0.119.2

premultiply, 2023-08-28:
Mmmh. Und welche Version war vorher?

premultiply, 2023-08-28:
Beide Boxen mal komplett spannungslos gemacht?

Was sagt denn `evcc charger --log trace` bei gestopptem Dienst?

premultiply, 2023-08-28:
> ```
> Aug 26 14:10:57 iobroker evcc[278263]: [lp-1  ] DEBUG 2023/08/26 14:10:57 vehicle range: 331km
> Aug 26 14:11:07 iobroker evcc[278263]: [site  ] DEBUG 2023/08/26 14:11:07 ----
> Aug 26 14:11:14 [...]

Da vergehen aber auch mehrere Sekunden. Da klingt diese Meldung zumindest mal plausibel und gerechtfertigt.
Sicher dass das System nicht mit irgendwas anderem ausgelastet ist?

Ich lese da was von iobroker...
Wenn Raspberry: Das Netzteil liefert (noch) stabil seine 5,1V?
Was sagt denn z. B. `top` zur Systemauslastung oder die SD-Kartenaktivität?
Schon mal das System neugestartet?

Ich kann da erstmal keine dafür relevante Änderung im Code erkennen.

panzerdev, 2023-08-28:
[image]
Vergleich mal die settings vor allem bei den "Mliti-Host Settings" und schau mal ob das der Grund sein kann. Musste da auch mal nachstellen und seitdem keine Probleme mehr.

premultiply, 2023-08-28:
Eigentlich müsste das da alles aus sein.
Sowohl Modbus-Konvertierung als auch alles was mit Multihost und sonstiger Konverterintelligenz zu tun hat.
Die evcc-Konfig arbeitet hier mit Modbus RTU-over-TCP/IP und nicht mit Modbus TCP. Also alles auf 1:1 Kommunikation.

panzerdev, 2023-08-28:
Was eigentlich müsste ist mir egal weil es so super per Modbus TCP funktioniert. Vielleicht mal ein wenig andere Lösungen akzeptieren statt Zeigefinger...

premultiply, 2023-08-28:
Die zugehörige Konfig steht ja oben.

panzerdev, 2023-08-28:
Ja dann macht man aus `modbus: rs485tcpip` eben `modbus: tcpip` keine große Änderung. Habe 3 Wallboxen so am Waveshare hängen und mehrere Clients die Werte abfragen im Netzwerk.

premultiply, 2023-08-28:
Das ist seitens Template nicht vorgesehen und wie gesagt oben auch so nicht konfiguriert.
Daher erstmal Nebelkerze die hier wenig weiterhilft.

redjudge13, 2023-08-28:
Version, welche vorher und jetzt stabil läuft: 0.119.0
System ist ein leistungsstarker Host, 10% CPU Load, 50% RAM frei

Log -trace unter 0.119.0
`[main  ] INFO 2023/08/28 20:16:50 evcc 0.119.0
[main  ] INFO 2023/08/28 20:16:50 using config file: /etc/evcc.yaml
[db    ] INFO 2023/08/28 20:16:50 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                             ='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT sql FROM sqlite_master WHERE type IN (                                                                                             "table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "                                                                                             table" desc 1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2023/08/28 20:16:50 connecting evcc-1188767939 at tcp://localhost:                                                                                             1883
[mqtt  ] DEBUG 2023/08/28 20:16:50 tcp://localhost:1883 connected
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                             ='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                             ='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                             ='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT sql FROM sqlite_master WHERE type IN (                                                                                             "table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "t                                                                                             able" desc 1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                             ='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT sql FROM sqlite_master WHERE type IN (                                                                                             "table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by ty                                                                                             pe = "table" desc 2 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT * FROM `config_details` LIMIT 1 -1 <ni                                                                                             l>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                              = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_confi                                                                                             gs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CON                                                                                             STRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" O                                                                                             R sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                              = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                              = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devic                                                                                             es_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CON                                                                                             STRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" O                                                                                             R sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/08/28 20:16:50 SELECT count(*) FROM sqlite_master WHERE type                                                                                              = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR s                                                                                             ql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" O                                                                                             R sql LIKE "%   device_id       %") -1 <nil>
[... cut ...]
[heidel] TRACE 2023/08/28 20:27:05 modbus: send 01 03 01 05 00 01 95 f7
[heidel] TRACE 2023/08/28 20:27:05 modbus: recv 01 03 02 00 00 b8 44
Power:          0W
Energy:         312.2kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 242V 240V 240V
Charge status:  B
Enabled:        false
`
Wallboxen waren jetzt stromlos.

Kann den Fehler  gerade nicht reproduzieren, was ja toll wäre.
Ich finde, dass @andig ein tolles Projekt gezündet hat, also bitte nicht untereinander streiten...

Ich beobachte weiter... Wie gesagt tolle Leistung...

premultiply, 2023-08-28:
Ok, dann mache ich hier erstmal zu da es nichts zu tun zu geben scheint.

closed 2023-08-28 by premultiply