evcc-io/evcc #9584: Easee to many requests 429

wattnpapa, 2023-08-29:
### Describe the bug

Moin

dieses Phänomen habe ich gerade gesehen
Easee Wallbox to many requests vielleicht liegt es daran das ich pallet noch Tibber habe für das preisabhängige laden 

sollte man abfangen denke ich und die Anzahl der requests dann reduzieren um aus dem Fehler rauszukommen 

ein paar Daten habe ich mal unkenntlich gemacht 
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.119.1

andig, 2023-08-29:
> sollte man abfangen denke ich und die Anzahl der requests dann reduzieren um aus dem Fehler rauszukommen

Das haben wir bereits. Ohne ein trace log können wir hier leider nichts weiter tun. Wäre Klasse wenn Du das ergänzen könntest.

closed 2023-08-29 by andig

g33kt00l, 2023-08-31:
Hatte das gleiche Problem mit v.0.119.2.
Ob EVCC der Grund für too many Requests ist, weis ich nicht. 
Ich habe EVCC in meinem HomeAssistant integriert und lasse gleichzeitig in meiner Übersicht die aktuelle Ladeleistung der Easee Wallbox anzeigen. (eingerichtet über Easee HACS integration).
In der Übersicht wurde mir auch angezeigt, dass momentan der Sensor nicht bereit ist.

Was bei mir dann geholfen hat, war EVCC für 10 min zu stoppen.
Dann wurde nach kurzer Zeit in der Home Assistant Übersicht wieder der Sensor bzw. der aktuelle Wert der Ladeleistung angezeigt.

Danach habe ich wieder den EVCC Dienst gestartet und seitdem keine "too many requests 429" Meldung. 

Hoffe, das hilft, auch ohne Log-File?

Kann das trace-log aber gerne nachreichen, wenn ich gefunden habe wie ich das in Home Assistant erstellen kann.