evcc-io/evcc #9586: Charging does not stop when evcc is restarted in PV mode

MarkusGH, 2023-08-29:
### Describe the bug

When evcc is restarted and charger is charging but disabled, it is not synchronized

### Steps to reproduce

1. Charger is charging
2. evcc is restarted in PV mode, no PV power available
3. Charging will not stop

### Configuration details

```yaml
""
```

### Log details

```text
Messsage charger logic error: disabled but charging is missing from log
```

### What type of operating system are you running?

Linux

### Version

Any evcc version after 0.118.11

MarkusGH, 2023-08-29:
I would assume this commit breaks it: https://github.com/evcc-io/evcc/pull/9075

MarkusGH, 2023-08-29:
@Andi: Ich bin mir fast zu 100% sicher, dass das ohne weiteres reproduzierbar ist. Hab bitte einfach ein bisschen Vertrauen und probier es aus.

MarkusGH, 2023-08-29:
Meinetwegen - aber es ist vergebliche Mühe weil man in den Logs nichts sieht. Und wie geschrieben: Es ist sicher ohne weiteres reproduzierbar.

MarkusGH, 2023-08-29:
Mit eebus hat das nix zu tun, lp-1 ist eine openwb. Die eebus Daten in der Konfig sind ein Relikt.

andig, 2023-08-29:
Sollte per https://github.com/evcc-io/evcc/pull/9570 behoben sein, testbar im Nightly.

closed 2023-08-29 by andig

MarkusGH, 2023-08-30:
Jetzt bekommt man einen stetigen Strom von Warnungen: charger logic error: disabled but charging.
Verhalten aber grundsätzlich in Ordnung, laden stoppt wenn es soll und danach sind die Warnungen weg.
Wäre jedoch gut wenn der Status des chargers aktualisiert würde damit die Meldung verschwindet.

andig, 2023-08-30:
> Wäre jedoch gut wenn der Status des chargers aktualisiert würde damit die Meldung verschwindet.

Wird er, aber anscheinend dauert das länger? Wir könnten auch hier ein Delay für die Meldung mit einbauen.

andig, 2023-08-30:
Tatsächlich ist das catch22:

	if !enabled && lp.charging() {
		if lp.guardGracePeriodElapsed() {
			lp.log.WARN.Println("charger logic error: disabled but charging")
		}
		enabled = true // treat as enabled when charging
		lp.elapseGuard()
		return nil
	}

Im ersten Durchlauf wird Guard raus genommen und im zweiten ist es dann raus. Bleibt die Frage, warum Dein Ladepunkt immer noch "charging" ist?

Achso, ja: die OpenWB aktualisiert sich nur alle 10s. Kann es daran liegen? Sonst müsstest Du nochmal mit Debugging einsteigen :(

MarkusGH, 2023-08-30:
Die Meldung kommt minutenlang und geht erst weg wenn der charger von evcc disabled wird.
Ich gehe mal davon aus, dass nie Enable(true) aufgerufen wird und sich deswegen lp.charger.Enabled() nicht ändert.

andig, 2023-08-30:
Ich denke eher https://github.com/evcc-io/evcc/issues/9540#issuecomment-1699571931, aber das wäre wieder mal im Logfile ersichtlich ;)