evcc-io/evcc #9594: Solax Hybrid Inverter

mstaber, 2023-08-29:
### Describe the bug

I think the information in parsed incorrectly to communicate with the solax cloud and to get the information from the system.
In the evcc.yaml I inserted the correct information, token ID and SN of the inverter. If I call the API directly in the browser it sends back an answer.
EVCC instead sends this error to the frontend:
`pv 1 power: add[0]: Get \"https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.022093002564524e%2B23&sn=SVYDPCXXXXX\": dial tcp 47.254.152.24:443: i/o timeout`
For me it seems as if the tokenid isn't transfered correctly. The . is obsolete and the last characters "e%2B23" are wrong, too.
Is this a known bug or how can I solve this?
Kind regards

### Steps to reproduce

1. The same error occurs at each start of the application
2.
3.
...

### Configuration details

```yaml
interval: 10s # control cycle interval

meters:
- type: template
  template: solax-hybrid-cloud
  usage: grid
  tokenid: 202209300256452411851314
  serial: SVYDPCXXXX
  name: grid1
- type: template
  template: solax-hybrid-cloud
  usage: pv
  tokenid: 202209300256452411851314
  serial: SVYDPCXXXX
  name: pv2
- type: template
  template: solax-hybrid-cloud
  usage: battery
  tokenid: 202209300256452411851314
  serial: SVYDPCXXXX
  name: battery3

chargers:
- type: template
  template: pulsarplus
  name: wallbox5
```

### Log details

```text
grid meter: Get \"https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.022093002564524e%2B23&sn=SVYDPCTNDA\": dial tcp 47.254.152.24:443: i/o timeout
2

2 minutes ago
pv 1 power: add[0]: Get \"https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.022093002564524e%2B23&sn=SVYDPCTNDA\": dial tcp 47.254.152.24:443: i/o timeout
3 minutes ago
grid meter: Get \"https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.022093002564524e%2B23&sn=SVYDPCTNDA\": dial tcp 47.254.152.24:443: i/o timeout
2

3 minutes ago
pv 1 power: add[0]: Get \"https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=2.022093002564524e%2B23&sn=SVYDPCTNDA\": dial tcp 47.254.152.24:443: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

evcc 0.119.2

mdkeil, 2023-08-29:
pls try to set the tokenID in quotation marks ( " " )

mstaber, 2023-08-29:
That worked. Perfect. Thanks!