evcc-io/evcc #9595: UI Loadpoint settings dialog

panzerdev, 2023-08-29:
### Describe the bug

[image]

Bei `minCurrent = 0` Wird im Dialog nix angezeigt

Obwohl `phases: 1` eingestellt ist, wird bei der "Vorschau" der Ladegeschwindigkeiten mit 3 Phasen gerechnet. 

Alles nur Darstellungsprobleme.

Evtl. wäre es auch sinvoll, wenn vom Charger supported, nich nur ganzzahlige Schritte im Dropdown zu machen sondern zm. "6.8A" oder so. Vielleicht sogar Freitext Feld?

### Steps to reproduce

open dialog of loadpoint from config

### Configuration details

```yaml
- title: "Heizstab"
    charger: ac-thor # Wallbox Referenz
    mode: pv # charge mode (off, now, minpv, pv)
    phases: 1
    minCurrent: 0
    maxCurrent: 13
    meter: ac-thor-meter
    priority: 5
```

### Log details

```text
UI issue
```

### What type of operating system are you running?

Linux

VolkerK62, 2023-08-29:
sorry, aber `mincurrent: 0` macht irgendwie gar keinen Sinn

panzerdev, 2023-08-29:
Hatte auch 0.1 drinnen, macht auch keinen Unterschied. Ja 0 ist nicht perfekt aber 1 ist zu viel.

panzerdev, 2023-08-30:
hab grad nachgeschaut und im loadpoint sind beide werte `int` und somit ist 1 das Minimum. Fürs Auto kein Ding.. bei Verbrauchern mit geringeren Ladeleistung wäre ein Wert zwischen 0 und 1 evtl. sinnvoll. 

```
	MinCurrent        int
	MaxCurrent        int
```

PR #9728 (2023-09-06): Improve min/maxcurrent limit UI
- add lower min current options (0.125, 0.25, 0.5). No free form number input, they are messy on mobile. fixes #9595
- existing non-integer values (e.g. set via API) will now show correctly in settings modal. fixes #9595
- fixed phase-based min/max power estimation based on `phasesConfigured` and `phasesActive` fixes #6178

**Note:** 🧪 Feature remains experimental until we replace yaml config of this value with db persistance.

lower min current options
[image]

add existing non-int values
[image]
closed 2023-09-06 by naltatis