evcc-io/evcc #9596: template: huawei-dongle-powersensor 0 values of battery SOC an charge/uncharge power

annotationshacker, 2023-08-29:
### Describe the bug

Hi Guys, first thanks for your terrific in this project!

The battery values (charge/uncharge, SOC) of my meter are every times "0"(not null, only 0).
My hardware: 
Huawei Luna battery, a SUN2000-KTL-M1 inverter and the dongle using Modbus TCP to get data.

Firmware:
Dongle: V100R001C00SPC133
Inverter: V100R001C00SPC155
Battery Software & Firmware: V100R002C00SPC119

I'm also using HomeAssistant, where I can see successfully those values of the battery using the Huawei Solar Integration by https://github.com/wlcrs.

Do I use the false template or there anywhere else an error in my config?

Thanks!
Donald

### Steps to reproduce

$ evcc meter -c evcc.yaml 
[main  ] INFO 2023/08/29 13:47:01 evcc 0.119.0
[main  ] INFO 2023/08/29 13:47:01 using config file: evcc.yaml
[db    ] INFO 2023/08/29 13:47:01 using sqlite database: /home/donald/.evcc/evcc.db
grid1
-----
Power:          34W
Energy:         1965.8kWh
Current L1..L3: -0.23A -0.25A 0.85A

pv2
---
Power:  424W
Energy: 5944.3kWh

battery3
--------
Power:  -0W
Energy: 0.0kWh
Soc:    0%

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ocpp: trace  

# unique installation id
plant: ***

interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.35  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.35
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.35
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: ocpp
  stationid: EVB-P*******
  connector: 1
  idtag: '***'
  connecttimeout: 5m
  timeout: 2m
  getconfiguration: false
  bootnotification: false
  name: evbox-elvi
  
vehicles:
- type: template
  template: mini 
  title: Mini Cooper SE  
  user: ***  
  password: ***  
  vin: ***  
  capacity: 28.9  
  name: ev4

loadpoints:
- title: Carport
  charger: evbox-elvi
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: MySite
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[lp-1  ] DEBUG 2023/08/29 14:17:12 phase timer inactive
[lp-1  ] DEBUG 2023/08/29 14:17:12 pv timer inactive
[lp-1  ] DEBUG 2023/08/29 14:17:12 guard timer inactive
[site  ] DEBUG 2023/08/29 14:17:12 ----
[lp-1  ] DEBUG 2023/08/29 14:17:12 charge power: 0W
[site  ] DEBUG 2023/08/29 14:17:14 pv power: 196W
[site  ] DEBUG 2023/08/29 14:17:17 battery soc: 0%
[site  ] DEBUG 2023/08/29 14:17:17 battery power: 0W
[site  ] DEBUG 2023/08/29 14:17:18 grid power: 1W
[site  ] DEBUG 2023/08/29 14:17:21 grid currents: [-0.3 -0.28 0.74]A
[site  ] DEBUG 2023/08/29 14:17:22 site power: 1W
[lp-1  ] DEBUG 2023/08/29 14:17:22 charger status: A
[lp-1  ] DEBUG 2023/08/29 14:17:22 set charge mode: minpv
[lp-1  ] DEBUG 2023/08/29 14:17:22 set min current: 6
[lp-1  ] DEBUG 2023/08/29 14:17:22 set max current: 16
[lp-1  ] DEBUG 2023/08/29 14:17:22 set priority: 0
[site  ] DEBUG 2023/08/29 14:17:22 ----
[lp-1  ] DEBUG 2023/08/29 14:17:22 charge power: 0W
[site  ] DEBUG 2023/08/29 14:17:24 pv power: 195W
[site  ] DEBUG 2023/08/29 14:17:27 battery soc: 0%
[site  ] DEBUG 2023/08/29 14:17:27 battery power: 0W
[site  ] DEBUG 2023/08/29 14:17:29 grid power: -0W
[site  ] DEBUG 2023/08/29 14:17:32 grid currents: [-0.29 -0.27 0.74]A
[site  ] DEBUG 2023/08/29 14:17:33 site power: 0W
[lp-1  ] DEBUG 2023/08/29 14:17:33 charger status: A
[site  ] DEBUG 2023/08/29 14:17:33 ----
[lp-1  ] DEBUG 2023/08/29 14:17:33 charge power: 0W
[site  ] DEBUG 2023/08/29 14:17:34 pv power: 199W
[site  ] DEBUG 2023/08/29 14:17:37 battery soc: 0%
[site  ] DEBUG 2023/08/29 14:17:37 battery power: 0W
[site  ] DEBUG 2023/08/29 14:17:38 grid power: -1W
[site  ] DEBUG 2023/08/29 14:17:41 grid currents: [-0.29 -0.27 0.77]A
[site  ] DEBUG 2023/08/29 14:17:43 site power: -1W
[lp-1  ] DEBUG 2023/08/29 14:17:43 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.119.0

fzzybllz, 2023-08-30:
Hi @annotationshacker,
I think I have a very similar issue with my Huawei setup and I have still no solution for it.
Could you check your battery details here:
https://region04eu5.fusionsolar.huawei.com/pvmswebsite/assets/build/index.html#/view/device/NE=35228666/battery/details
I have one battery but it shows me 2 ESU on that page.
ESU 1 does not contain any values while ESU 2 does show the correct values. It seems like evcc is pulling the data of ESU 1 but not ESU 2 and I have no idea how to change this.

premultiply, 2023-08-30:
Add `storageunit: 2` to battery meter or use
`evcc configure --advanced` to build a new config file by answering the advanced parameter settings.