evcc-io/evcc #9599: Go-e: Frc operator (Neutral=0 vs On=2)

binsentsu, 2023-08-29:
### Describe the bug

I'm using the Go-E together with RFID authentication. When a valid RFID tag is scanned, the go-e charger always directly started the charging process although no enabled bit was sent by EVCC (eg. no sun). I contacted Go-e support about this if there was a way to handle the RFID scan as a 'release for charging' signal, without actually starting the charging process. They answered that this is standard behavior in 'standard mode' and proposed a workaround by setting the GO-E charging mode in ECO modus in the app iso normal loading mode and set it up that it never reaches the eco threshold.  This advice works, and now after scanning RFID, the loading process is directly paused by the charger with reason 'eco threshold not reached'.
Now the problem arises when EVCC eventually gives the enable signal: it sends 'Neutral = 0'. Using this value actually gives the go-e the assignment to start loading if 'it's ok according to its own rules'. Using the value 'On=2' will always start the loading process, no matter if there are ecosettings or plannings activated in the GO-E app charger settings. I tested it by triggering the local API and for this usecase value 'On=2' would be the correct value. The EVCC implementation used to sent 'On=2', but it was changed with PR #5597 after some discussions. I can't really catch why the 'on' signal was changed from 2 to 0, but I can confirm that sending 'On=2' without an authenticated RFID does _not_ start the loading process. Also switching between the 3 different values does not seems to block charging as mentioned in #5595

For the record: according to documentation the valid states for frc are: 
forceState (Neutral=0, Off=1, On=2)

If some cases need to be tested with the hardware, I'm happy to perform them.

Go-E firmware version: 055.5 (latest non beta release)

### Steps to reproduce

1. Put GO-E in Eco-modus through app
2. Authenticate with RFID
3. Start charging process through EVCC
4. Charging will not start

### Configuration details

```yaml
vehicles:
  - name: tesla_y
    type: template
    template: teslamate
    title: Tesla Model Y
    id: 1
    capacity: 75
chargers:
  - name: go-e_charger
    type: template
    template: go-e-v3
    host: XX # IP-Adresse oder Hostname
```

### Log details

```text
/
```

### What type of operating system are you running?

Docker container

### Version

0.119.1

andig, 2023-08-29:
/cc @premultiply

andig, 2023-08-29:
@binsentsu thank you for this issue and the detailed analysis. It seems that switching "on" instead of "neutral" would indeed be correct here and we should revert #5597.

closed 2023-09-06 by andig