evcc-io/evcc #9606: Cannot add Vehicles Cupra Leon eHybrid Gen3

TheHendla, 2023-08-29:
### Describe the bug

I cannot add the Cupra Leon eHybrid Gen3 to the evcc config.

The Error Message in Logfile: RS.security.9007: The context has not gained a legitimation due to missing access rights.

Even tried template cupra instead of seat, but with no success.

### Steps to reproduce

1. add car
2. read logfile

### Configuration details

```yaml
vehicles:
- type: template
  template: seat
  title: Cupra-eHybrid2
  user: <my-email>
  password: <my-pass>
  vin: VSSZZZKL9PR047111
  capacity: 12
  name: ev4
```

### Log details

```text
[main  ] ERROR 2023/08/29 23:00:55 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
```

### What type of operating system are you running?

Linux

### Version

0.119.2

andig, 2023-08-30:
Da gibts leider keine Lösung- das API sieht dein Auto nicht, warum auch immer. Du könntest noch `cupra` probieren.

TheHendla, 2023-08-30:
Wenn ich auf cupra Stelle, sagt er das er die FIN nicht kennt. Gibt es zu der API eine Doku? Dann könnte ich mir das Problem mal genauer anschauen.

andig, 2023-08-30:
Leider nein. Du kannst aber mal iobroker probieren und ggf. dort ein Issue aufmachen. Das ist meine Kopiervorlage.

TheHendla, 2023-08-31:
Ich werde naher einen Mitbenutzer anlegen und ihn rüber schicken. Weil wenn ich das Passwort ändere, loggt der mich überall aus.

blenni, 2023-09-04:
Vielleicht hilft auch die Info:

https://github.com/Farfar/seatconnect/pull/7

andig, 2023-09-04:
Spontan nicht. Gut wäre mal das durch einen Proxy laufen zu lassen und den HTTP Mitschnitt mit info@evcc.io zu teilen- danke!

andig, 2023-09-04:
Depends on https://github.com/TA2k/ioBroker.vw-connect/issues/256

blenni, 2023-09-04:
> Spontan nicht. Gut wäre mal das durch einen Proxy laufen zu lassen und den HTTP Mitschnitt mit [info@evcc.io](mailto:info@evcc.io) zu teilen- danke!

Kann ich heute Abend zur Verfügung stellen

hermann-59, 2023-09-05:
Ich habe seit dem 1.9. auch das Problem, dass mein Curpa Leon offline ist (siehe auch #9397). Vorher funktionierte es ohne Probleme. Eine Änderung der Konfiguration habe ich nicht gemacht, es gab keinerlei Änderungen am LAN, nur die verfügbaren Updates habe ich zeitnah eingespielt.
Ein login mit meinen Zugangsdaten auf https://www.cupraofficial.de/ funktioniert.

Hat noch irgendjemand eine Idee, was ich probieren könnte?

sh4qyn, 2023-09-05:
Gleiches Thema bei mir, hatte schon testweise von Seat auf Cupra gestellt, bringt aber nichts.

Matze03, 2023-09-06:
Das selbe hier, auch bei unserem Mii...

andig, 2023-09-06:
Gibts jemanden bei dem die Seat App funktioniert? Welche App? Bitte Zugangsdaten an info@evcc.io.

bsteimer, 2023-09-06:
> Gibts jemanden bei dem die Seat App funktioniert? Welche App? Bitte Zugangsdaten an [info@evcc.io](mailto:info@evcc.io).

Meinst du die Seat App oder auch die Cupra App. Letztere funktioniert bei mir unter iOS.

Matze03, 2023-09-06:
> Gibts jemanden bei dem die Seat App funktioniert? Welche App? Bitte Zugangsdaten an [info@evcc.io](mailto:info@evcc.io).

Seat & Cupra App funktionieren. Hab Dir gerade eine Einladung geschickt...

Targus87, 2023-09-06:
> Gibts jemanden bei dem die Seat App funktioniert? Welche App? Bitte Zugangsdaten an [info@evcc.io](mailto:info@evcc.io).

Meine funktioniert auch. Zugangsdaten habe ich dir Montag schon gesendet.

hermann-59, 2023-09-06:
Noch zur Info: Mein Cupra Leon war vom 26. bis 27.8. offline und dann ab dem 1.9. (bis heute).

derHeidecker, 2023-09-06:
na da hoffen wir mal auf eine baldige Lösung,... 
bin gestern vom Urlaub zurückgekommen und seitdem ist er auch offline :(

gr5604, 2023-09-06:
Nur zur Ergänzung in der Hoffnung, dass es weiter hilft.

Unser Seat Me electric ist in evcc ebenfalls offline. Problem zuletzt nachvollzogen mit der nightly 0.119.4 (d7ca1fc3)
und dem Kommando $ evcc vehicle

[main  ] INFO 2023/09/06 23:01:11 evcc 0.119.4 (d7ca1fc3)
[main  ] INFO 2023/09/06 23:01:11 using config file: /home/pi/evcc/evcc.yaml
[db    ] INFO 2023/09/06 23:01:11 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2023/09/06 23:01:13 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
ev4
---
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'seat': cannot get vehicles: RS.security.9007: The context has not gained a legitimation due to missing access rights.
Capacity: 36.8kWh
Features: [Offline]

Konfigurationsdatei evcc.yaml ist angehängt

Was bei mir allerdings funktioniert ist der connect mit IOBroker mit der aktuellen Version 0.1.2 ( heute herunter geladen ) des VW Adapters ( siehe Log im Anhang )

Solltest du meine Zugangsdaten für Seat brauchen, bitte einfach melden. Dann bekommst du sie wieder als PN

Ich hoffe der Input hilft weiter.

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/12542750/evcc.yaml.txt)

[iobroker.log.txt](https://github.com/evcc-io/evcc/files/12542748/iobroker.log.txt)

andig, 2023-09-07:
Fix ist in 30min im nightly- dann müssen wir mal ein paar Tage die Dauen drücken dass es keine andere Fahrzeugklasse gibt die dadurch kaputt geht...

closed 2023-09-07 by andig

Targus87, 2023-09-08:
> Fix ist in 30min im nightly- dann müssen wir mal ein paar Tage die Dauen drücken dass es keine andere Fahrzeugklasse gibt die dadurch kaputt geht...

Danke für den Fix.
Seat scheint wieder online zu sein.

sh4qyn, 2023-09-08:
Bei mir gehts auch wieder, danke!

hermann-59, 2023-09-08:
bei mir auch. danke.

TheHendla, 2023-09-11:
Bei mir geht es auch mit der Version: 0.119.5.

Danke und viele Grüße