evcc-io/evcc #9607: Zielladen startet nicht

tisoft, 2023-08-30:
### Describe the bug

Seit Version 0.119.0 funktioniert bei mir das Zielladen nicht mehr. Der Charger wird nicht enabled, wenn die Startzeit gekommen ist. In Version 0.118.x hat alles mit der selben Konfiguration noch funktioniert.

### Steps to reproduce

1. Zielladen in der UI aktivieren
2. Warten bis der Startzeitpunkt erreicht ist
3. Es sollte laden, tut aber nicht

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug

# unique installation id
plant: *****

interval: 10s # control cycle interval

# mqtt message broker
mqtt:
  broker: 192.168.178.25:11883
  topic: evcc
  user: *****
  password: *****

tariffs:
  grid:
    type: tibber
    token: *****
    homeid: "*****"
  feedin:
    type: fixed
    price: 0.0769

vehicles:
  - type: template
    template: mg2mqtt
    title: ***** # Wird in der Benutzeroberfläche angezeigt (Optional)
    user: *****
    vin: *****
    capacity: 46 # Akkukapazität in kWh (Optional)
    name: ev1

site:
  title: Mein Zuhause
  meters:
    pv: kaco
    grid: pulse

meters:
  - name: goe1_meter
    type: custom
    power:
      source: mqtt
      topic: go-eCharger/100565/nrg
      jq: .[11]
    energy:
      source: mqtt
      topic: go-eCharger/100565/eto
      jq: ./1000
  - name: kaco
    type: custom
[... 143 lines cut ...]
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:01 charger status: B
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:02 charger: guard elapse
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:02 guard timer inactive
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:02 planned 2h6m31s until 2023-08-30 06:00:00 +0200 CEST at 11040W: total plan duration: 2h6m31s, avg cost: 0.277
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:02 max charge current: 16A
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:02 pv timer elapse
Aug 30 02:00:02 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:02 pv timer inactive
Aug 30 02:00:11 evcc evcc[58131]: [site  ] DEBUG 2023/08/30 02:00:11 ----
Aug 30 02:00:11 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:11 charge power: 0W
Aug 30 02:00:11 evcc evcc[58131]: [site  ] DEBUG 2023/08/30 02:00:11 pv power: 0W
Aug 30 02:00:11 evcc evcc[58131]: [site  ] DEBUG 2023/08/30 02:00:11 grid power: 495W
Aug 30 02:00:11 evcc evcc[58131]: [site  ] DEBUG 2023/08/30 02:00:11 grid currents: [1.79 0.72 0.85]A
Aug 30 02:00:11 evcc evcc[58131]: [site  ] DEBUG 2023/08/30 02:00:11 site power: 495W
Aug 30 02:00:12 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:12 charge total import: 6583.457kWh
Aug 30 02:00:12 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:12 charger status: B
Aug 30 02:00:12 evcc evcc[58131]: [lp-1  ] DEBUG 2023/08/30 02:00:12 planned 2h6m31s until 2023-08-30 06:00:00 +0200 CEST at 11040W: total plan duration: 2h6m31s, avg cost: 0.277
```

### What type of operating system are you running?

Linux

### Version

0.119.2

andig, 2023-08-30:
Leider ist aus dem Logfile nicht zu erkennen, wann das laden starten sollte. Könntest Du bitte die Nightly Version probieren?

andig, 2023-08-30:
Die Logfiles sagen leider nichts aus. Auch das 118er Logfile zeigt keine Ladung (das wäre Status C). Bitte probier morgen nochmal das Nightly- da ist auch das Logging für die Planung verbessert. Bis dahin schließe ich hier- danke!

tisoft, 2023-08-30:
Mit dem nightly hat es gerade funktioniert:

```
Aug 30 20:48:08 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:08 ----
Aug 30 20:48:08 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:08 charge power: 0W
Aug 30 20:48:08 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:08 pv power: 0W
Aug 30 20:48:08 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:08 grid power: 950W
Aug 30 20:48:08 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:08 grid currents: [1.95 0.74 2.23]A
Aug 30 20:48:08 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:08 site power: 950W
Aug 30 20:48:08 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:08 charge total import: 6602.142kWh
Aug 30 20:48:08 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:08 charger status: B
Aug 30 20:48:08 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:08 planned 2h11m31s until 2023-08-30 23:00:00 +0200 CEST at 11040W: total plan duration: 2h11m31s, avg cost: 0.318
Aug 30 20:48:08 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:08 pv charge current: 0A = 0A + -4.13A (950W @ 1p)
Aug 30 20:48:18 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:18 ----
Aug 30 20:48:18 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:18 charge power: 0W
Aug 30 20:48:18 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:18 pv power: 0W
Aug 30 20:48:18 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:18 grid power: 944W
Aug 30 20:48:18 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:18 grid currents: [1.95 0.74 2.22]A
Aug 30 20:48:18 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:18 site power: 944W
Aug 30 20:48:18 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:18 charge total import: 6602.142kWh
Aug 30 20:48:18 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:18 charger status: B
Aug 30 20:48:18 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:18 planned 2h11m31s until 2023-08-30 23:00:00 +0200 CEST at 11040W: total plan duration: 2h11m31s, avg cost: 0.318
Aug 30 20:48:18 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:18 pv charge current: 0A = 0A + -4.1A (944W @ 1p)
Aug 30 20:48:28 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:28 ----
Aug 30 20:48:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:28 charge power: 0W
Aug 30 20:48:28 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:28 pv power: 0W
Aug 30 20:48:28 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:28 grid power: 947W
Aug 30 20:48:28 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:28 grid currents: [1.95 0.74 2.22]A
Aug 30 20:48:28 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:28 site power: 947W
Aug 30 20:48:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:28 charge total import: 6602.142kWh
Aug 30 20:48:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:28 charger status: B
Aug 30 20:48:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:28 planned 2h11m31s until 2023-08-30 23:00:00 +0200 CEST at 11040W: total plan duration: 2h11m31s, avg cost: 0.318
Aug 30 20:48:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:28 pv charge current: 0A = 0A + -4.12A (947W @ 1p)
Aug 30 20:48:38 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:38 ----
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 charge power: 0W
Aug 30 20:48:38 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:38 pv power: 0W
Aug 30 20:48:38 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:38 grid power: 951W
Aug 30 20:48:38 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:38 grid currents: [1.96 0.75 2.22]A
Aug 30 20:48:38 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:38 site power: 951W
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 charge total import: 6602.142kWh
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 charger status: B
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 planned 2h11m31s until 2023-08-30 23:00:00 +0200 CEST at 11040W: total plan duration: 2h11m21s, avg cost: 0.318
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 max charge current: 16A
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 charger enable
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 wake-up timer: start
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 pv timer elapse
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 pv timer inactive
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 charger: guard elapse
Aug 30 20:48:38 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:38 guard timer inactive
Aug 30 20:48:48 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:48 ----
Aug 30 20:48:48 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:48 charge power: 0W
Aug 30 20:48:48 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:48 pv power: 0W
Aug 30 20:48:48 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:48 grid power: 948W
Aug 30 20:48:48 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:48 grid currents: [1.95 0.74 2.23]A
Aug 30 20:48:48 evcc evcc[59343]: [site  ] DEBUG 2023/08/30 20:48:48 site power: 948W
Aug 30 20:48:48 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:48 charge total import: 6602.142kWh
Aug 30 20:48:48 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:48 charger status: B
Aug 30 20:48:48 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:48:48 planned 2h11m31s until 2023-08-30 23:00:00 +0200 CEST at 11040W: total plan duration: 2h11m11s, avg cost: 0.318
[... cut ...]
Aug 30 20:49:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:49:28 soc estimated: 61.26% (vehicle: 61.20%)
Aug 30 20:49:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:49:28 vehicle soc: 61%
Aug 30 20:49:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:49:28 vehicle range: 174km
Aug 30 20:49:28 evcc evcc[59343]: [lp-1  ] DEBUG 2023/08/30 20:49:28 planned 2h11m21s until 2023-08-30 23:00:00 +0200 CEST at 11040W: total plan duration: 2h10m31s, avg cost: 0.318
```

Ich beobachte das mal weiter. 😄

tisoft, 2023-09-12:
@andig Ist heute nacht mit der 0.19.5 wieder aufgetreten.

```
Sep 12 00:55:00 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:00 ----
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 charge power: 0W
Sep 12 00:55:00 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:00 pv power: 0W
Sep 12 00:55:00 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:00 grid power: 380W
Sep 12 00:55:00 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:00 grid currents: [1.57 0.7 0.71]A
Sep 12 00:55:00 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:00 site power: 380W
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 charge total import: 6862.310kWh
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 charger status: B
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 charger: guard elapse
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 guard timer inactive
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 plan: charge 54m34s starting at 2023-09-12 01:00:00 +0200 CEST until 2023-09-12 06:00:00 +0200 CEST (power: 11040W, avg cost: 0.288)
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 pv charge current: 0A = 0A + -1.65A (380W @ 1p)
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 site power 380W >= 0W disable threshold
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 pv disable in 0s
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 pv disable timer elapsed
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 charger disable
Sep 12 00:55:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:00 wake-up timer: stop
Sep 12 00:55:10 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:10 ----
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 charge power: 0W
Sep 12 00:55:10 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:10 pv power: 0W
Sep 12 00:55:10 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:10 grid power: 387W
Sep 12 00:55:10 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:10 grid currents: [1.58 0.7 0.71]A
Sep 12 00:55:10 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:10 site power: 387W
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 charge total import: 6862.310kWh
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 charger status: B
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 charger: guard elapse
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 guard timer inactive
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 plan: charge 54m34s starting at 2023-09-12 01:00:00 +0200 CEST until 2023-09-12 06:00:00 +0200 CEST (power: 11040W, avg cost: 0.288)
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 pv charge current: 0A = 0A + -1.68A (387W @ 1p)
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 site power 387W >= 0W disable threshold
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 pv disable in 0s
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 pv disable timer elapsed
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 charger disable
Sep 12 00:55:10 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:10 wake-up timer: stop
Sep 12 00:55:20 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:20 ----
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 charge power: 0W
Sep 12 00:55:20 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:20 pv power: 0W
Sep 12 00:55:20 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:20 grid power: 392W
Sep 12 00:55:20 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:20 grid currents: [1.59 0.7 0.71]A
Sep 12 00:55:20 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:20 site power: 392W
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 charge total import: 6862.310kWh
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 charger status: B
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 charger: guard elapse
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 guard timer inactive
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 plan: charge 54m34s starting at 2023-09-12 01:00:00 +0200 CEST until 2023-09-12 06:00:00 +0200 CEST (power: 11040W, avg cost: 0.288)
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 pv charge current: 0A = 0A + -1.7A (392W @ 1p)
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 site power 392W >= 0W disable threshold
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 pv disable in 0s
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 pv disable timer elapsed
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 charger disable
Sep 12 00:55:20 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:20 wake-up timer: stop
Sep 12 00:55:30 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:30 ----
Sep 12 00:55:30 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 00:55:30 charge power: 0W
Sep 12 00:55:30 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:30 pv power: 0W
Sep 12 00:55:30 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:30 grid power: 509W
Sep 12 00:55:30 evcc evcc[74235]: [site  ] DEBUG 2023/09/12 00:55:30 grid currents: [1.58 0.7 0.96]A
[... cut ...]
Sep 12 01:04:50 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 01:04:50 plan: charge 54m34s starting at 2023-09-12 01:04:50 +0200 CEST until 2023-09-12 06:00:00 +0200 CEST (power: 11040W, avg cost: 0.288)
```

 Zielladen sollte um 01:00:00 Uhr starten, aber der enable Befehl zur Wallbox ging nicht raus. Es sieht für mich so aus, als ob evcc denken würde, dass die Wallbox schon enabled ist. Der Letzte Ladevorgang vor dem Zielladen war 1-Phasig. Eventuell kommt das Problem durch das Umschalten auf 3-Phasig zustande?

Die Ladeleistung scheint ja noch gesetzt zu werden:

```
Sep 12 01:00:00 evcc evcc[74235]: [lp-1  ] DEBUG 2023/09/12 01:00:00 max charge current: 16A
```

andig, 2023-09-15:
Ich glaube hier gibt es irgendeinen inkonsistenten Zustand. Bei Dir steht in jedem einzelnen Zyklus:

```
[lp-1  ] DEBUG 2023/09/12 00:59:30 charger: guard elapse
[lp-1  ] DEBUG 2023/09/12 00:59:30 guard timer inactive
[lp-1  ] DEBUG 2023/09/12 00:59:30 pv disable in 0s
[lp-1  ] DEBUG 2023/09/12 00:59:30 pv disable timer elapsed
[lp-1  ] DEBUG 2023/09/12 00:59:30 charger disable
[lp-1  ] DEBUG 2023/09/12 00:59:30 wake-up timer: stop
```

Das kann mal vorkommen, sollte aber nur genau einmal der Fall sein.

/cc @GrimmiMeloni vielleicht hast Du eine Idee, was hier los ist?

andig, 2023-09-15:
@tisoft ich würde Dich mal bitten, das mit der normalen GoE Charger Implementierung zu versuchen um weitere Fehlerquellen auszuschließen. Wir machen mit neuem Logfile wieder auf.

closed 2023-09-15 by andig

GrimmiMeloni, 2023-09-16:
> /cc @GrimmiMeloni vielleicht hast Du eine Idee, was hier los ist?

Ich habe mal geschaut. Deine Analyse passt soweit. Um 1 Uhr landen wir offensichtlich in dem fastCharge block von Loadpoint.Update(). Bis setLimit() mit 16A passt alles. Es fehlt aber der Enable() Aufruf. Hmmm... neues Log wie vorgeschlagen wäre nochmal gut.

andig, 2023-09-16:
Bitte auch mal den Go-E auf trace setzen. Dann sehen wir in jedem Falle, wer da was startet.