evcc-io/evcc #9639: Manueller Lademodus mit über UI einstellbarer Ladestärke

allcoolusernamesaregone, 2023-08-31:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when ... die PV Anlage auf dem Dach 1,8 kW liefert und wegen Wolken die Leistung um +-1 kW schwankt. Kommen 2,5 kW vom Dach steuert EVCC in PV oder MinPV natürlich nach, kaum erfolgt geht's runter auf 1,1 kW und "jede Menge" Netzbezug. Im Modus PV wird dann das Laden gestoppt und kurz darauf mit 1,6 kW vom Dach wieder gestartet und ... ...
Das ist bei mir unnötig, da das Auto sich langsam aber stetig einfach mit der Mindestleistung vollnuckeln könnte...

**Describe the solution you'd like**
Mit einem manuell wählbaren Lademodus "Fix X Ampere auf Y Phasen" oder "x kW" könnte man sich das hoch und runter und nachsteuern sparen und einfach mit z.B. 7 A / 1,5 kW laden. Natürlich ist dann auch wieder Netzbezug wenn die Leistung runtergeht, aber wenn das nicht gewünscht ist => Modus PV.

**Describe alternatives you've considered**
EVCC ausschalten und Ladestärke manuell selbst einstellen..

allcoolusernamesaregone, 2023-08-31:
Nachtrag: 
2. Anwendungsfall, es ist zwar bewölkt und heitert immer mal wieder auf und Sonne pur - auch hier könnte man mit z.B. manuell eingestellten 3,4 kW laden. Dann wird zwar nicht hochgeregelt wenn mal kurz die Wolken verschwinden, aber wenn sie wieder da sind, wird nicht mit Netzbezug geladen bis runtergeregelt wird. Auch hier, stetiges langsam(er)es Laden, aber dann wirklich mit 100% Sonne.

naltatis, 2023-08-31:
Diese Funktion ist aktuell bereits über die API möglich. Wenn du nightly Builds verwendest gibts da für auch eine UI mit der du die Min/Max Leistung pro Ladepunkt einstellen kannst. Diese hat Auswirkung auf alle Modi. Heißt du könntest du "schnell" Laden in der von dir gewünschten Geschwindigkeit.

[image]

closed 2023-08-31 by naltatis