evcc-io/evcc #9640: ResidualPower skalieren mit PV Leistung

allcoolusernamesaregone, 2023-08-31:
**Is your feature request related to a problem? Please describe.**
Ein zu großes residualPower (z.B. 500) führt beim Modus PV ggf. nicht zum Ladestart mit 1,2 kW, wenn nur 1,5 kW PV Leistung vorhanden sind, aber trotzdem 300 Watt Überschuss da sind, aber 200 zu wenig lt. residualPower. 
Während bei 10 kW PV Leistung residualPower für auch größer 1 kW sein könnte, mit dem Hintergrund, dass zwar schnell(er) geladen wird, aber trotzdem ausreichend Reserve für übliche Haushaltsgeräte sind, damit diese nicht gleich bei kurzer Nutzung (z.B. Kaffeemaschine, 1000 Watt) zu Netzbezug führen. 

**Describe the solution you'd like**
ResidualPower könnte ein relativer Wert, z.B. auf x % der PV Leistung, gesetzt werden.

**Describe alternatives you've considered**
Keine

StefanSchoof, 2023-08-31:
Kannst du ein für dich sinnvolles Verhalten mit enable threshold (und ggf. Kleinem residualPower) erreichen? https://docs.evcc.io/docs/reference/configuration/loadpoints/#threshold

allcoolusernamesaregone, 2023-08-31:
Hmm, ich sehe jetzt nicht wie das das "Problem" lösen könnte, dass eine nennenswerte/größere Reserve (ResidualPower) bei wenig PV Leistung ein Laden verhindert, eine dann alternativ verkleinerte Reserve dann aber bei viel PV Leistung aber kaum erwähnenswert ist bzw. größere Verbraucher dann doch wieder Netzbezug verursachen... 
Letztlich verschiebt der Treshold doch effektiv nur den effektiven Wert von ResidualPower oder?

VolkerK62, 2023-08-31:
Ich verstehe das Problem mit dem Netzbezug nicht. Bei einem normal üblichen Intervall (z.B. 15 Sek) wird doch ruckzuck nachgeregelt.
Oder musst du (aus irgendwelchen Gründen) ein hohes Intervall einstellen?