evcc-io/evcc #9641: Set soc limit to a default value when a vehicle is connected

naltatis, 2023-08-31:
Since the last release evcc stores the last configured targetSoc per vehicle. A side effect of this is, that the previous `resetOnDisconnect` behavior does not work any more.

With this change, we will introduce a UI configurable option to set the target soc to a fixed default value. The current behavior (keep last selected limit) will still be possible.

TheMr-S, 2023-09-01:
So what do I have to do if I want EVCC to everytime use the targetsoc value that is set in the .yaml when the car connects?
I recently had it several times that my car would load to 100% whith me not realizing it because I thought it would reset everything every time after a disconnect.

naltatis, 2023-09-02:
The targetsoc values in the evcc yaml dont have an effect any more. The current behaviour is, that we remeber the last used targetsoc per vehicle. This will also be used for the next charging session. Do you experience a behaviour where evcc switches back to 100% without you having set the limit to this value? If yes please explain your setup so that we can reproduce it.

TheMr-S, 2023-09-02:
I will investigate and report if this is the case.

naltatis, 2023-09-02:
Quickfix to enable previous behavior #9685

naltatis, 2023-09-03:
https://github.com/evcc-io/evcc/pull/9685 Ist nun gemerged. Damit ist `targetSoc` am Fahrzeug in der yaml wieder aktiv. Bitte mit dem heutigen Nightly einmal testen.

Die Lösung über die UI machen wir trotzdem noch. Allerdings werden wir in dem Zuge das ganze Thema "wer überschreibt was" und "welches Setting lebt wie lange" überarbeiten. Das wird mehr Zeit in Anspruch nehmen.

andig, 2023-09-09:
Ich denke wir brauchen:
- einen Defaultwert je Fahrzeug (der heute aus yaml kommt) 
- einen Sessionwert am Loadpoint (NICHT am Fahrzeug)
- Der Sessionwert verschwindet beim Abstecken oder neuen Anstecken (analog heutigem Verhalten von `resetOnDisconnect`)

Wir brauchen:
- neue Api: für Default am Fahrzeug
- generell ein API für Fahrzeuge
- einen Mechanismus um Fahrzeugsettings aus API für Loadpoint erreichbar zu machen 
- ...und Updates zu konsumieren 

Wir sollten zusätzlich das kopieren entfernen und die Werte zur Laufzeit ermitteln.

fow0ryl, 2023-09-18:
Hallo,
ich nutze EVCC zur Zeit nur,  um bei meinen Twingo den Ladevorgang bei 80% abzubrechen (PV kommt erst noch..). Im Fahrzeug gibt es keine Möglichkeit da etwas zu stellen. Der kann nur 100%.
Eine dynamische Ermittlung ist also gar nicht sinnvoll möglich. Insofern muss der Default für ein Fahrzeug irgendwo gespeichert werden...

naltatis, 2023-09-18:
@fow0ryl Ich verstehe nicht ganz genau was die Aussage ist. Das Ladelimit lässt sich immer per UI einstellen. Über `targetSoc: 80` an der Fahrzeugkonfiguration kannst du zusätzlich auch einen Wert angeben der beim Anstecken des Fahrzeuges verwendet wird. Dieses Feature war kurzzeitig weg, ist inzwischen aber wieder da.

In diesem Issue geht es (inzwischen) nur noch darum die yaml Einstellung zu einer UI-Einstellung zu überführen.

fow0ryl, 2023-09-18:
Da muss ich etwas ausholen, obwohl mir das hier nicht der richtige Ort zu sein scheint.

Schauen wir mal hier: https://docs.evcc.io/docs/guides/
Warum dieser Punkt "Tipps und FAQ" heißt ist mir bis heute verborgen geblieben.
Nach der Installation wird ja mit "evcc configure" eine erste Konfiguration vorgenommen. Wenn da irgendwas schief geht, ist zwar beschrieben, wie man die einzelnen Konfigurationen prüft, aber leider nicht, wie die Syntax der einzelnen Parameter ist.
Das ist zumindest im Zweig "Debian..." so.

Wählt man gar den Zweig "Manuell", dann erfährt man nicht einmal was über die "Test-Funktionen". Mit Glück zufällig, oder eben, wenn man alles durchliest. Aber wozu dann eine Gliederung?

Für micht hätte der ganze Punkt "Installation -> Debian -> Testen" eher in den Punkt "Konfiguration" gehört. 

Da ich noch keine PV Anlage habe, habe ich EVCC vorgegaukelt, das ich immer 30kW Überschuß produiziere. Bislang hatte ich dazu monatelang den Modus "PV" verwendet. Seit ein paar Releases hat das aber dazu geführt, das das Fahrzeug nach Einstecken des Typ2 Steckers nicht mehr geladen wurde.
Ich musste also jedesmal in GUI gehen und auf "Schnell" oder "Min+PV" klicken. Leider hatte ich noch keine Zeit das Problem näher zu untersuchen.
Um das Klicken zu ersparen habe ich dann den Ladepunkt in der evcc.yaml fest auf "MinPV" gesetzt.
Die ganzen Warnungen wegen der deprecated Parameter haben mich dann allerding nervös gemacht ...

Unter "Geräte -> Hausinstallation -> Generische Unterstützung" wird z.B. nicht erklärt, was der Parameter "Energy:" bedeutet. Er ist optional und die Angabe hat in kWh zu erfolgen.
Aber was bewirkt "Energy" genau ?

Wenn ich mir dann https://docs.evcc.io/docs/reference/configuration/vehicles ansehe, steht dort bei "onIdentify", das "minSOC" veraltet ist, und der Wert im Fahrzeug gespeichert wird, wenn er über die GUI verändert wird.
Ich habe in der GUI bislang nichts derartiges wahrgenommen. Außerdem, wie soll das im Fahrzeug gespeichert werden, wenn das vom Fahrzeug gar nicht unterstützt wird?
Meine spontane Frage, wie soll das Feature bei solchen Fahrzeugen verwendet werden?

Das Einstellen des Ladelimits per UI hat hier noch nie funktioniert. Vielleicht liegt das aber wieder an meinem Fahrzeug. Dort kann man ja derartige Werte nirgends angeben und smoti vermutlich auch weder speichern, noch in irgendeiner Weise benutzen. 
"targetSoc" war daher die einzige Möglichkeit den Ladezustand zu begrenzen.

Wenn es hier jetzt nur noch um die Unterstützung der Parameter in der GUI geht, so hätte ich erwartet, das der Issue  geschlossen wird und ein neuer aufgemacht wird.

All dies überfordert einen alten Maschinenbauer.
Daher meine Anmerkung ....

closed 2023-12-02 by andig