evcc-io/evcc #9654: Augment logging output for "charger: timeout"

mrandt, 2023-08-31:
**Is your feature request related to a problem? Please describe.**

I operate two Volkswagen (aka Elli / EV Box) ID Chargers (poor me). 

Whenever the EEBUS over IP communication fails, I receive a "charger timeout" message in UI and also in log file. However I cannot tell which charger it is from.

Example:

`[lp-1  ] ERROR 2023/08/31 16:07:57 charger: timeout`

Obviously not an issue if you have one charger, but if you have multiple it is hard to figure out which one is at fault.

**Describe the solution you'd like**

It would be really helpful if the logging output contained a reference to the charger which timed out.

**Describe alternatives you've considered**

I have set log levels to DEBUG to try and figure out which charger is giving the problems based on timely correlation, but that was difficult.

**Additional context**

Here is a screenshot of the message in UI:

[image]

In my evcc.yaml I have defined the following chargers:

```
chargers:
- type: template
  template: ellipro
  ski: [truncated]
  ip: 192.168.1.244
  name: wallbox1
- type: template
  template: ellipro
  ski: [truncated]
  ip: 192.168.1.250
  name: wallbox2
```

So the log output could probably contain the name of the charger, like:

`[lp-1  ] ERROR 2023/08/31 16:07:57 charger wallbox1: timeout`

Thank you for considering this, greatly appreciated.

StefanSchoof, 2023-08-31:
Sounds like the same as https://github.com/evcc-io/evcc/issues/9654

andig, 2023-08-31:
Yeah, really sorry for this. Maybe we can do something without refactoring the entire logging.

@naltatis could we look into this together?

naltatis, 2023-09-01:
Yes, we should!

mrandt, 2023-09-01:
> Sounds like the same as #9654
> 
> Edit: Copy and Paste error. I wanted to link #7971

Sorry for creating a duplicate, I did not find the issue mentioned as I only searched using english words.

Also, from reading the other issue, I learned out that the logger `[lp-1  ]` indicates the loadpoint - I guess I could have guessed... 

But I still think making this explicit - and visible on the UI - would be helpful.

andig, 2023-09-06:
Das ist ein Duplikat zu https://github.com/evcc-io/evcc/issues/7971

closed 2023-09-06 by andig