evcc-io/evcc #9667: Vehicle MinSoc Option is not working anymore

thierolm, 2023-09-01:
### Describe the bug

My vehicle (Renault Zoe) is configured with `MinSoc` = 40.
Home battery SoC is greater than site prioritySoC 89% > 70% 
Loadpoint is in PV mode.
Vehicle is connected with a SoC < 40%,

but, evcc is not starting charging ... :-(

Vehicle config is providing MinSoc:
```sh
/app # evcc -l debug vehicle

renault
-------
Soc:            15%
Capacity:       52.0kWh
Charge status:  B
Range:          49km
Odometer:       41869km
Finish time:    2023-09-03 01:14:00 +0200 CEST
Climate active: false
Position:       49.81697585,9.05620257
OnIdentified:   Mode:pv, MinCurrent:9, MaxCurrent:16, MinSoc:40, TargetSoc:95
Features:       []
```

[image]

### Steps to reproduce

1. start evcc with connected car and Soc < 15%

### Configuration details

```yaml
...
vehicles:
- type: template
  template: renault
  title: ZOE Experience R135/Z.E. 50
  user: xxx
  password: xxx
  vin: xxx
  capacity: 52
  mode: pv
  minSoC: 40
  targetSoC: 95
  phases: 1
  minCurrent: 9
  maxCurrent: 16
  name: renault

...

site:
  title: Grid/Solar 8.5kWp  # display name for UI
  meters:
    grid: grid # grid meter
    pv:       # pv meters
    - pv
    - balkonpv
    battery: battery # battery meter
  prioritySoC: 70 # give home battery priority up to this soc (0 to disable)
  residualPower: 100

...

loadpoints:
- title: go-e # display name for UI
[... 359 lines cut ...]
[site  ] DEBUG 2023/09/01 17:52:11 ----
[lp-1  ] DEBUG 2023/09/01 17:52:11 charge power: 0W
[site  ] DEBUG 2023/09/01 17:52:12 pv power: 953W
[site  ] DEBUG 2023/09/01 17:52:12 battery soc: 89%
[site  ] DEBUG 2023/09/01 17:52:12 battery power: -469W
[site  ] DEBUG 2023/09/01 17:52:12 grid power: 5W
[site  ] DEBUG 2023/09/01 17:52:12 grid currents: [0.511 1.72 0.667]A
[site  ] DEBUG 2023/09/01 17:52:12 site power: -364W
[site  ] DEBUG 2023/09/01 17:52:26 ----
[lp-1  ] DEBUG 2023/09/01 17:52:26 charge power: 0W
[site  ] DEBUG 2023/09/01 17:52:27 pv power: 951W
[site  ] DEBUG 2023/09/01 17:52:27 battery soc: 90%
[site  ] DEBUG 2023/09/01 17:52:27 battery power: -498W
[site  ] DEBUG 2023/09/01 17:52:27 grid power: 13W
[site  ] DEBUG 2023/09/01 17:52:27 grid currents: [0.405 1.65 0.549]A
[site  ] DEBUG 2023/09/01 17:52:27 site power: -385W
```

### What type of operating system are you running?

Linux

### Version

0.112.3

thierolm, 2023-09-01:
Im Log taucht nirgendwo der gesetzte `MinSoc` Wert 40 auf.
Auch die für die Zoe konfigierten `phases` = 1, finde ich nicht im Log.

StefanSchoof, 2023-09-01:
Take a look at https://github.com/evcc-io/evcc/discussions/9621#discussioncomment-6875076

thierolm, 2023-09-01:
OK, war im Urlaub .... habe https://github.com/evcc-io/evcc/pull/8115 übersehen ...

closed 2023-09-01 by thierolm