evcc-io/evcc #9668: Ladevorgänge nicht mehr aufrufbar nach 0.119.x hinter Reverse Proxy

Tedfierro, 2023-09-01:
### Describe the bug

Seit dem Update auf Version 0.119.1 und höher, lässt sich das Menü "Ladevorgänge" nicht mehr aufrufen... Wenn ich das Downgrade auf 0.118.11 durchführe, geht es wieder. Dies tritt nur hinter meinem Apache2 Reverse Proxy auf, beim Aufruf der lokalen IP geht es nach wie vor.
Hier der Error in der Konsole:
`VM411 index-c6fb1d50.js:11 DOMException: Failed to execute 'define' on 'CustomElementRegistry': the name "shopicon-regular-arrowup" has already been used with this registry at https://my.domain/assets/index-c6fb1d50.js:21:359`

Hier die Config der Apache Site-Enabled:
`<IfModule mod_ssl.c>
<VirtualHost *:443>
ServerAdmin x
ServerName my.domain
ProxyPreserveHost On
DocumentRoot /var/www/html

ProxyPass / http://127.0.0.1:7070/
ProxyPassReverse / http://127.0.0.1:7070/
RewriteEngine on

RewriteCond %{HTTP:Upgrade} =websocket [NC]
RewriteRule /(.*)           ws://127.0.0.1:7070/$1 [P]

RewriteCond %{HTTP:Upgrade} !=websocket [NC]
RewriteRule /(.*)           http://127.0.0.1:7070/$1 [P]

<Location />
 Deny from all
 #Allow from 127.0.0.1  ##Set IP to allow access without password
 AuthUserFile /etc/apache2/.htpasswd
 AuthName "Restricted Area"
 AuthType Basic
 Satisfy Any
 require valid-user
</Location>

Include /etc/letsencrypt/options-ssl-apache.conf
SSLCertificateFile /etc/letsencrypt/live/my.domain/fullchain.pem
SSLCertificateKeyFile /etc/letsencrypt/live/my.domain/privkey.pem

</VirtualHost>
</IfModule>`

### Steps to reproduce

1. Update auf 0.119.x
2. Aufruf der UI über den Reverse Proxy

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.119.3

andig, 2023-09-02:
Ich habs mal vorsichtig auf Bug gesetzt auch wenn reverse proxy nicht als Feature beworben wird.

jheinitz, 2023-09-02:
Nur mal so als Hinweis. Bei mir läuft Version `0.119.3` einwandfrei hinter einem Traefik 2.x Proxy und die Ladevorgänge lassen sich auch aufrufen.

andig, 2023-09-03:
@Tedfierro irgendwas spannendes in der Browserkonsole?

naltatis, 2023-09-03:
Könnte ein Seiteneffekt von https://github.com/evcc-io/evcc/pull/9464 sein. Ich muss das aber reproduzieren. Welchen Browser verwendest du?

davidgiga1993, 2023-09-03:
> Nur mal so als Hinweis. Bei mir läuft Version `0.119.3` einwandfrei hinter einem Traefik 2.x Proxy und die Ladevorgänge lassen sich auch aufrufen.

Läuft bei mir auch hinter nginx mit authentic als IDP. Vermute ein Konfigurationsproblem beim Apache.

Tedfierro, 2023-09-03:
@naltatis Chrome 116.x Stable, aktuellste Version. Tritt aber auch auf Chrome Android auf. Gerade habe ich Firefox getestet - dort gehts! 
@andig In der Konsole steht nur obige Fehlermeldung. Und sonst halt "websocket connected", "websocket disconnected" Aber das war vorher genau auch schon so.

naltatis, 2023-09-03:
@Tedfierro Tritt das Problem auch auf, wenn du die Demo-Instanz aufrufst?
https://demo.evcc.io/
 
Poste bitte mal den Inhalt des Netzwerk-Tabs vom Chrome. Also so:
[image]

Tedfierro, 2023-09-03:
@naltatis Nein, bei der Demo tritt das nicht auf...
Hier das Foto vom Netzwerk der Demo:
[image]
Hier von meiner eigenen Seite:
[image]

naltatis, 2023-09-03:
Magst du auch noch mal den Konsole Tab vom zweiten posten?

Tedfierro, 2023-09-04:
Gerne
[image]

Ich hab jetzt selbst nochmals getestet: Rufe ich im Incognito Modus die Seite auf, gebe Name und Passwort in der Abfrage ein, kann ich die Ladevorgänge aufrufen. Gehe ich aber über mein Lesezeichen im Sinne von "https://name:passwort@my.domain", dann geht es nicht... Also muss es wohl mit der Authentifizierung bei Apache2 liegen, nur kann ich mir nicht erklären, warum das nach deiner Optimierung ab 0.119.x auftritt...

naltatis, 2023-09-04:
Die Ursache der Fehlermeldung ist, dass die `/assets/index-.*.js` Datei zwei Mal ausgeführt wird. Dadurch probiert der Dinge zu registrieren, die aber bereits registriert waren. Wieso das passiert ist mir allerdings unklar. Kannst du mal testweise das Basic-Auth deaktivieren. Nach dem was du geschrieben hast läuft es dann, richtig?

Tedfierro, 2023-09-04:
Genau, nehme ich die Basic Auth raus aus der Config, dann gehts. Gerade probiert.

naltatis, 2023-09-04:
Vielleicht ist es ein Caching Problem. Also dass der Browser die Datei nicht behält weil Credentials im Pfad sind. Momentan liefern wir CSS/JS mit Cache-Control: no-cache aus. In den Release-Builds könnten wir das Caching erlauben da wir eindeutige URLs (content-hash) haben.

Kannst du bei dir im Apache mal probieren einen Cache-Header hinzuzufügen? Also dass die `/assets/*` Ressourcen sowas bekommen: `cache-control: private, max-age=3600`

Tedfierro, 2023-09-04:
Ich habe diesen Code nun hinzugefügt und vorher noch den "Headers" Mod aktiviert:
`<Directory "/assets/*>
Header set Cache-Control "max-age=3600, private"
</Directory>
`
Gleiches Verhalten. Ist das so richtig?

naltatis, 2023-09-05:
> Gleiches Verhalten.
Schade, hatte gehofft, dass das evtl. einen Unterschied macht.

Chrome supportet Credentials in URL aus Sicherheitsgründen auch schon länger nicht mehr:
https://medium.com/@lmakarov/say-goodbye-to-urls-with-embedded-credentials-b051f6c7b6a3

Ich würd den Issue hier zu machen. Wenn du einen einfachen Weg findest das doch noch zu supporten können wir das gerne machen, aber würde dafür jetzt keine extra Hacks oder so einführen.

closed 2023-09-05 by naltatis