evcc-io/evcc #9669: Experimental: neues Fahrzeug wird erst nach Neustart angezeigt

VolkerK62, 2023-09-01:
### Describe the bug

Ich habe im UI über "Configuration" eine neues generisches Fahrzeug angelegt.
In den Fahrzeugauswahl am Loadpoint erscheint dieses jedoch erst nach einem Restart.
Wenn man das Fahrzeug wieder löscht, wird es auch erst nach einem Restart aus der Liste entfernt.

@naltatis meinte auf Slack, dass sollte auch ohne Restart funktionieren.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.119.2 (1faa6662)

VolkerK62, 2023-09-01:
Das dieses Fahrzeug bei `evcc vehicle` nicht aufgeführt wird, ist wahrscheinlich richtig, weil es nicht in der yaml enthalten ist?

andig, 2023-09-02:
@naltatis sollen wir auf einanderes Api umsteign an der Stelle oder- alternativ oder zusätzlich- die vehicles in /state nochmal publizieren?

naltatis, 2023-09-03:
Gute Frage. Das einfachste wäre jetzt hier vmtl. ein neues Publishing. Momentan ists ja nur die Liste der `titles`.

Wir sollten aber durchaus mal darüber nachdenken, die (`vehicle*`) Eigenschaften am `lp` in `vehicle` Objekte zu verlagern und am `lp` nur noch eine Fahrzeugreferenz (erstmal idx, später id) zu veröffentlichen. Das ist aber ein größeres Thema. Zu der Anschlussfrage: WS-Publish oder REST-Endpunkt habe ich noch keine klare Meinung. Ich würde momentan zu  publish tendieren, weil dann kein separater Invalidation Mechnamius in die UI muss.

andig, 2023-09-09:
Weitere Lücke: `loadpoint.Coordinator` kennt die geänderten Fahrzeuge nicht.

closed 2023-09-17 by andig