evcc-io/evcc #9671: Fast Charging did not adjust

infinus, 2023-09-01:
### Describe the bug

Changing setting to charge now/fast did not alter speed

### Steps to reproduce

1. plug in vehicle
2. set charge status to fast

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 5e866f7fa74360b3e83390f7677cc63bcaa6a0e5542a0cdccffec90af47b531c

interval: 10s # control cycle interval

meters:
- type: template
  template: enphase
  usage: grid
  host: 192.168.1.225
  token: Removed for security
  name: grid1
- type: template
  template: enphase
  usage: pv
  host: 192.168.1.225
  token: removed for security
  name: pv2

chargers:
- type: template
  template: twc3
  host: 192.168.1.108
  name: wallbox4

vehicles:
- type: template
  template: tesla
  title: Infinus
  accessToken: removed for security
  refreshToken: removed for security
  vin: removed
  name: ev3

loadpoints:
- title: Tesla
  charger: wallbox4
  vehicle: ev3
  mode: pv
  phases: 1
  mincurrent: 5
  maxcurrent: 48
  resetOnDisconnect: true

tariffs:
  currency: USD
  grid:
    # static grid price
[... 391 lines cut ...]
[lp-1  ] DEBUG 2023/09/01 23:51:29 vehicle soc: 36%
[lp-1  ] DEBUG 2023/09/01 23:51:29 vehicle soc limit: 90%
[lp-1  ] DEBUG 2023/09/01 23:51:29 vehicle range: 155km
[site  ] DEBUG 2023/09/01 23:51:35 ----
[lp-1  ] DEBUG 2023/09/01 23:51:35 charge power: 1233W
[site  ] DEBUG 2023/09/01 23:51:37 pv power: 2709W
[site  ] DEBUG 2023/09/01 23:51:38 grid power: 4247W
[site  ] DEBUG 2023/09/01 23:51:38 site power: 4247W
[lp-1  ] DEBUG 2023/09/01 23:51:38 charge voltages: [247 0 0]V
[lp-1  ] DEBUG 2023/09/01 23:51:38 detected connected phases: 1p
[lp-1  ] DEBUG 2023/09/01 23:51:38 charge currents: [5 0 0]A
[lp-1  ] DEBUG 2023/09/01 23:51:38 detected active phases: 1p
[lp-1  ] DEBUG 2023/09/01 23:51:38 charger status: C
[lp-1  ] DEBUG 2023/09/01 23:51:38 vehicle soc: 36%
[lp-1  ] DEBUG 2023/09/01 23:51:38 vehicle soc limit: 90%
[lp-1  ] DEBUG 2023/09/01 23:51:38 vehicle range: 155km
```

### What type of operating system are you running?

Docker container

### Version

0.119.3

infinus, 2023-09-01:
I restarted the container and now it seems to work but not sure why this occured in the first place? I noticed earlier in the day when I first connected the vehicle SOC did not update and status shows asleep. When first connecting charger vehicle should not be asleep?

`
```
[site  ] DEBUG 2023/09/01 22:58:25 ----
--
[lp-1  ] DEBUG 2023/09/01 22:58:25 charge power: 0W
[site  ] DEBUG 2023/09/01 22:58:27 pv power: 1283W
[site  ] DEBUG 2023/09/01 22:58:28 grid power: 280W
[site  ] DEBUG 2023/09/01 22:58:28 site power: 280W
[lp-1  ] DEBUG 2023/09/01 22:58:28 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/09/01 22:58:28 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/01 22:58:28 charger status: A
[site  ] DEBUG 2023/09/01 22:58:35 ----
[lp-1  ] DEBUG 2023/09/01 22:58:36 charge power: 0W
[site  ] DEBUG 2023/09/01 22:58:38 pv power: 1288W
[site  ] DEBUG 2023/09/01 22:58:39 grid power: 195W
[site  ] DEBUG 2023/09/01 22:58:39 site power: 195W
[lp-1  ] DEBUG 2023/09/01 22:58:39 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/09/01 22:58:39 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/01 22:58:39 charger status: B
[lp-1  ] INFO 2023/09/01 22:58:39 car connected
[lp-1  ] DEBUG 2023/09/01 22:58:39 pv timer elapse
[lp-1  ] DEBUG 2023/09/01 22:58:39 pv timer inactive
[lp-1  ] DEBUG 2023/09/01 22:58:39 charger: guard elapse
[lp-1  ] DEBUG 2023/09/01 22:58:39 guard timer inactive
[lp-1  ] ERROR 2023/09/01 22:58:49 vehicle soc: asleep
[lp-1  ] DEBUG 2023/09/01 22:58:49 pv charge current: 0A = 0A + -0.849A (195W @ 1p)
[lp-1  ] DEBUG 2023/09/01 22:58:49 pv enable timer reset
[lp-1  ] DEBUG 2023/09/01 22:58:49 pv timer inactive
[site  ] DEBUG 2023/09/01 22:58:49 ----
[lp-1  ] DEBUG 2023/09/01 22:58:49 charge power: 0W
[site  ] DEBUG 2023/09/01 22:58:51 pv power: 1290W
[site  ] DEBUG 2023/09/01 22:58:52 grid power: 303W
[site  ] DEBUG 2023/09/01 22:58:52 site power: 303W
[lp-1  ] ERROR 2023/09/01 22:58:52 vehicle odometer: asleep
[lp-1  ] DEBUG 2023/09/01 22:58:52 charge voltages: [2.1 2.1 2.1]V
[lp-1  ] DEBUG 2023/09/01 22:58:52 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/09/01 22:58:52 charger status: B
[lp-1  ] DEBUG 2023/09/01 22:58:52 pv charge current: 0A = 0A + -1.32A (303W @ 1p)
```
`

andig, 2023-09-02:
With twc, control works via the Tesla. For diagnosis please set Tesla log to trace to see whats going on.

closed 2023-09-02 by andig